bluez/bluez #896: HID-over-GATT: Extra byte in output reports

hglassdyb, 2024-07-10:
Hello,

I've been trying to adapt an open source FIDO security key firmware to Zephyr RTOS. It worked fine when using HID with the USB transport layer. However, it had issues when using HID-over-GATT with BlueZ.

It seems that the output reports have an extra byte `0x00` that precedes the expected report.

```
Report from USB: ff ff ff ff 86 00 08 4c b7 c4 91 c9 e7 4c 3f 00 00 00 00 ... (64 bytes sent)
Report from HOG: 00 ff ff ff ff 86 00 08 4c b7 c4 91 c9 e7 4c 3f 00 00 00 00 ... (65 bytes sent)
```

This happens when trying to authenticate on a browser with the security key or when using the configuration tool. But the data is correctly transferred when writing directly to the hidraw device.

I thought it could be the report ID, but my device doesn't define any in the report descriptor, so it should be omitted.

I managed to make it work by using a dirty hack where I strip the extra byte:

 ```diff
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 1b4aca074..48f4e79fd 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -758,7 +758,7 @@ static void forward_report(struct uhid_event *ev, void *user_data)
        data = ev->u.output.data;
        size = ev->u.output.size;
 
-   if (report->numbered && size > 0) {
+ if (!ev->u.output.data[0] || (report->numbered && size > 0)) {
                data++;
                --size;
        }
```

But I'm not sure where the problem is, if my device is non-compliant or if it's coming from the computer side.

Vudentz, 2024-07-10:
@hglassdyb we detect if the report is numered, has a report ID by doing:

```
	case HOG_REPORT_TYPE_OUTPUT:
		if (hog->uhid_flags & UHID_DEV_NUMBERED_OUTPUT_REPORTS)
			report->numbered = true;
		break;
```

Not really sure why the kernel would append a the report ID if it is not indicating it via uhid_flags, anyway  don't think we can't just check the first byte like that since it may be valid to send data starting with by 0x00.

hglassdyb, 2024-07-11:
@Vudentz Indeed, that's what I thought.

And the weird thing is that when I write directly to the hidraw device, the extra byte is not there. For example:

```bash
# Data sent from the host
printf "\xde\xad\xbe\xef" > /dev/hidraw4

# Debug log on the device
[DUMP] >> de ad be ef
```

So maybe this extra byte is added from the application side?

ValdikSS, 2024-09-10:
> Not really sure why the kernel would append a the report ID if it is not indicating it via uhid_flags

Sorry I'm totally ignorant and haven't read this conversation thoroughly, but buf[0] is ignored by the kernel's `hid_hw_raw_request` function (it's skipped), but not by `hid_hw_output_report` for example.

https://github.com/torvalds/linux/blob/8d8d276ba2fb5f9ac4984f5c10ae60858090babc/drivers/hid/hid-lenovo.c#L495

closed 2025-07-22 by github-actions[bot]