bluez/bluez #898: The LE Audio broadcast sink does not probe

dabeixiaomibujiatang, 2024-07-11:
BlueZ 5.77 + PipeWire1.2.0 + Linux version 6.6.18
    The broadcast assistant helps the broadcast sink sync to the broadcast source. The broadcast sink is able to receive BIS data from the broadcast source, but no sound is heard from the broadcast sink. I analyzed the bluetoothd.log of the broadcast sink and found no bap_bcast_probe(). Comparing the bluetoothd.log of the LE Audio CIS, I noticed that bap_probe() needs to connect to the remote BAP profile. However, the broadcast assistant only has the Broadcast Audio Scan profile and does not have the BAP profile. Does the bap_bcast_probe() of the broadcast sink also require the broadcast assistant to contain the BAP profile? Is my understanding incorrect?
logs:
cis_sink:
[bluetoothd_sink.log](https://github.com/user-attachments/files/16175405/bluetoothd_sink.log)
bis_sink:
[bluetoothd.log](https://github.com/user-attachments/files/16175409/bluetoothd.log)
[btsnoop_sink.cfa.log](https://github.com/user-attachments/files/16175427/btsnoop_sink.cfa.log)
bis_assistant_hci_log:
[btsnoop_assistant.cfa.log](https://github.com/user-attachments/files/16175428/btsnoop_assistant.cfa.log)

Vudentz, 2024-07-11:
@dabeixiaomibujiatang looks like you are confusing roles, the assitant is for transferring a broadcast to another device, if you want to act as broadcast sink yourself then you will need to scan, you can check out how it is done via bluetoothctl with the likes of https://github.com/bluez/bluez/blob/master/client/scripts/broadcast-sink.bt

dabeixiaomibujiatang, 2024-07-12:
Thank you for your reply.
    My buildroot environment is set to the broadcast sink role. My broadcast assistant can scan the surrounding broadcasts and sync the broadcast sink to the broadcast source via the Broadcast Audio Scan profile. I found that BlueZ can also implement the scanning broadcast feature.
    Based on your suggestion, I created a broadcast sink using the commands in broadcast-sink.bt. I found that the effect of the commands in broadcast-sink.bt is the same as adding bluez5.roles = [ bap_sink bap_bcast_sink ] in /etc/wireplumber/wireplumber.conf.d/bluetooth.conf, which also created Endpoint registered: sender=:1.4 path=/MediaEndpointLE/BAPBroadcastSink/lc3.
    Now I have synced the BlueZ-related code in the kernel with bluez/bluetooth-next, and the bluez5-utils code with bluez/bluez. One change is that the log now shows that the broadcast source is also connected. However, endpoint.list and transport.list are still empty, and there is no sound from the broadcast sink.

logs:
bis_sink:
[bluetoothd.log](https://github.com/user-attachments/files/16192387/bluetoothd.log)
bis_source_addr:46:7A:28:32:93:AB
bis_assistant_addr:C0:07:E8:A6:B5:5B
bis_hci_log:
[btsnoop.cfa.log](https://github.com/user-attachments/files/16192388/btsnoop.cfa.log)
pipewire_log:
[logs-wireplumber.txt](https://github.com/user-attachments/files/16192389/logs-wireplumber.txt)

dabeixiaomibujiatang, 2024-07-16:
Sorry, I misunderstood. The issue has been resolved. 
Once the broadcast source is ready and starts broadcasting, the broadcast sink registers the endpoint according to the broadcast-sink.bt script, then turns on scanning. When the broadcast source is detected, it automatically syncs to the broadcast source and begins receiving ISO data. The entire process does not require a broadcast assistant.

closed 2024-07-16 by dabeixiaomibujiatang

michael-kong754, 2024-12-10:
> Sorry, I misunderstood. The issue has been resolved. Once the broadcast source is ready and starts broadcasting, the broadcast sink registers the endpoint according to the broadcast-sink.bt script, then turns on scanning. When the broadcast source is detected, broadcast sink will automatically syncs to the broadcast source and begins receiving ISO data. The entire process does not require a broadcast assistant.

Hi @dabeixiaomibujiatang ,
  I also use the broadcast-source.bt and broad-sink.bt in https://github.com/bluez/bluez/tree/master/client/scripts to test LE Audio, I use the BlueZ 5.77 and kernel 6.6.50 and without pipewire. However, the broadcast sink cannot detect any broadcast source device, are there any additional instructions or configurations that need to be set (like in src/main.conf)?
Thank you!

dabeixiaomibujiatang, 2024-12-11:
hi, michael-kong754

main.conf
ControllerMode = le
Experimental = true
KernelExperimental = 6fbaf188-05e0-496a-9885-d6ddfdb4e03e

Without pipewire, you don't have a complete le audio environment and can't test le audio functionality. You can use isotest to test only cis &amp; bis in iso links. The isotest tool is a specialized tool for testing iso links provided by bluez, whose source code is located in tools\isotest.c.

         原始邮件

发件时间：         2024年12月10日 16:58 
收件人：                  "bluez/bluez"         ***@***.***&gt;                 
抄送：                  "bujia_tang"         ***@***.***&gt;         ,                "Mention"         ***@***.***&gt;

michael-kong754, 2025-02-21:
Hi @dabeixiaomibujiatang ,
  Thank you for your replying, I can use run broadcast-source role in bluez use pipewire now, and i can see ISO Date TX use btmon.
  However, the broadcast-sink cannot sync with broadcast-source. Have you ever met this problem? Here is my config and test procedure:
 For broadcast-source: 1) run pipewire and bluetoothd; 2)then execute pw-play;
 For boradcast-sink: 1)run pipewire and bluetoothd; 2)run "scan on" in bluetoothctl;
 For bluetooth.conf:
   bluez5.roles = [ bap_bcast_source bap_bcast_sink ]
   bluez5.bcast_source.config = [
    {
      "broadcast_code": "Børne House",
      "encryption": false,
      "bis": [
        { # BIS configuration
          "qos_preset": "16_2_1", # QOS preset name from table Table 6.4 from BAP_v1.0.1.
          "audio_channel_allocation": 1, # audio channel allocation configuration for the BIS
          "metadata": [ # metadata configurations for the BIS
             { "type": 1, "value": [ 1, 1 ] }
          ]
        }
      ]
    }
  ]

Best Regards!

dabeixiaomibujiatang, 2025-03-07:
hi,michael-kong754
https://github.com/bluez/bluez/issues/1033
it need 
transport.select                 /org/bluez/hci0/dev_17_7A_80_64_A7_93/bis1/fd0
or 
transport.select                 /org/bluez/hci0/dev_17_7A_80_64_A7_93/bis1/fd0                 /org/bluez/hci0/dev_17_7A_80_64_A7_93/bis2/fd1
 

         原始邮件

发件时间：2025年2月21日 18:19

raidoum, 2025-03-29:
@dabeixiaomibujiatang @michael-kong754 

I have the same problem, and if you have already solved it, could you please tell me how to solve it?

The following is what I have confirmed. What procedure is missing?

1. [Source] Run Broadcast-source.bt
2. [Sink] Run Broadcast-sink.bt
3. [Sink] "Device XX:XX:XX:XX:XX Connected: yes" log is displayed
4. [Source] Run pw-play ***.mp3
➔ Check that ISO packets are being sent with btmon
5. [Sink] "Transport /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/bis1/fd0" log is displayed
6. [Sink] When I check the btmon log, I can see that Broadcast Audio Announcement is detected, but ISO packets are not being received.

michael-kong754, 2025-04-02:
> [@dabeixiaomibujiatang](https://github.com/dabeixiaomibujiatang) [@michael-kong754](https://github.com/michael-kong754)
>
> I have the same problem, and if you have already solved it, could you please [...]

Hi @raidoum 
  In my opinion, you should not register endpoints with both pipewire and bluez at the same time, as this will cause them to affect each other. See Vudentz's reply in:
https://github.com/bluez/bluez/issues/1030

raidoum, 2025-04-16:
@michael-kong754 
Thank you for your reply.

You said that I should try it in an environment where pipewire doesn't work, but does that mean that you haven't been able to confirm that broadcast works with pipewire+bluez in your environment?