bluez/bluez #901: Unable to monitor for patterns with data types other than 1 (Raspberry Pi 3b+)

LeCyberDucky, 2024-07-15:
Hey there!

Please note that the title of this issue is not the best, since I don't exactly know whether the data type is at the core of the problem. But consider the following: 

I have a device that produces the following output in `btmon`: 

```
> HCI Event: LE Meta Event (0x3e) plen 36                                                        #34 [hci0] 1134.929495
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: D8:E7:2F:A6:11:21 (Chipsea Technologies (Shenzhen) Corp.)
        Data length: 24
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 1 entry
          Body Composition (0x181b)
        Service Data: Body Composition (0x181b)
          Data[13]: 0204b207011209080a0000083e
        RSSI: -50 dBm (0xce)
@ MGMT Event: ADV Monitor Device Found (0x002f) plen 40                                     {0x0001} [hci0] 1134.929549
        Handle: 0
        Address: D8:E7:2F:A6:11:21 (Chipsea Technologies (Shenzhen) Corp.)
        Addr Type: 1
        RSSI: -50
        Flags: 0x00000000
        AD Data Len: 24
        AD Data[24]: 02010603021b1810161b180204b207011209080a0000083e
```
I would like to monitor for the advertisement data of this device. Based on discussion in https://github.com/hbldh/bleak/discussions/1612 (and this issue that I just stumbled upon,  https://github.com/bluez/bluez/issues/652), I have tried to add some filters derived from the `AD Data` field, like so: 

```bash
bluetoothctl
menu monitor
add-or-pattern 0 16 1b18
add-or-pattern 0 02 1b18
```

This gives me no hits. 
The only thing that works, is `add-or-pattern 0 01 06`, which instantly yields the results I'm looking for. 
Any ideas as to what is going wrong here?

I'm using a Raspberry Pi 3b+ running Raspbian 12, bookworm (Linux 6.1.41-v7+ armv7l). I have just built bluez from master a few minutes ago.

Vudentz, 2024-07-15:
@LeCyberDucky the format is the following:
```

	pat->start_pos = atoi(parameter_list[0]);
	pat->ad_data_type = atoi(parameter_list[1]);
	pat->content_len = str2bytearray(parameter_list[2], pat->content);
```

So it should probably always start from 0, then ad_data_type should be BT_AD_UUID16_SOME(0x02) followed by "0x1b 0x18":

`bluetoothctl> monitor.add-or-pattern 0 0x02 "0x1b 0x18"`

LeCyberDucky, 2024-07-15:
Sorry, I may be misunderstanding, but 

`monitor.add-or-pattern 0 0x02 "0x1b 0x18"`

gives me 

`pattern-list malformed`.

If I remove the space from `"0x1b 0x18"`, the pattern is accepted, but the whole pattern ends up looking strange, I think: 
```
Advertisement Monitor 0
        path: /org/bluez/adv_monitor_app/0
        type: or_patterns
        pattern 1:
                start position: 0
                AD data type: 0
                content: 001b0018
```

dlech, 2024-07-16:
I think it doesn't like the `0x` prefix and thinks you are wanting `0`.

Try `add-or-pattern 0 16 1b18` or `add-or-pattern 0 2 1b18`.

LeCyberDucky, 2024-07-17:
@dlech Right, I should have clarified that, sorry. The first thing I tried was `add-or-pattern 0 16 1b18`, which yields no results. But let me summarize below: 

When I do 

```
add-or-pattern 0 16 1b18
add-or-pattern 0 02 1b18
```
I get the following output from `btmon`: 

```
@ MGMT Command: Add Advertisement Patterns Monitor (0x0052) plen 35                           {0x0001} [hci0] 68.272534
        Number of patterns: 1
          Pattern 1:
            AD type: 16
            Offset: 0
            Length: 2
            Value [2]: 1b18
@ MGMT Event: Command Complete (0x0001) plen 5                                                {0x0001} [hci0] 68.272625
      Add Advertisement Patterns Monitor (0x0052) plen 2
        Status: Success (0x00)
        Handle: 1
```

```
@ MGMT Command: Add Advertisement Patterns Monitor (0x0052) plen 35                          {0x0001} [hci0] 180.610648
        Number of patterns: 1
          Pattern 1:
            AD type: 2
            Offset: 0
            Length: 2
            Value [2]: 1b18
@ MGMT Event: Command Complete (0x0001) plen 5                                               {0x0001} [hci0] 180.610730
      Add Advertisement Patterns Monitor (0x0052) plen 2
        Status: Success (0x00)
        Handle: 2
```

Additionally, here is the output from `get-pattern all`: 

```
[bluetooth]# monitor.get-pattern all
Advertisement Monitor 0
        path: /org/bluez/adv_monitor_app/0
        type: or_patterns
        pattern 1:
                start position: 0
                AD data type: 16
                content: 1b18
Advertisement Monitor 1
        path: /org/bluez/adv_monitor_app/1
        type: or_patterns
        pattern 1:
                start position: 0
                AD data type: 2
                content: 1b18
```

I believe this shows that the commands are parsed as intended. 
But I receive absolutely no results with these patterns. 

As soon as I do `add-or-pattern 0 01 06`, however, I immediately start receiving results for this specific pattern. See: 

```
[bluetooth]# monitor.add-or-pattern 0 01 06
Advertisement Monitor 2 added
[bluetooth]# Advertisement monitor 2 activated
[bluetooth]# [NEW] Device D8:E7:2F:A6:11:21 MIBFS
[bluetooth]# Advertisement monitor 2 found device /org/bluez/hci0/dev_D8_E7_2F_A6_11_21
[bluetooth]# [NEW] Device A0:CD:F3:94:96:41 EOSR6m2_949640
[bluetooth]# Advertisement monitor 2 found device /org/bluez/hci0/dev_A0_CD_F3_94_96_41
```

For good measure, here's the `btmon` output for that pattern as well: 

```
@ MGMT Command: Add Advertisement Patterns Monitor (0x0052) plen 35                          {0x0001} [hci0] 402.620725
        Number of patterns: 1
          Pattern 1:
            AD type: 1
            Offset: 0
            Length: 1
            Value [1]: 06
@ MGMT Event: Command Complete (0x0001) plen 5                                               {0x0001} [hci0] 402.620814
      Add Advertisement Patterns Monitor (0x0052) plen 2
        Status: Success (0x00)
        Handle: 3
```

And `get-pattern all`:

```
Advertisement Monitor 0
        path: /org/bluez/adv_monitor_app/0
        type: or_patterns
        pattern 1:
                start position: 0
                AD data type: 16
                content: 1b18
Advertisement Monitor 1
        path: /org/bluez/adv_monitor_app/1
        type: or_patterns
        pattern 1:
                start position: 0
                AD data type: 2
                content: 1b18
Advertisement Monitor 2
        path: /org/bluez/adv_monitor_app/2
        type: or_patterns
        pattern 1:
                start position: 0
                AD data type: 1
                content: 06
```

@Vudentz Following the above, I believe that the issue does not stem from the parsing of my commands. Do you agree, or am I missing something? Like @dlech points out, I believe that a hex-formatted content string causes funky things to happen in the parsing of patterns (misinterpreting characters as zeros). 

Btw, I also tried `add-or-pattern 0 22 1b18`, just to make sure. That didn't work either.

hw-lunemann, 2024-09-17:
I ran into the same problem using `bluer`. Nothing works except for the `0x1` data type.

Vudentz, 2024-09-18:
Correction, I think the standard replacement is actually what is described as 5. Monitoring Advertisers:

https://www.bluetooth.com/core-specification-6-feature-overview/?utm_source=internal&utm_medium=blog&utm_campaign=technical&utm_content=now-available-new-version-of-the-bluetooth-core-specification

hw-lunemann, 2024-09-18:
I've run into the exact same issue, maybe it has to do with filtering on a data type that isn't the first one in the LE advertising data?

logan12358, 2025-03-20:
I ran into this issue as well, trying to match against a 16 bit service data UUID.

From looking through the code, it seems related to how [`struct bt_ad`](https://github.com/bluez/bluez/blob/f77e72c3ab11d8c06f487ff7892afef2aefbd0f6/src/shared/ad.c#L28) has separate structures for some specific AD types (i.e. service UUIDs (0x02-0x07), manufacturer data (0xff), service data (0x16, 0x20, 0x21), name (0x08, 0x09), and all other data together), parsed [here](https://github.com/bluez/bluez/blob/f77e72c3ab11d8c06f487ff7892afef2aefbd0f6/src/shared/ad.c#L357). In the [`pattern_match`](https://github.com/bluez/bluez/blob/f77e72c3ab11d8c06f487ff7892afef2aefbd0f6/src/shared/ad.c#L1430) implementation there is _some_ special handling for the different AD types, but I don't think it works as expected:
* For patterns with a service data AD type, [`match_service`](https://github.com/bluez/bluez/blob/f77e72c3ab11d8c06f487ff7892afef2aefbd0f6/src/shared/ad.c#L1365) compares the pattern with `bt_ad_service_data`'s `data` but not `uuid`, which is parsed out of the AD data earlier.
* For patterns with a service UUID AD type (same for names), they are matched against `bt_ad`'s `data` field, not the `service_uuids` field that service UUIDs are parsed into.

In my case I have a small hacky patch (included below) to `match_service` which compares against the UUID if `offset` is specified as `BT_AD_MAX_DATA_LEN+1` (32). It's not a good fix, but it matches as I expect.

Hope this makes sense, all the Bluetooth terminology is dense 😬 I'm happy to try to contribute a fix, if there's a clear direction on the correct API. Also happy to test anything with what hardware I have available.

<details><summary>Hacky patch</summary>

```
diff --git a/src/shared/ad.c b/src/shared/ad.c
index d08ce7af9..dd9c72298 100644
--- a/src/shared/ad.c
+++ b/src/shared/ad.c
@@ -1362,6 +1362,63 @@ static bool match_manufacturer(const void *data, const void *user_data)
 	return false;
 }
 
+static bool bt_uuid16_create_with_data(bt_uuid_t *uuid, struct iovec *iov)
+{
+	uint16_t value;
+
+	if (!util_iov_pull_le16(iov, &value))
+		return false;
+
+	if (bt_uuid16_create(uuid, value))
+		return false;
+
+	return true;
+}
+
+static bool bt_uuid32_create_with_data(bt_uuid_t *uuid, struct iovec *iov)
+{
+	uint32_t value;
+
+	if (!util_iov_pull_le32(iov, &value))
+		return false;
+
+	if (bt_uuid32_create(uuid, value))
+		return false;
+
+	return true;
+}
+
+static bool bt_uuid128_create_with_data(bt_uuid_t *uuid, struct iovec *iov)
+{
+	void *data;
+	uint128_t value;
+
+	data = util_iov_pull_mem(iov, 16);
+	if (!data)
+		return false;
+
+	bswap_128(data, &value);
+
+	if (bt_uuid128_create(uuid, value))
+		return false;
+
+	return true;
+}
+
+static bool bt_uuid_create_with_data(bt_uuid_t *uuid, struct iovec *iov)
+{
+	switch (iov->iov_len) {
+		case 2:
+			return bt_uuid16_create_with_data(uuid, iov);
+		case 4:
+			return bt_uuid32_create_with_data(uuid, iov);
+		case 16:
+			return bt_uuid128_create_with_data(uuid, iov);
+		default:
+			return false;
+	}
+}
+
 static bool match_service(const void *data, const void *user_data)
 {
 	const struct bt_ad_service_data *service_data = data;
@@ -1388,6 +1445,22 @@ static bool match_service(const void *data, const void *user_data)
 		return false;
 	}
 
+	if (pattern->offset == BT_AD_MAX_DATA_LEN + 1) {
+		struct iovec iov = {
+			.iov_base = (void *)pattern->data,
+			.iov_len = pattern->len,
+		};
+
+		bt_uuid_t pattern_uuid;
+		if (!bt_uuid_create_with_data(&pattern_uuid, &iov))
+			return false;
+
+		if (!bt_uuid_cmp(&pattern_uuid, &service_data->uuid))
+			return true;
+
+		return false;
+	}
+
 	if (service_data->len < pattern->offset + pattern->len)
 		return false;
```

</details>