bluez/bluez #902: After pairing successfully, low power mouse and keyboard connections cannot be received

caitao123456789, 2024-07-16:
Bluez version is greater than or equal to 5.69 , when paired successfully,  turn Bluetooth off and then on, unable to receive a low-power mouse and keyboard connection.

...
...

After analysis, the mgmt command(MGMT_OP_ADD_DEVICE) was not sent to the kernel after the pairing connection was completed.

...
...

_**Here's how to fix the patch**_
```c
diff --git a/src/device.c b/src/device.c
index 097b1fbba..7b18781c8 100644
--- a/src/device.c
+++ b/src/device.c
@@ -6062,6 +6062,10 @@ void btd_device_set_temporary(struct btd_device *device, bool temporary)
 
 	if (device->bredr)
 		adapter_accept_list_add(device->adapter, device);
+	else if (device->le) {
+		device->disable_auto_connect = FALSE;
+		device_set_auto_connect(device, TRUE);
+	}
 
 	store_device_info(device);
```

gabriele2000, 2024-07-21:
If you know how to fix it, why didn't you open a pull request?
Go my friend, make everyone proud and open a pull request ;-)

Vudentz, 2024-07-22:
> ```c
> diff --git a/src/device.c b/src/device.c
> index 097b1fbba..7b18781c8 100644
> --- a/src/device.c
> +++ b/src/device.c
> @@ -6062,6 +6062,10 @@ void btd_device_set_temporary(struct btd_device *device, [...]

Wrong fix, you would have to check if device->auto_connect has been set and that is done by probe_service which probably means that we are not setting auto_connect flag because there are no matching services:

```c
	/* Only set auto connect if profile has set the flag and can really
	 * accept connections.
	 */
	if (profile->auto_connect && profile->accept) {
		/* If temporary mark auto_connect as disabled so when the
		 * device is connected it attempts to enable it.
		 */
		if (device->temporary)
			device->disable_auto_connect = TRUE;
		else
			device_set_auto_connect(device, TRUE);
	}
```

If you want to set auto_connect for a specific service not covered by any plugin then you need to use GattProfile interface:

https://github.com/bluez/bluez/blob/master/doc/org.bluez.GattProfile.rst

caitao123456789, 2024-07-23:
patch 507ba1248 ("profile: Remove probe_on_discover", 2023-08-17)
```c 
 	/* Only set auto connect if profile has set the flag and can really
 	 * accept connections.
 	 */
-	if (profile->auto_connect && profile->accept)
-		device_set_auto_connect(device, TRUE);
+	if (profile->auto_connect && profile->accept) {
+		/* If temporary mark auto_connect as disabled so when the
+		 * device is connected it attempts to enable it.
+		 */
+		if (device->temporary)
+			device->disable_auto_connect = TRUE;
+		else
+			device_set_auto_connect(device, TRUE);
+	}
 
 	return service;
``` 

In the paired connection test, disable_auto_connect is set to true because the device is temporary.  If you restore this patch code, the function device_set_auto_connect(device, TRUE) is executed.

This is new patch, determine the profile status after the device's temporary status is set to false.
```c
diff --git a/src/device.c b/src/device.c
index 882465162..7df5f1c3a 100644
--- a/src/device.c
+++ b/src/device.c
@@ -2323,6 +2323,25 @@ static struct btd_service *find_connectable_service(struct btd_device *dev,
 	return NULL;
 }
 
+
+static bool device_is_set_auto_connect(struct btd_device *dev)
+{
+	if (dev->services) {
+		GSList *l;
+
+		for (l = dev->services; l != NULL; l = g_slist_next(l)) {
+			struct btd_service *service = l->data;
+			struct btd_profile *p = btd_service_get_profile(service);
+
+			if (p->auto_connect && p->accept)
+				return true;
+		}
+	}
+
+	return false;
+}
+
+
 static int service_prio_cmp(gconstpointer a, gconstpointer b)
 {
 	struct btd_profile *p1 = btd_service_get_profile(a);
@@ -6063,7 +6082,7 @@ void btd_device_set_temporary(struct btd_device *device, bool temporary)
 
 	if (device->bredr)
 		adapter_accept_list_add(device->adapter, device);
-	else if (device->le) {
+	else if (device->le && device_is_set_auto_connect(device)) {
 		device->disable_auto_connect = FALSE;
 		device_set_auto_connect(device, TRUE);
 	}

```

Vudentz, 2024-07-24:
@caitao123456789 could you please add the bluetoothd logs? Does it only happens if you turn it power off or it still happens when you just disconnect? As for the code perhaps you want to do just: else if (device->le && device->disable_auto_connect) since setting disable_auto_connect already indicates that auto_connect has been disabled.

caitao123456789, 2024-07-24:
Use branch 68864d1aa ("Release 5.77", 2024-07-08)

patch : add log
```c
diff --git a/src/device.c b/src/device.c
index 097b1fbba..75e12b98d 100644
--- a/src/device.c
+++ b/src/device.c
@@ -4922,10 +4922,15 @@ static struct btd_service *probe_service(struct btd_device *device,
 		/* If temporary mark auto_connect as disabled so when the
 		 * device is connected it attempts to enable it.
 		 */
-		if (device->temporary)
+		if (device->temporary) {
 			device->disable_auto_connect = TRUE;
-		else
+			DBG("device->temporary is false");
+		}	
+		else {
 			device_set_auto_connect(device, TRUE);
+			DBG("device_set_auto_connect set true");
+		}
+			
 	}
 
 	return service;
diff --git a/src/log.h b/src/log.h
index 1ed742a0d..6260df88f 100644
--- a/src/log.h
+++ b/src/log.h
@@ -57,8 +57,8 @@ void __btd_enable_debug(struct btd_debug_desc *start,
 } while (0)
 
 #define DBG(fmt, arg...) \
-	DBG_IDX(0xffff, "%s:%s() " fmt, __FILE__, __func__, ## arg)
+	DBG_IDX(0xffff, "%s:%s():%d " fmt, __FILE__, __func__, __LINE__, ## arg)
 #define error(fmt, arg...) \
-	btd_error(0xffff, "%s:%s() " fmt, __FILE__, __func__, ## arg)
+	btd_error(0xffff, "%s:%s():%d " fmt, __FILE__, __func__, __LINE__, ## arg)
 #define warn(fmt, arg...) \
-	btd_warn(0xffff, "%s:%s() " fmt, __FILE__, __func__, ## arg)
+	btd_warn(0xffff, "%s:%s():%d " fmt, __FILE__, __func__, __LINE__, ## arg)

``` 

[bluetoothd.log](https://github.com/user-attachments/files/16358033/bluetoothd.log)
bluetoothd key log:
```
bluetoothd[300213]: src/device.c:device_set_legacy():6131 legacy 0
bluetoothd[300213]: src/device.c:device_set_rssi_with_delta():6249 rssi -45
bluetoothd[300213]: src/device.c:device_set_tx_power():6286 tx_power 4
bluetoothd[300213]: src/device.c:btd_device_device_set_name():4398 /org/bluez/hci0/dev_F5_56_0B_E4_2F_3B M585/M590
bluetoothd[300213]: profiles/input/hog.c:hog_probe():151 path /org/bluez/hci0/dev_F5_56_0B_E4_2F_3B
bluetoothd[300213]: src/service.c:change_state():91 0x5603887b1840: device F5:56:0B:E4:2F:3B profile input-hog state changed: unavailable -> disconnected (0)
bluetoothd[300213]: src/device.c:probe_service():4927 device->temporary is false
bluetoothd[300213]: src/device.c:device_set_flags():6299 flags 5
bluetoothd[300213]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0012
bluetoothd[300213]: src/adapter.c:device_found_callback():7486 hci0 addr 65:EC:82:D0:83:48, rssi -78 flags 0x0000 eir_len 17
bluetoothd[300213]: src/device.c:device_create():4343 dst 65:EC:82:D0:83:48
bluetoothd[300213]: src/device.c:device_new():4251 address 65:EC:82:D0:83:48
bluetoothd[300213]: src/device.c:device_new():4288 Creating device /org/bluez/hci0/dev_65_EC_82_D0_83_48
bluetoothd[300213]: src/device.c:device_set_legacy():6131 legacy 0
bluetoothd[300213]: src/device.c:device_set_rssi_with_delta():6249 rssi -78
bluetoothd[300213]: src/device.c:device_set_tx_power():6286 tx_power 12
bluetoothd[300213]: src/device.c:device_set_flags():6299 flags 26
bluetoothd[300213]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0012
``` 

[btmon1.log](https://github.com/user-attachments/files/16358076/btmon1.log)
btmon key log:
Search for "MGMT Command:" and MGMT_OP_ADD_DEVICE is not found
```
  7,1: @ MGMT Command: Set Powered (0x0005) plen 1                                                    {0x0001} [hci0] 14:24:47.027798
  826,1: @ MGMT Command: Start Discovery (0x0023) plen 1                                                {0x0001} [hci0] 14:24:47.221048
  2203,1: @ MGMT Command: Confirm Name (0x0025) plen 8                                                   {0x0001} [hci0] 14:24:49.618722
  2346,1: @ MGMT Command: Confirm Name (0x0025) plen 8                                                   {0x0001} [hci0] 14:24:50.146435
  2389,1: @ MGMT Command: Set Connectable (0x0007) plen 1                                                {0x0001} [hci0] 14:24:50.269245
  2404,1: @ MGMT Command: Set Discoverable (0x0006) plen 3                                               {0x0001} [hci0] 14:24:50.269289
  2704,1: @ MGMT Command: Confirm Name (0x0025) plen 8                                                   {0x0001} [hci0] 14:24:51.441453
  8384,1: @ MGMT Command: Start Discovery (0x0023) plen 1                                                {0x0001} [hci0] 14:25:03.269229
  8960,1: @ MGMT Command: Stop Discovery (0x0024) plen 1                                                 {0x0001} [hci0] 14:25:04.484616
  9035,1: @ MGMT Command: Pair Device (0x0019) plen 8                                                    {0x0001} [hci0] 14:25:04.503288
  10791,1: @ MGMT Command: Start Discovery (0x0023) plen 1                                                {0x0001} [hci0] 14:25:06.050781
  11260,1: @ MGMT Command: Confirm Name (0x0025) plen 8                                                   {0x0001} [hci0] 14:25:07.792798
  11296,1: @ MGMT Command: Confirm Name (0x0025) plen 8                                                   {0x0001} [hci0] 14:25:07.889719
```

caitao123456789, 2024-07-24:
Use branch a3f9970f7 ("client: Add assistant submenu", 2024-07-16)

Use patch
```c
diff --git a/src/device.c b/src/device.c
index 7df5f1c3a..178cf0c9e 100644
--- a/src/device.c
+++ b/src/device.c
@@ -2323,25 +2323,6 @@ static struct btd_service *find_connectable_service(struct btd_device *dev,
 	return NULL;
 }
 
-
-static bool device_is_set_auto_connect(struct btd_device *dev)
-{
-	if (dev->services) {
-		GSList *l;
-
-		for (l = dev->services; l != NULL; l = g_slist_next(l)) {
-			struct btd_service *service = l->data;
-			struct btd_profile *p = btd_service_get_profile(service);
-
-			if (p->auto_connect && p->accept)
-				return true;
-		}
-	}
-
-	return false;
-}
-
-
 static int service_prio_cmp(gconstpointer a, gconstpointer b)
 {
 	struct btd_profile *p1 = btd_service_get_profile(a);
@@ -6082,7 +6063,7 @@ void btd_device_set_temporary(struct btd_device *device, bool temporary)
 
 	if (device->bredr)
 		adapter_accept_list_add(device->adapter, device);
-	else if (device->le && device_is_set_auto_connect(device)) {
+	else if (device->le && device->disable_auto_connect) {
 		device->disable_auto_connect = FALSE;
 		device_set_auto_connect(device, TRUE);
 	}

``` 

[bluetoothd2.txt](https://github.com/user-attachments/files/16358978/bluetoothd2.txt)
bluetoothd key log:
```
bluetoothd[403009]: src/device.c:device_set_rssi_with_delta() rssi 0
bluetoothd[403009]: src/device.c:device_set_rssi_with_delta() rssi 0
bluetoothd[403009]: src/device.c:device_set_tx_power() tx_power 127
bluetoothd[403009]: src/device.c:device_set_rssi_with_delta() rssi 0
bluetoothd[403009]: src/device.c:device_set_rssi_with_delta() rssi 0
bluetoothd[403009]: src/device.c:device_set_rssi_with_delta() rssi 0
bluetoothd[403009]: src/device.c:device_set_tx_power() tx_power 127
bluetoothd[403009]: src/adapter.c:trigger_passive_scanning() 
bluetoothd[403009]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0013
bluetoothd[403009]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0 method 0
bluetoothd[403009]: src/device.c:btd_device_set_temporary() temporary 0
bluetoothd[403009]: src/device.c:device_set_auto_connect() F5:56:0B:E4:2F:3E auto connect: 1
bluetoothd[403009]: src/agent.c:agent_ref() 0x55a5de81cea0: ref=2
bluetoothd[403009]: src/device.c:bonding_request_new() Requesting bonding for F5:56:0B:E4:2F:3E
bluetoothd[403009]: src/agent.c:agent_ref() 0x55a5de81cea0: ref=3
bluetoothd[403009]: src/agent.c:agent_unref() 0x55a5de81cea0: ref=2
bluetoothd[403009]: src/adapter.c:suspend_discovery() 
bluetoothd[403009]: src/adapter.c:adapter_bonding_attempt() hci0 bdaddr F5:56:0B:E4:2F:3E type 2 io_cap 0x01
bluetoothd[403009]: src/shared/mgmt.c:send_request() [0x0000] command 0x0033
bluetoothd[403009]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0033 complete: 0x00
bluetoothd[403009]: src/adapter.c:add_device_complete() F5:56:0B:E4:2F:3E (2) added to kernel connect list
bluetoothd[403009]: src/shared/mgmt.c:send_request() [0x0000] command 0x0019
``` 

[btmon2.txt](https://github.com/user-attachments/files/16359134/btmon2.txt)
btmon key log:
Search for "MGMT Command:" 
```
14713,1: @ MGMT Command: Start Discov.. (0x0023) plen 1  {0x0001} [hci0] 15:55:09.269805
  15062,1: @ MGMT Command: Confirm Name (0x0025) plen 8    {0x0001} [hci0] 15:55:09.816919
  15742,1: @ MGMT Command: Confirm Name (0x0025) plen 8    {0x0001} [hci0] 15:55:10.783965
  16914,1: @ MGMT Command: Stop Discovery (0x0024) plen 1  {0x0001} [hci0] 15:55:14.122570
  16960,1: @ MGMT Command: Add Device (0x0033) plen 8      {0x0001} [hci0] 15:55:14.131586
  16967,1: @ MGMT Command: Pair Device (0x0019) plen 8     {0x0001} [hci0] 15:55:14.131632
  19167,1: @ MGMT Command: Start Discov.. (0x0023) plen 1  {0x0001} [hci0] 15:55:15.922642
  19552,1: @ MGMT Command: Confirm Name (0x0025) plen 8    {0x0001} [hci0] 15:55:16.363743
  19659,1: @ MGMT Command: Confirm Name (0x0025) plen 8    {0x0001} [hci0] 15:55:16.411871
  20541,1: @ MGMT Command: Confirm Name (0x0025) plen 8    {0x0001} [hci0] 15:55:19.013158
  20711,1: @ MGMT Command: Confirm Name (0x0025) plen 8    {0x0001} [hci0] 15:55:19.342735
  20946,1: @ MGMT Command: Stop Discovery (0x0024) plen 1  {0x0001} [hci0] 15:55:19.932414
  21205,1: @ MGMT Command: Set Connecta.. (0x0007) plen 1  {0x0001} [hci0] 15:55:20.166284
  21252,1: @ MGMT Command: Set Powered (0x0005) plen 1     {0x0001} [hci0] 15:55:20.182926
```

caitao123456789, 2024-07-25:
> @caitao123456789 could you please add the bluetoothd logs? Does it only happens if you turn it power off or it still happens when you just disconnect? As for the code perhaps you want to do just: [...]

**Above is related log information**

@Vudentz 
After testing and code analysis, the problem can be solved by using variable device->disable_auto_connect as well as using functions device_is_set_auto_connect. However, using variable device->disable_auto_connect means less code modification  and is more concise.

caitao123456789, 2024-12-25:
394675782 ("device: Fix Pair method not setting auto_connect", 2024-08-29)

This patch has solved the issue, close the issue.

closed 2024-12-25 by caitao123456789