bluez/bluez #903: bluez removes pairing data for samsung buds on shutdown

LuanVSO, 2024-07-18:
device is more specifically a samsung buds 2 pro
steps:
1. pair device with LE support enabled on bluez
2. restart bluetooth.service
3. notice logs:
```
Failed to load IRKs for hci0: Invalid Parameters (0x0d)
src/battery.c:provided_battery_added_cb() Ignoring non-existent device path for battery /org/bluez/hci0/dev_40_35_E6_06_A1_B8
```
4. mask bluetooth.service
5. stop bluetooth.service
6. at this point the folder for the device will no longer exist on /var/lib/bluetooth forcing a repairing on the next time you want to use the device.

[btmon.log](https://github.com/user-attachments/files/16286836/btmon.log)
[bluetoothd-d.log](https://github.com/user-attachments/files/16286809/bluetoothd-d.log)
thanks for the help

CEOofFrance, 2024-08-11:
I have this exact same issue with the Galaxy Buds 2 Pro.

principis, 2024-08-29:
This seems to be related to (some?) Intel chipsets. I can reproduce it on the AX210 and know it is an issue on the AX200. I cannot reproduce it on the AMD RZ616/MT7922. If needed, I can test on others.

I have tried the following bluez versions, all of which have the issue:

- 5.77
- 5.75
- 5.73
- 5.68

I would be happy to further look into this, but I don't know where to start.

LuanVSO, 2024-08-29:
can confirm my Bluetooth card is also a intel ax210

SergSel2006, 2024-09-18:
I have similar issue on Intel AX101 with bluez 5.77,
I have two LE devices: My keyboard and a mouse. They both end up failing to work after reboot. 
I have these two errors in my log in particular:
```
Failed to load LTKs for hci0: Invalid Parameters (0x0d)
Failed to load IRKs for hci0: Invalid Parameters (0x0d)
```

SergSel2006, 2024-09-21:
I have managed to find workaround for this problem: I set `UserspaceHID` to `false` and it did the trick. So the problem should be in the userspace HID driver?

Insane-Owl, 2024-09-21:
> I have managed to find workaround for this problem: I set `UserspaceHID` to `false` and it did the trick. So the problem should be in the userspace HID driver?

Interesting, I tried this and it didn't seem to work however.

Insane-Owl, 2024-09-21:
I also noticed something that _may_ be related; the buds 2 pro show up twice in the pairing screen.

charliefrance, 2024-09-21:
Me too @Insane-Owl. One entry for audio device, one is an unknown device. I think this is specific to Buds 2 Pro, and that's why @SergSel2006's fix didn't work for you.

Green-D-683, 2024-11-25:
Also having this problem, ax201 Bluetooth card

joshthegraphitemechanic, 2025-02-03:
Having the same issue, unpairs on both shutdown and when using the earbuds with any other device.

SuperSandro2000, 2025-02-16:
> I also noticed something that _may_ be related; the buds 2 pro show up twice in the pairing screen.

They show up once as an headset and once as serial something where you can update the firmware and change settings.

SuperSandro2000, 2025-05-07:
I am going to post some spam to fight the stale bot.

charliefrance, 2025-05-10:
Yeah this is super annoying, really wish someone woukd check it out

charliefrance, 2025-10-07:
Most certainly not stale

LuanVSO, 2025-11-07:
just tested this again on fedora 43 and the issue seems to be fixed, could someone else also retest?
versions:
```
bluez-libs-5.84-2
bluez-5.84-2
bluez-obexd-5.84-2
bluez-cups-5.84-2
NetworkManager-bluetooth-1.54.0-2

kernel-6.17.7-300
```

charliefrance, 2025-11-07:
I think this is also fixed for me, on Arch Linux

SuperSandro2000, 2025-11-07:
NixOS does not have 5.84, yet, because no github release exists for it, only a tag...

LuanVSO, 2025-11-07:
> NixOS does not have 5.84, yet, because no github release exists for it, only a tag...

but is it still broken on whichever version it has now?

SuperSandro2000, 2025-11-07:
We are on NixOS on 5.83 right now and I think when I last tested it a week ago, it was still broken.