bluez/bluez #908: Logitech MX Master 3 cannot load driver logiops to function properly, after update from 5.69 to 5.77

lazarosgogos, 2024-07-22:
I was using bluez-5.69 under Fedora 39 mainly due to my MX Master 3 not working with the newer **bluez** versions.

The mouse worked fine, its driver loaded appropriately and everything was smooth.

Having updated to Fedora 40, under which I cannot downgrade to bluez-5.69 and stuck with the lowest available version being 5.73, my system cannot load the driver for the MX Master 3, logiops.

I had once tried to update, under Fedora 39 from bluez-5.69 to bluez-5.73, but that broke my mouse from working, so I reverted back to bluez-5.69

Any idea what's going on? The only difference was the update of this package, if I am not mistaken. Other than that it worked, and it should work fine.

I am certain this problem affects others as well and I am here for further clarifications,

Thank you.

lazarosgogos, 2024-07-22:
Seems related to #778

c-kr, 2024-07-22:
See also https://github.com/PixlOne/logiops/issues/468 - not sure if the bug comes from logiops or bluez and the end

c-kr, 2024-07-22:
Probably also related to #899 - i am also unable to create a btmon dump (got the same 'buffer overflow detected' message)

c-kr, 2024-07-25:
I was able to create a dump with btmon directly on tty while the bug occurred. If a dev needs it i can send it by mail (dont want to publish public as it is big and i dont know which data it contains)

lazarosgogos, 2024-07-25:
> I was able to create a dump with btmon directly on tty while the bug occurred. If a dev needs it i can send it by mail (dont want to publish public as it is big and i dont know which data it contains)

Can you share the log with me, or share the commands you used to reproduce it?

To reproduce it, as far as I know, you can turn the mouse off and then on again. So an output dump would be valueable I think, to find what part fails exactly.

Or send me an email at lazarosg@csd.auth.gr

c-kr, 2024-07-25:
@lazarosgogos i used the capture command from the wiki https://github.com/bluez/bluez/wiki/btmon overnight just before leaving the computer and stopped capturing next morning just after i checked that the bug occurred.

ValdikSS, 2024-09-11:
Try the following patch:
```
--- bluez-5.78/profiles/input/hog-lib.c        2024-09-11 02:40:28.913942567 +0300
+++ bluez-5.78/profiles/input/hog-lib.c 2024-09-11 02:41:59.017178257 +0300
@@ -834,6 +834,7 @@
                return;
 
        bt_uhid_unregister_all(hog->uhid);
+       force = true;
 
        err = bt_uhid_destroy(hog->uhid, force);
        if (err < 0) {
```

If it works, it's #949 / #933

c-kr, 2024-09-17:
Updating to bluez 5.78 solved the issue for me

lazarosgogos, 2024-09-17:
> Updating to bluez 5.78 solved the issue for me

Same here. Fedora 40 GNOME 46. 5.78 seems to have solved it

closed 2024-09-17 by lazarosgogos