bluez/bluez #911: Wiimote Bluetooth pairing does not work above 5.72

MetalfaceScout, 2024-07-23:
I've been having trouble pairing my Wiimotes, and I downgraded past 5.73 and found that below that version my Wiimotes pair properly. They still connect above 5.73 but don't pair. The Wiimote will flash until the pair request happens then it turns off. I can provide any logs or information needed.

teinturman, 2024-07-24:
Hi, i have exactly same issue with my Wiiu Pro Controller device  : 
these devices are really great and cheap and it is a shame that they no longer work with the new versions : 

If i do not pair and i connect : it works, without prompting for a code pin. 
As soon as i pair my device, the connect function does not work anymore, and i am prompted for a code pin ( which always fails, as the wiiu pro controller does not expect any pincode... )  

This was not the case in the previous versions. ( was working well in raspberry pi bullseye, but no longer works in raspberry pi bookworms). 

Could it be that in the dev_connect function, there is the below code that enforce medium security if device is bonded ? 

profiles/device.c
 	/* encrypt connection if device is bonded */
	if (input_device_bonded(idev))
		sec_level = BT_IO_SEC_MEDIUM;
	else
		sec_level = BT_IO_SEC_LOW;

if  any guru bluetooth could help us understand what is the exact change that occured that made this no longer work, and ideally have a setting in the conf file to allow previous behaviour for legacy devices...  thanks a lot for your help ...

Vudentz, 2024-07-24:
There has been some changes to move to uHID by default:

https://github.com/bluez/bluez/blob/master/profiles/input/input.conf#L14

Try changing that back to false to see if that helps, also check with 5.77 if you still can reproduced it since 5.73.

MetalfaceScout, 2024-07-24:
> There has been some changes to move to uHID by default:
> 
> https://github.com/bluez/bluez/blob/master/profiles/input/input.conf#L14
> 
> Try changing that back to false to see if that helps, also check with 5.77 if you still can reproduced it since 5.73.

This didn't work running bluez 5.77-1 on my Arch system. I uncommented the line and set it to false, like so:
```
# Enable HID protocol handling in userspace input profile
# Defaults to true (Use UHID instead of kernel HIDP)
UserspaceHID=false
```
The same issue occurs, the Wiimote fails to pair.

Vudentz, 2024-07-24:
@MetalfaceScout collect the logs with btmon -w <finame>, you may also try with bluetoothctl -a auto to see if that helps.

MetalfaceScout, 2024-07-24:
Here's a log of the failed pair,
[output.txt](https://github.com/user-attachments/files/16367831/output.txt)
as well as the output of the btmon command.
[btmonoutput.zip](https://github.com/user-attachments/files/16367837/btmonoutput.zip)

Authentication happens at line 343 and pairing is initiated at line 274.

MetalfaceScout, 2024-07-25:
I did some more investigation, I think this has something to do with PIN code generation to pair with the Wiimote.
I read [this information on WiiBrew](https://www.wiibrew.org/wiki/Wiimote#Bluetooth_Pairing) about the Wiimote and it needs a pin generated to match the Wiimote to pair with it. I logged a request from bluez 2.72 and it looks like this:
```
bluetoothd[1868041]: @ MGMT Command: PIN Code Reply (0x0016) plen 24                {0x0001} [hci0] 15.891165
BR/EDR Address: 00:16:95:01:55:B0 (AVC Technology (International) Limited)
PIN length: 6
PIN code[16]: 89ca64ef14bc00000000000000000000
< HCI Command: PIN Code Request Reply (0x01|0x000d) plen 23                              #40 [hci0] 15.891171
Address: 00:16:95:01:55:B0 (AVC Technology (International) Limited)
PIN length: 6
PIN code: ��d��
```

Whereas the log from bluez 2.77 looks like this:
```
bluetoothd[1594454]: @ MGMT Command: PIN Code Reply (0x0016) plen 24                {0x0001} [hci0] 18.143497
BR/EDR Address: 00:25:A0:C9:57:D3 (Nintendo Co., Ltd.)
PIN length: 4
PIN code[16]: 30303030000000000000000000000000
< HCI Command: PIN Code Request Reply (0x01|0x000d) plen 23                              #37 [hci0] 18.143503
Address: 00:25:A0:C9:57:D3 (Nintendo Co., Ltd.)
PIN length: 4
PIN code: 0000
```

The pin code generated by bluez 2.77 is just 0000, and the pin code from 2.72 isn't.

As a sidenote, the addresses are different because one of the Wiimotes I used is different. The result is the same otherwise.

teinturman, 2024-07-26:
hi all , i did some more investigations on my side as well : i am in version 5.66    
i tried to upgrade to later version but it is not available in the Raspberry pi aptget space yet. not sure how to force the upgrade...

What i discovered with btmon is that when i am prompted for a code, and i tried to enter the macadress  characters in reverse order ( mac = 18:2A:7B:3C:FF:4F ; code = Oÿ<{*↑ ) 

The btmon showed that what was sent to the controller was the code in UTF8, whereas , the controller is expecting unicode encoding.
not sure if this is only related to version 5.66 , 

@MetalfaceScout  , do you confirm that in your 5.72 environment, it is indeed the mac adress of the controller,  in reverse order that is sent to the controller? 

@ MGMT Command: PIN Code Reply (0x0016) plen 24      {0x0001} [hci0] 546.733327
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        PIN length: 9
        PIN code: e286912a7b3cc3bf4f00000000000000
< HCI Command: PIN Code Request... (0x01|0x000d) plen 23  #68 [hci0] 546.733477
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        PIN length: 9
        PIN code: ↑*{<ÿO

teinturman, 2024-07-28:
i spent again uite a few time on this issue : and did  some more investigations : i have now installed version 5.77 : and i believe that i successfully paired the device, but once paired, i am unable to connect. in my setup : 

Mac adress of the raspberry pi bluetooth controller : cd:58:b2:dd:3a:d8 / mac of the wiiu pro controller gamepad : 18:2A:7B:3C:FF:4F
This is important because the pincode that is expected with WIIU is ( i do not know exactly yet ) is one of these macaddress, either in forward order or in reverse order. 

i did a trace, which i analyzed as much as i could with my knowlefdge in excel : 
[logs_btmon.xlsx](https://github.com/user-attachments/files/16402830/logs_btmon.xlsx)

what i understand is that during the pairing, during the Auth Request process :  first the device tries with "0000" ==> it fails  but then, it retries and sends the correct password ( macaddress of Raspberry pi in forward order ) ==> success..

Then , the pairing is a success and a link key is exchanged. 

This is not enough to have the gamepad to be able to reconnect automatically, as for the wiiu to reconnect automatically ,  needs to have the 3 steps ( pair , connect and trust ) completed... 

Once the device is paired, if i launch the command "Connect 18:2A:7B:3C:FF:4F" , then there is again 2 attempts 
-first attempt : 
> HCI Event: PIN Code Request (0x16) plen 6               #20 [hci0] 275.634082
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
@ MGMT Event: PIN Code Request (0x000e) plen 8       {0x0002} [hci0] 275.634091
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        Secure PIN: 0x00
@ MGMT Event: PIN Code Request (0x000e) plen 8       {0x0001} [hci0] 275.634091
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        Secure PIN: 0x00
bluetoothd[33577]: = profiles/input/device.c:control_connect_cb(..   315.418220
bluetoothd[33577]: = src/agent.c:pincode_reply() Agent /org/blue..   335.655418
bluetoothd[33577]: @ MGMT Command.. (0x0017) plen 7  {0x0001} [hci0] 335.655429
then there is a response (PN Code NEgATIVE Response)  and authentication fails

- second attempt : 
i get prompted for the pin code interactively. 
when i type the password ( mac address of raspberry pi in forward order, ) 

the raspberry pi sends what i typed, but in UTF8 format this is leading to mismatch with the Wiiu which is expecting simply the mac address in standard Hexa code... 
therefore, the connect fails. 

==> there are potentially 2 issues there : 

1) Raspberry pi should have been autonomous to send the pin code to the wiiu, like it did during the paired mode, without prompting me for a password. 
2) when i get prompted for the code pin, and i enter the code pin, i believe what is sent to the Wiiu should not be encoded in utf8 format. 

could someone else reconfirm that my understanding is correct? 

and would someone knows where in the code, the behaviour to decide what pin code should be sent  and how it is sent is written ?

thank you all for your help in resolving this topic... 

Below is the trace for the second attempt of the connect process ( where we can see that the raspbeery pi sends : 
c3983ac39dc2b258c38d   to the wiiu ( 10 characters ) which is the utf-8 version of what i enterered : D83ADDB258CD
i am pretty sure the WIIU is expecting only 6 characters, in hexa format.

< HCI Command: Authentication Req.. (0x01|0x0011) plen 2  #40 [hci0] 576.925123
        Handle: 12 Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
> HCI Event: Command Status (0x0f) plen 4                 #41 [hci0] 576.925543
      Authentication Requested (0x01|0x0011) ncmd 1
        Status: Success (0x00)
> HCI Event: Link Key Request (0x17) plen 6               #42 [hci0] 576.925802
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
< HCI Command: Link Key Request... (0x01|0x000b) plen 22  #43 [hci0] 576.925812
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        Link key[16]: 001e50ed9d6b3b7ff8cc210a638f3929
> HCI Event: Command Complete (0x0e) plen 10              #44 [hci0] 576.927082
      Link Key Request Reply (0x01|0x000b) ncmd 1
        Status: Success (0x00)
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
> HCI Event: PIN Code Request (0x16) plen 6               #45 [hci0] 576.937609
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
@ MGMT Event: PIN Code Request (0x000e) plen 8       {0x0002} [hci0] 576.937676
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        Secure PIN: 0x00
@ MGMT Event: PIN Code Request (0x000e) plen 8       {0x0001} [hci0] 576.937676
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        Secure PIN: 0x00
bluetoothd[33577]: @ MGMT Comman.. (0x0016) plen 24  {0x0001} [hci0] 579.485616
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        PIN length: 10
        PIN code[16]: c3983ac39dc2b258c38d000000000000
< HCI Command: PIN Code Request... (0x01|0x000d) plen 23  #46 [hci0] 579.485639
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
        PIN length: 10
        PIN code: Ø:Ý²XÍ
> HCI Event: Command Complete (0x0e) plen 10              #47 [hci0] 579.487368
      PIN Code Request Reply (0x01|0x000d) ncmd 1
        Status: Success (0x00)
        Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
@ MGMT Event: Command Complete (0x0001) plen 10      {0x0001} [hci0] 579.487382
      PIN Code Reply (0x0016) plen 7
        Status: Success (0x00)
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
> HCI Event: Auth Complete (0x06) plen 3                  #48 [hci0] 579.537727
        Status: Authentication Failure (0x05)
        Handle: 12 Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.)
@ MGMT Event: Authentication Failed (0x0011) plen 8  {0x0002} [hci0] 579.537757
        BR/EDR Address: 18:2A:7B:3C:FF:4F (Nintendo Co., Ltd.

AlphaCraft9658, 2024-08-12:
This issue seems to also apply to Wii U Pro Controllers.

Vudentz, 2024-08-12:
Hmm, looks like there is a dedicated plugin for handling Wii controllers:

https://github.com/luetzel/bluez/blob/master/plugins/wiimote.c

But that hasn't received changes in ages, anyway perhaps it worth checking if there is a change with respect to not loading or loading it in the distro package, anyway if that is actually in use there should be a line like the following in the journalctl:

"Forcing fixed pin on detected wiimote XX:XX:XX:XX:XX:XX"

mikezila, 2024-08-29:
I'm also having issues pairing Wii U Pro controllers. Some discussion and logs are available on #800 from myself and others.

ValdikSS, 2024-09-11:
Try the following patch:
```
--- bluez-5.78/profiles/input/hog-lib.c        2024-09-11 02:40:28.913942567 +0300
+++ bluez-5.78/profiles/input/hog-lib.c 2024-09-11 02:41:59.017178257 +0300
@@ -834,6 +834,7 @@
                return;
 
        bt_uhid_unregister_all(hog->uhid);
+       force = true;
 
        err = bt_uhid_destroy(hog->uhid, force);
        if (err < 0) {
```

If it works, it's #949 / #933

John-Gee, 2024-12-23:
> Try the following patch:
>
> ```
> --- bluez-5.78/profiles/input/hog-lib.c        2024-09-11 02:40:28.913942567 +0300
> +++ bluez-5.78/profiles/input/hog-lib.c 2024-09-11 02:41:59.017178257 +0300
> @@ -834,6 [...]

That patch did not work for me with the Wii U Pro connection issue.

Chris660, 2025-01-04:
I have the same problem as the original reporter, in that I can no longer pair (genuine) wiimote controllers on more recent bluez releases.

> I've been having trouble pairing my Wiimotes, and I downgraded past 5.73 and found that below that version my Wiimotes pair properly. They still connect above 5.73 but don't pair.

I bisected the changes between 5.72 and 5.73, and identified the following as the commit that "broke" things for me:

```
commit f174724c76c6c8b91276f444142fed9c2537e5c1
Author: Emil Velikov <[emil.velikov@collabora.com](mailto:emil.velikov@collabora.com)>
Date:   Mon Jan 29 14:44:21 2024 +0000

bluetoothd: change plugin loading alike obexd
```

I think what's happening is that since these changes, the "autopair" plugin PIN callback runs *before* the "wiimote" one - and this is why you saw the PIN of `0000` being sent on 5.77.  See [autopair_pin_cb](https://github.com/bluez/bluez/blob/dfb1ffdc95a00bc06d81a75c11ab5ad2e24d37bf/plugins/autopair.c#L76).

Disabling the  `autopair` plugin with `bluetoothd -P autopair` on 5.79 allows the wiimotes to pair again:

```
$ bluetoothctl --version
bluetoothctl: 5.79

$ bluetoothctl 
[Nintendo RVL-CNT-01]# Agent registered
[Nintendo RVL-CNT-01]# info
Device A4:C0:E1:36:B5:1F (public)
	Name: Nintendo RVL-CNT-01
	Alias: Nintendo RVL-CNT-01
	Class: 0x00002504 (9476)
	Icon: input-gaming
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: no
	LegacyPairing: yes
	UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v057Ep0306d8614
```

I'm totally unfamiliar with bluez, so I'm not sure what the best way to allow these plugins to coexist is.  I'm very happy to test changes if that's of any use.

MetalfaceScout, 2025-01-05:
> I think what's happening is that since these changes, the "autopair" plugin PIN callback runs _before_ the "wiimote" one - and this is why you saw the PIN of `0000` being sent on 5.77. See [...]

Good find, can confirm this does work. Might fix the Wii U Pro stuff as well because the plugin handles those to my knowledge.

I looked at the codebase, looks like magic to me. I can't find a function or a way that bluez decides which callback to use. All I found was ```BLUETOOTH_PLUGIN_DEFINE``` which passes a priority when registering a plugin.

Chris660, 2025-01-05:
There are three plugin priorities defined: `BLUETOOTH_PLUGIN_PRIORITY_LOW`, `*_DEFAULT` and `*_HIGH`.

`BLUETOOTH_PLUGIN_DEFINE` adds the plugins to a list, sorted by (descending) priority:

```c
   95 static void add_plugin(const struct bluetooth_plugin_desc *desc)
   96 {
    // ...
  112     plugins = g_slist_insert_sorted(plugins, plugin, compare_priority);
  113     DBG("Plugin %s loaded", desc->name);
```

The wiimote plugin is currently `BLUETOOTH_PLUGIN_PRIORITY_LOW` and autopair `BLUETOOTH_PLUGIN_PRIORITY_DEFAULT`.

If we bumped the wiimote plugin up to `BLUETOOTH_PLUGIN_PRIORITY_DEFAULT`, on par with autopair, then IIUC their relative position will depend on the order of static initialisation.  The wiimote plugin would need to be `BLUETOOTH_PLUGIN_PRIORITY_HIGH` to guarantee priority over autopair...

TBH, I think the cleanest solution would be to fold the wiimote plugin code into autopair rather than trying to solve the issue with plugin priorities (less obvious, more fragile).  I'll try this, and report back if it works.

John-Gee, 2025-01-07:
> I have the same problem as the original reporter. I can no longer pair (genuine) wiimote controllers on more recent bluez releases.
> 
> > I've been having trouble pairing my Wiimotes, and I downgraded past 5.73 and found that below that version my Wiimotes pair properly. They still connect above 5.73 but don't pair.
> 
> I bisected the changes between 5.72 and 5.73, and identified the following as the commit that "broke" things for me:
> 
> ```

Your findings are similar to mine in https://github.com/bluez/bluez/issues/800#issuecomment-2563794673

My understanding is that priorities aren't used anymore since the _bad_ commit, and that's the issue.

Vudentz, 2025-01-07:
@Chris660 can you try with the above changes that merge the handling of wii controllers into autopair plugin?

Chris660, 2025-01-07:
@Vudentz  - I tested the changes on that branch, and can confirm they work for me (tested wiimotes + balance board).

FWIW, on my branch I added the wiimote test in the existing Peripheral branch of the nested switch in `autopair_pincb`.  I was trying to reuse the already extracted vendor product and name data: https://github.com/Chris660/bluez/commit/7bd03fa25b098e6e111f4632c5294c4a64bb8a4b

HurricanePootis, 2025-05-20:
I just compiled bluez with git master, `5.82.r127.g0402c0674`, and I am still unable to pair a Wii U Pro Controller to my device. It says pairing successful but not connected.

John-Gee, 2025-05-20:
> I just compiled bluez with git master, `5.82.r127.g0402c0674`, and I am still unable to pair a Wii U Pro Controller to my device. It says pairing successful but not connected.

I had issues yesterday connecting my Wii U Pro controller to a new install using Plasma, it kept asking for a pin to connect.

This is what worked for me, from memory (not sure which step was necessary): 
- Log out from Plasma
- Kill bluedevil 
- Restart the Bluetooth service
- In bluetoothctl do:
  * power on
  * agent off
  * agent NoInputNoOutput
  * default-agent
  * remove MAC (if necessary)
  * scan on
  * press reset on the controller 
  * scan off, when the controller is found
  * pair MAC
  * trust MAC
  * connect MAC
  
The whole thing in one go, not allowing the controller to turn before connecting to it.

I couldn't get it to work with the old .71 stack but it worked with the .82.

This issue was about pairing though so unlikely to be yours now.

HurricanePootis, 2025-05-20:
Whenever I do that, it still asks for a pin code?