bluez/bluez #912: BLE Development - Connection Unreliability

WNacci, 2024-07-25:
I am working on developing BLE peripherals to stream sensor data but have been facing consistent issues with BLE reliability on Linux/Bluez, whereas my windows machine performs flawlessly in comparison. Additionally - the Linux device performs properly with most peripherals once paired.

The issue manifests in disconnects from the device soon after initial connection as shown in btmon (within a second or two), with a CO_ERROR_DIFF_TRANSACTION_COLLISION 0x2A being observed on the peripheral. Upon successful pairing, which seems to happen upon a retry or two, a second error of a 0x08 timeout error occurs between the devices:

btmon:
```
Reason: Invalid LMP Parameters / Invalid LL Parameters (0x1e)
Upon disconnect: 
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                 #12271 [hci0] 1556.503035  
       Status: Success (0x00)  
       Handle: 2048 Address: 48:23:35:F4:00:06 (Dialog Semiconductor Hellas SA)  
       Reason: Connection Timeout (0x08)  
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                            {0x0001} [hci0] 1556.503105  
       LE Address: 48:23:35:F4:00:06 (Dialog Semiconductor Hellas SA)  
       Reason: Connection timeout (0x01)

Reconnections:
       Reason: Connection Failed to be Established (0x3e)
```
       
I realize that my BLE implementation on the peripheral may have problems that are causing this behavior, but what I'm wondering is how I can get my linux machine (running Bluez) to perform as robustly as the windows BLE stack, and what might be the cause for such a difference between the two systems. As the device connects consistently & reliably to the windows machine and my iPhone, I'm wondering what is causing these issues in Bluez in comparison. Additionally, my BLE implementation is basically exactly the same as the manufacturer's (Renesas) recommendation. Are there settings that can be changed to improve Bluez performance in BLE? I have tried altering connection intervals and timeouts to larger or smaller values but neither have seemed to affect this behavior; I believe the values are being renegotiated upon connection.

Vudentz, 2024-07-25:
@WNacci sounds like a Link-Layer problem (controller), do you have the same controller on windows as well? Except if windows is using vendor commands the HCI commands should look the same and obviously we can't go around the hardware.

About the claims of not having robust software and then throwing windows into the mix sounds like a joke after the bsod episode with cloudstrike, no software is imune to bugs, so next time please come with something more concrete and if there is something we could do better we will attempt to do it.

WNacci, 2024-07-25:
@Vudentz I’ll do some further investigation and try to come back to you with some more info focused on the Link-Layer. Do you have any requests for information other than more thorough btmon readouts?

Also, I’ve edited my initial comment- did not mean to suggest that Bluez is not robust, meant to emphasize that the behavior in this particular (non-user-facing) context has seemed inconsistent and strange relative to other stacks.

I believe that I have tested with the same controller before but I’ll re-run some tests with better logging soon to verify and help narrow things down. Thank you for your quick response, I’ll hopefully have some more info to share in the near future.

closed 2025-07-21 by github-actions[bot]