bluez/bluez #913: Disable settings such as cis-central cis-peripheral iso-broadcaster sync-receiver

maniacx, 2024-07-25:
On **Fedora** 40 using **Mediatek bluetooth adapte**r, the `btmgmt info` reports the following settings are supported and enabled
```
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech cis-central cis-peripheral 
	current settings: powered ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral iso-broadcaster sync-receiver 
```

Is there a way to disable `cis-central`, `cis-peripheral, `iso-broadcaster` , `sync-receiver` using bluetoothctl or any other method?

Using Gnome extension `Airpod Battery Monitor` works on **Debian**, but **not** on **Fedora** 40 for some reason. `Airpod Battery Monitor`scans the nearest BLE beacons of the connected Airpods in discovery mode, as it containg Manufacturers Data which contains Battery Levels.

Only different I see in Fedora is that it is using newer kernel and newer bluez version and supports these additional 4 settings.`cis-central cis-peripheral iso-broadcaster sync-receiver`. I want to test the `Airpod Battery Monitor` by disabling these setting, just to check if these new settings could be the reason why Airpods BLE signal aren't detected when connected.

Some details.

Fedora
```
kernel version : 6.9.9-200.fc40.x86_64
bluez version : 5.77
btmgmt info :
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech cis-central cis-peripheral 
	current settings: powered ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral iso-broadcaster sync-receiver 
```

Debian
```
kernel version : 6.1.0-22-amd64
bluez version : 5.66
btmgmt info :
 	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech 
	current settings: powered ssp br/edr le secure-conn 
```

Is there a way to disable `cis-central`, `cis-peripheral, `iso-broadcaster` , `sync-receiver` using bluetoothctl or any other method?

Vudentz, 2024-07-26:
Weird, they shouldn't be enabled by default, perhaps kernel ISO sockets experimental is enabled? Anyway I don't think that has anything to do with Airpod Battery Monitor, that usually is a feature done over AT vendor commands, so these flags don't really matter with airpods.

maniacx, 2024-07-26:
Hello Vudentz,

Airpod Battery Monitor does NOT use AT vendor commands (unlike other bransds earphones, Sony, Samsung they do use   AT vendor commands over RFCOMM)

Instead it scans for nearbly le beacon (random address), (similar to how all android Airpod battery level apps work such as CAPOD).
It just scan for a random address that broadcast LE beacon, gets the manufacturing data, verifies on manufacturing data is an Airpods and assumes the nearest Airpod is the connected Airpods and gets battery levels from the manufacturing data.

In any case, is there a way to disable those settings (is-central, cis-peripheral, iso-broadcaster,sync-receiver) , by bluetoothctl command or editing /etc/bluetooth/main.conf or some other way? Just for testing.

Vudentz, 2024-07-26:
> Hello Vudentz,
>
> Airpod Battery Monitor does NOT use AT vendor commands (unlike other bransds earphones, Sony, Samsung they do use AT vendor commands over RFCOMM)
>
> Instead it scans for nearbly le [...]

Good to know, though that part of it assuming it is the nearest Airpod got me thinking how easy would it be to mess it up by just advertising the same data, anyway.

> In any case, is there a way to disable those settings (cis-central, cis-peripheral, iso-broadcaster,sync-receiver) , by bluetoothctl command or editing /etc/bluetooth/main.conf or some other way? Just for testing.

In theory they shall only be set if ISO socket support has been enabled, but perhaps there is a bug somewhere that sets it anyway.

closed 2025-07-21 by github-actions[bot]