bluez/bluez #914: Concurrent BLE Peripheral and Central

claplante88, 2024-07-29:
I have been trying to setup my BT Controller (CY20704 based) so it is Central to one of my sensors and a peripheral which I can connect via mobile app. I have configured /etc/bluetooth/main.conf with Experimental-true.

If I start with a peripheral device first via the following commands using bluetoothctl:

power on
     discoverable on
     pairable on
     agent on
     default-agent
     advertise on
I can see my device advertising and I can connect to it. It won't scan once anything peripheral is initialized. I get the error:
Failed to start discovery: org.bluez.Error.InProgress

If I start as a Central first and issue command:
scan on
I can connect to the sensor of interest:

With the controller connected as a Central once I type 
discoverable on
the connection is dropped, I can't scan or do anything central related anymore.

At one point I was at least able to get connections as central and peripheral concurrently but once I dropped the Central connection I could not reconnect again until dropping the peripheral connection. This was my initial issue I was fighting and was going to submit here with btmon logs. I can't even seem to get that working now doing the same sequences. The overall issue is common though with central not working when peripheral is either connected or already advertising.

Any suggestions would be greatly appreciated.

Refer to attached logs.

Thanks
Chris
[btmon_log.txt](https://github.com/user-attachments/files/16419339/btmon_log.txt)

claplante88, 2024-07-31:
I should have added this was using BlueZ v5.64 with Ubuntu 22.04.4 LTS. I had a colleague also try this using a different controller that was BT/WiFi device and he was using a newer version of BlueZ v5.66 with Linux kernel 6.1.36 and he saw the same issues. He could connect both Central and peripheral BLE concurrently but once disconnecting the Central from a remote peripheral he was not able to connect again until disconnecting the peripheral side.

jamsliu2009, 2024-08-02:
Hi claplante88,

I have also observed same issue several month agon when I tried to setup a multi-role topology. This symptom is that if one link is created as peripheral role, initiating another connection results in a failure.

Watching the HCI log, BlueZ did not create connection after getting an extended advertising report from a passive scan. After tracing the kernel code, besides checking controller's response of "LE Read Supported States (0x08|0x001c)", BlueZ also checks the HCI device "HCI_QUIRK_VALID_LE_STATES" bit. Only devices that support both could pass the check and enable multi-role behavior.

The "HCI_QUIRK_VALID_LE_STATES" bit is set in Bluetooth HCI device driver (kernel), it seems some vendors have made this feature built-in in the kernel, e.g. Intel, MTK, RTK and Qualcomm. 

As the comment written in hci.h, 
	/* When this quirk is set, the controller has validated that
	 * LE states reported through the HCI_LE_READ_SUPPORTED_STATES are
	 * valid.  This mechanism is necessary as many controllers have
	 * been seen has having trouble initiating a connectable
	 * advertisement despite the state combination being reported as
	 * supported.
	 */
	HCI_QUIRK_VALID_LE_STATES,

I have try modify set_le_simultaneous_roles_func( ) function in kernel as following.
`static int set_le_simultaneous_roles_func(struct sock *sk, struct hci_dev *hdev,
                      struct mgmt_cp_set_exp_feature *cp,
                      u16 data_len)

{
...
    set_bit(HCI_QUIRK_VALID_LE_STATES, &hdev->quirks);
    printk("set HCI_QUIRK_VALID_LE_STATES (%lx->%d)\n", hdev->quirks, changed);
...
}`

And then setting MGMT_OP_SET_EXP_FEATURE via Mgmt API to set current HCI device's quirk flag HCI_LE_SIMULTANEOUS_ROLES on. In such that I confirm multi-roles is workable.

But this is just to confirm the BlueZ's behavior. I do not consider adding this hack in my project due to the risk. So, if your Bluetooth SOC does not have a driver in Linux kernel to support multi-role, multi-role is not doable.

claplante88, 2024-08-02:
> EXTERNAL EMAIL: Be careful with attachments and links.

Thank you for the detailed information of your investigation. The BT Controller I am using is the CYW20704 from Infineon/Cypress. I need to look into if it has a driver in Linux kernel to support multi-role or not as it seems that is the key from what you learned.

Would you be able to share which BT SOC you were seeing this issue on? I had a colleague try this on a dual mode WiFi/BT device which I believe was NXP based and he also saw the same behavior.

Chris

THE INFORMATION CONTAINED IN THIS DOCUMENT IS OF A PROPRIETARY NATURE AND IS INTENDED TO BE KEPT CONFIDENTIAL BETWEEN THE SENDER AND THE INTENDED RECIPIENT. IT MAY NOT BE REPRODUCED OR USED WITHOUT EXPRESS WRITTEN PERMISSION OF EZURIO

abqjln, 2024-09-24:
Chiming in with exactly the same issue. Central can connect to client, and then multiple peripherals, but if client disconnects it cannot reconnect, regardless of the state of peripheral connections. Same behavior on:

bluetoothd -v is 5.66

rpi5 arm64 Debian 12, Linux 6.6.47+rpt-rpi-2712
and
rpi4 arm64 Debian 12, Linux 6.6.31+rpt-rpi-v8.

Not a kernel guy (for now) so I went to a second adapter, where everything works fine.

avanmalleghem, 2024-12-27:
Has someone solved with this concurrency issue ?

> Not a kernel guy (for now) so I went to a second adapter, where everything works fine.

@abqjln can you give me the reference to your adapter ?

abqjln, 2024-12-27:
Adapter is an Edimax BT-8500.

abqjln, 2025-01-15:
Inspired by @jamsliu2009 debug hack, and this https://github.com/JaapvanEkris/openrowingmonitor/issues/69, I found I could get simultaneous central and server working on a single adapter _if_ I hard coded which device(s) were to be treated as centrals (I used two different test centrals) and the rest as peripherals. I had to use device names because the addresses are private random resolvable.

On r(e)connection my code would get confused as to the roles of the (re)connecting devices because both were on the same adapter. I think that's why to order of disconnect/connect mattered and made it inconsistent. In my physical environment, there is much noise so I expect disconnections, just need to reconnect quickly.

My hope lies in detecting the "initiator" state that determines central or peripheral at the moment of connection. It exists at the Link Layer (Core Spec 6(B)) upon connection. The state is in the daemon in btd_device_is_initiator(device) in src/device.c for LE.  But I cannot find where it is exposed to dbus.

My preferred solution would be a new device property "Initiator" that I could access the same way I access "Connected" or "Paired" or "Bonded" using doc/org.bluez.Device.rst.

Before I dive deeper or go requesting things, does this help anyone? Or even make sense.

closed 2025-07-21 by github-actions[bot]