bluez/bluez #915: Does any encounter problems about Data Buffer Overflow

jamsliu2009, 2024-07-31:
I am developing a BLE application in a multi-link setup. I have observed that a few HCI ACL packets are occasionally missed.

I accumulate: (a) the number of completed counts in "HCI Event: Number of Completed Packets (0x13)" and (b) the number of "ACL Data TX" in the HCI log captured via a Bluetooth sniffer.

Before the "HCI Event: Data Buffer Overflow (0x1a)" is received from the BlueZ Host, (b) - (a) is larger than the controller's buffer size, which is obtained from the attach procedure, as shown below. 

< HCI Command: LE Read Buffer Size (0x08|0x0002) plen 0     #7 [hci0] 32.122530
> HCI Event: Command Complete (0x0e) plen 7                 #8 [hci0] 32.126803
      LE Read Buffer Size (0x08|0x0002) ncmd 1
        Status: Success (0x00)
        Data packet length: 340
        Num data packets: 12

It seems BlueZ Host sending ACL data overflowed the Tx buffer in controller. Does anyone encounter same problem?

jamsliu2009, 2024-08-02:
In hci_core.c hci_sched_le(), "cnt" is assigned value from "hdev->le_cnt", while the "cnt" variable is declared as an int but  "hdev->le_cnt" is declared as an unsigned int. When "cnt" is underflow, the outer while loop while(cnt && hci_chan_sent()) will be also true if there are packets to send in the socket, things get worse. Why does "cnt" underflow? Because the inner while loop does not check "cnt" variable.

====

static void hci_sched_le(struct hci_dev *hdev)
{
	struct hci_chan *chan;
	struct sk_buff *skb;
	int quote, cnt, tmp;

	BT_DBG("%s", hdev->name);

	if (!hci_conn_num(hdev, LE_LINK))
		return;

	cnt = hdev->le_pkts ? hdev->le_cnt : hdev->acl_cnt;

	__check_timeout(hdev, cnt, LE_LINK);

	tmp = cnt;
	while (cnt && (chan = hci_chan_sent(hdev, LE_LINK, &quote))) {
		u32 priority = (skb_peek(&chan->data_q))->priority;
		while (quote-- && (skb = skb_peek(&chan->data_q))) {
			BT_DBG("chan %p skb %p len %d priority %u", chan, skb,
			       skb->len, skb->priority);

			/* Stop if priority has changed */
			if (skb->priority < priority)
				break;

			skb = skb_dequeue(&chan->data_q);

			hci_send_frame(hdev, skb);
			hdev->le_last_tx = jiffies;

			cnt--;
			chan->sent++;
			chan->conn->sent++;

			/* Send pending SCO packets right away */
			hci_sched_sco(hdev);
			hci_sched_esco(hdev);
		}
	}

	if (hdev->le_pkts)
		hdev->le_cnt = cnt;
	else
		hdev->acl_cnt = cnt;

	if (cnt != tmp)
		hci_prio_recalculate(hdev, LE_LINK);
}

Vudentz, 2024-08-08:
@jamsliu2009 could you please attach the btmon traces?

`hci_sched_le(cnt -1 quote 0)[A 623,P 610]`

cnt underflowing or being bellow quote sounds really like a bug and then it is assigned back to hdev->le_cnt, so perhaps the culprit is hci_quote_sent returning a quote bigger than cnt, anyway with the logs we could see how the credits are being restored perhaps there is something else at play since this is the first time seeing something like this in a very very long time.

jamsliu2009, 2024-08-09:
Hi Vudentz,

Thanks for your reply. I read the hci_quote_sent() again. I think the reason why the quote is bigger than cnt is that it is calculated by hdev->le_cnt. However in the hci_sched_le() while loop, cnt is not assigned back to hdev->le_cnt after frame sent. Such that  it use old hdev->le_cnt value to calculate next quote. The following log snippet includes  more debug information for your reference.

For details please find 
[btmon-0809-trim.zip](https://github.com/user-attachments/files/16555609/btmon-0809-trim.zip)
[syslog-0809-trim.zip](https://github.com/user-attachments/files/16555610/syslog-0809-trim.zip)

Aug  9 10:29:16 james-vm3 kernel: [ 2067.771881] num_pkt=1,[A11383,P11372]
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771889] num_pkt=1,[A11383,P11373]
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771891] hci0 acl 0 sco 0 le 2 iso 0
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771893] hci_quote_sent q 2 cnt 2 num 1
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771894] chan 00000000ec43bcff num 1 quote 2
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771896] chan 00000000ec43bcff skb 0000000065af90c3 len 255 priority 6
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771901] hci_sched_le(cnt 1 quote 1)[A 11384,P 11373]
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771902] hci_quote_sent q 2 cnt 2 num 1
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771903] chan 000000008e09a8eb num 1 quote 2
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771903] chan 000000008e09a8eb skb 00000000e8ccfaf3 len 255 priority 0
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771905] hci_sched_le(cnt 0 quote 1)[A 11385,P 11373]
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771906] chan 000000008e09a8eb skb 00000000a379becd len 255 priority 0
Aug  9 10:29:16 james-vm3 kernel: [ 2067.771908] hci_sched_le(cnt -1 quote 0)[A 11386,P 11373]
Aug  9 10:29:16 james-vm3 kernel: [ 2067.772779] hci0 acl 0 sco 0 le -1 iso 0
Aug  9 10:29:16 james-vm3 kernel: [ 2067.772782] hci_quote_sent q -1 cnt -1 num 1
Aug  9 10:29:16 james-vm3 kernel: [ 2067.772783] chan 000000008e09a8eb num 1 quote -1
Aug  9 10:29:16 james-vm3 kernel: [ 2067.772785] chan 000000008e09a8eb skb 00000000340b5998 len 255 priority 0
Aug  9 10:29:16 james-vm3 kernel: [ 2067.772795] hci_sched_le(cnt -2 quote -2)[A 11387,P 11373]

Vudentz, 2024-08-09:
> Hi Vudentz,
>
> Thanks for your reply. I read the hci_quote_sent() again. I think the reason why the quote is bigger than cnt is that it is calculated by hdev->le_cnt. However in the hci_sched_le() [...]

Oh, maybe we need to follow what hci_sched_iso is doing and passing the cnt by reference to avoid having cnt being out of sync, that said this still sound very weird that we have not detected such problem for this long.

jamsliu2009, 2024-08-12:
Yes, even we are surprised to observe that this bug existed in the kernel for such a long time. We look forward to seeing the patch, thank you.

Vudentz, 2024-08-12:
@jamsliu2009 https://patchwork.kernel.org/project/bluetooth/patch/20240812153219.290388-1-luiz.dentz@gmail.com/

jamsliu2009, 2024-08-15:
Applied the patch and verified it for 50 iterations, the issue was not observed.

Bi11, 2024-11-08:
Eight months ago, I sent[ a patch](https://patchwork.kernel.org/project/bluetooth/patch/20240305182349.4182578-1-yuxinwang9999@gmail.com/) to the Linux-Bluetooth mailing list. Unfortunately, it was not merged, even though the changes are identical to the current fix. It seems that the mailing list is less active than GitHub. 😒

Vudentz, 2024-11-08:
> Eight months ago, I sent[ a patch](https://patchwork.kernel.org/project/bluetooth/patch/20240305182349.4182578-1-yuxinwang9999@gmail.com/) to the Linux-Bluetooth mailing list. Unfortunately, it was [...]

Your change was archived before we had a chance to review it, have you resend it would probably be merged, anyway it is definitely not very productive to create a fuss about it if it was fixed.

closed 2025-07-22 by github-actions[bot]