bluez/bluez #917: Re-pairing the Xbox Series controller leads to a flood of failed BT connection attempts

RyuzakiKK, 2024-08-02:
# Description

Re-pairing the Xbox Series X|S Wireless Controller to a second PC, result in the first PC to repeatedly fail to re-connect to the controller. Flooding the BT bandwidth with failed connection attempts.

# Steps to reproduce

1. Long press the pairing button on the Xbox Series controller
2. Pair it to a PC (A)
3. Shut down the PC (A)
4. Long press the pairing button and pair the controller to another PC (B)
5. Shut down that new PC (B)
6. Turn on the initial PC (A)
7. Turn on the Xbox controller

# Expected result

A sensible expected result should probably be one of the followings:
- Bluez should try to connect a few times and then give up
- Bluez should increase the timeout between each failed attempts to avoid flooding the BT bandwidth with pointless requests

# Actual result

Bluez of PC (A) is constantly attempting to connect to the controller without any success.

When this is happening, `btmon` was continuously reporting the following:
```
> HCI Event: LE Meta Event (0x3e) plen 12                                      #245 [hci1] 7.425583
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 58 Address: 3C:FA:06:46:53:ED (Microsoft Corporation)
        Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          LE Encryption
< HCI Command: LE Start Encryption (0x08|0x0019) plen 28                       #246 [hci1] 7.425696
        Handle: 58 Address: 3C:FA:06:46:53:ED (Microsoft Corporation)
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
        Long term key[16]: 4341742aaa9be78345c66ef92a896f32
> HCI Event: Command Status (0x0f) plen 4                                      #247 [hci1] 7.431576
      LE Start Encryption (0x08|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: Encryption Change (0x08) plen 4                                   #248 [hci1] 7.441576
        Status: PIN or Key Missing (0x06)
        Handle: 58 Address: 3C:FA:06:46:53:ED (Microsoft Corporation)
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                 #249 [hci1] 7.441599
        Handle: 58 Address: 3C:FA:06:46:53:ED (Microsoft Corporation)
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                      #250 [hci1] 7.442575
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                 #251 [hci1] 7.448576
        Status: Success (0x00)
        Handle: 58 Address: 3C:FA:06:46:53:ED (Microsoft Corporation)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                          {0x0001} [hci1] 7.448582
        LE Address: 3C:FA:06:46:53:ED (Microsoft Corporation)
        Reason: Connection terminated due to authentication failure (0x04)
bluetoothd[28098]: = profiles/input/hog-lib.c:info_read_cb() HID Information read fail..   7.449004
bluetoothd[28098]: = profiles/input/hog-lib.c:report_reference_cb() Read Report Refere..   7.449079
bluetoothd[28098]: = profiles/input/hog-lib.c:report_reference_cb() Read Report Refere..   7.449146
bluetoothd[28098]: = profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PN..   7.449213
```

Btmon output the first time you successfully connect the Xbox controller: [xbox_ok.btsnoop.zip](https://github.com/user-attachments/files/16471180/xbox_ok.btsnoop.zip)

Btmon output when it continuously fails to connect, after you paired the device to another PC: [xbox_ko.btsnoop.zip](https://github.com/user-attachments/files/16471192/xbox_ko.btsnoop.zip)

Journal of the failed state, with `bluetoothd -d`: [xbox_ko_journal_bluetoothd.log](https://github.com/user-attachments/files/16471199/xbox_ko_journal_bluetoothd.log)

# Test Environment

- Arch Linux
- Bluez 5.77
- Linux 6.10.1

```
❯ bluetoothctl info 3C:FA:06:46:53:ED
Device 3C:FA:06:46:53:ED (public)
	Name: Xbox Wireless Controller
	Alias: Xbox Wireless Controller
	Appearance: 0x03c4 (964)
	Icon: input-gaming
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Vendor specific           (00000001-5f60-4c4f-9c83-a7953298d40d)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v045Ep0B13d0515
```

RyuzakiKK, 2025-05-06:
> Remove stale label or comment

Still an issue. It seems like I can't change the labels, so I'm leaving this comment.

thomasnappee, 2025-09-14:
Can confirm on **Manjaro 25.0.8** using **package bluez-5.83-1** and controller **Microsoft Xbox Series S|X Controller**
No workaround found

TommyAmberson, 2025-10-28:
I can't get it reconnect again. If I forget and re-pair it still doesn't work. Is there a way to reset my controller?

TommyAmberson, 2025-10-28:
seems to be related to https://github.com/bluez/bluez/issues/605

thomasnappee, 2025-12-17:
Updating the firmware and bluez to 5.85 fixed the issue on Manjaro 25.1