bluez/bluez #918: concurrent LE advertisments and extended advertisements

dakejahl, 2024-08-03:
Is it possible to enable both classic LE advertisements and extended advertisements concurrently? I am working on implementing drone "Remote ID" which requires broadcasting both. I have tested my code successfully with classic and LE individually, but they do not work concurrently. The behavior is that whichever commands are issued first, that is what the controller is configured for. For example if I issue the command **LE Set Extended Advertising Parameters** followed by **LE Set Advertising Parameters**, the second command returns **Command Disallowed** and vice versa.

How can I enable these concurrently?

Vudentz, 2024-08-05:
> Is it possible to enable both classic LE advertisements and extended advertisements concurrently? I am working on implementing drone "Remote ID" which requires broadcasting both. I have tested my [...]

You seem to be confusing the use of Extended command doesn't mean the Extended version of the PDU is used, Legacy PDUs can still be selected. That said if you are issuing this manually that means you are probably not really using BlueZ since this is controlled by the kernel and the daemon automatically uses the right PDU based on the advertisement length or when a secondary PHY is selected.

dakejahl, 2024-08-05:
I am using bluez source code in my project. The work around I found is to toggle between standard advertisements and extended advertisements on every iteration through my main loop.
https://github.com/ARK-Electronics/RemoteIDTransmitter/blob/main/src/Transmitter/Transmitter.cpp#L89-L143

Remote ID for drones is essentially hijacking the advertising data field of the advertisements to broadcast the required information.

Set extended advertising params
https://github.com/ARK-Electronics/RemoteIDTransmitter/blob/main/src/Bluetooth/Bluetooth.cpp#L323
Set advertising params
https://github.com/ARK-Electronics/RemoteIDTransmitter/blob/main/src/Bluetooth/BluetoothLegacy.cpp#L76

Ideally I'd be able to have them both enabled simultaneously, but for whatever reason this is not possible.

closed 2024-08-05 by dakejahl