bluez/bluez #922: Dualshock 4 controllers disconnect when DualSense controller is also connected

dklementowski, 2024-08-10:
I've got weird problem on one PC that I can't reproduce elsewhere, so this might be hardware-specific. It doesn't seem like it's kernel because , because I still had an old version that used to work correctly on that machine and it's the same.

Dualshock 4 controllers (up to 2) work perfectly fine, but whenever there is DualSense controller connected, DS4 works initially for around 30s and then stops and turns off some more seconds later.

Some logs I've found that may be relevant seen when it happens or around that time:
```
bluetoothd[1111]: profiles/input/device.c:control_connect_cb() connect to 58:10:31:8E:CE:99: Host is down (112)
bluetoothd[1111]: profiles/input/device.c:hidp_send_message() BT socket write error: Resource temporarily unavailable (11)
bluetoothd[1111]: profiles/input/device.c:hidp_send_message() BT socket write error: Resource temporarily unavailable (11)

kernel: playstation 0005:054C:0CE6.000D: DualSense input CRC's check failed

kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: kernel bug: Touch jump detected and discarded.
                    See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: kernel bug: Touch jump detected and discarded.
                    See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: kernel bug: Touch jump detected and discarded.
                    See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: kernel bug: Touch jump detected and discarded.
                    See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: kernel bug: Touch jump detected and discarded.
                    See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: WARNING: log rate limit exceeded (5 msgs per 24h). Discarding future messages.

kwin_wayland[3233]: kwin_libinput: Libinput: event266 - Wireless Controller Touchpad: kernel bug: Touch jump detected and discarded.
                                   See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details

```

DualSense always works properly itself. Haven't tried to pair multiple DualSense controllers. It makes sense that there are some issues, as both controllers now use the same hid_playstation driver, but with exactly the same software configuration on different hardware I can't reproduce it.

Bluez version: 5.77-1.fc40
OS: Nobara Linux 
Affected bluetooth adapter: 8087:0025 Intel Corp. Wireless-AC 9260 Bluetooth Adapter

gabriele2000, 2024-08-10:
The Dualsense is an asshole.
I have it and I also have a pair of bluetooth SONY headphones, but it doesn't even matter... what matters is that the Dualsense hijacks the connection is the audio node goes to standby, meaning that if you don't connect the headphones to another "perpetual" source of signal, it'll go to "standby" and the Dualsense will kill the connection to the headphones, meaning that the Dualsense will remain alone, uncontested, the king.

dklementowski, 2024-08-15:
> The Dualsense is an asshole.
> 
> I have it and I also have a pair of bluetooth SONY headphones, but it doesn't even matter... what matters is that the Dualsense hijacks the connection is the audio node goes to standby, meaning that if you don't connect the headphones to another "perpetual" source of signal, it'll go to "standby" and the Dualsense will kill the connection to the headphones, meaning that the Dualsense will remain alone, uncontested, the king.
> 
> No matter the BlueZ version, this problem just exists and I couldn't help the devs to diagnose it and fix, since a "fix" has been created but never worked.

Would you mind sharing related issue and the "fix" and what hardware you had that problem on?
I think for now I can work around the issue by using separate bluetooth adapter for Dualsense, but maybe it's worth trying to at least figure out where the problem lies and as it's adapter specific, maybe it's specific kernel driver issue (like some power-saving features malfunctioning), hid_playstation (being unable to handle some features of different controllers at the same time) or something in the user space like bluez.

gabriele2000, 2024-08-15:
> Would you mind sharing related issue and the "fix" and what hardware you had that problem on?

I will certainly tell how I work around the problem.
The "fix" was the 5.66 version, for a week it gave me no problem, although it could be pure luck.

### Getting started

Let's start by saying: the wifi/bluetooth card that I'm using is the `Intel Corp. AX210 Bluetooth`

You need a pair of bluetooth headphones or earphones from SONY, that's what I tested, it could also "work" with every bluetooth device.
You also need the dualsense, no matter the firmware or version.

### The issues

There are multiple version of this issue:
1 - The Dualsense makes it impossible to connect other bluetooth devices
2 - The Dualsense hijacks the connection of existing connected bluetooth devices

I have a pair of WH-1000XM5 (wireless headphones), but this issue exists with my GF's WF-C500 too (true-wireless earbuds).

**[1]**
Let's connect the Dualsense first, all goes smoothly.
After we connect the Dualsense, let's connect my WH-1000XM5 (Just to be clear: other hardware is affected too, just connect something else, a new controller, a speaker, whatever)... they will not connect.

To fix that you have to disconnect the Dualsense, now you'll be able to connect everything else.

This issue will trigger the second one though, read further to find out!

**[2]**
After we connected the WH-1000XM5, we can now use them, the connection is very stable.
To trigger this issue, you must put the node on standby, by simply not playing a single sound.

How to do that?
Simple.

This is Helvum, it allows you to interface with Wireplumber and connect various nodes.
This is the situation now:
[image]
As you can see, both input nodes and output nodes are on standby (intermittent line).

Now, let's reproduce something:
[image]
As you can see, the nodes are active.

When they're active, the Dualsense can't hijack the connection by disconnecting other devices.
The Dualsense NEVER goes on standby mode, it's always active.
The headphones node will go on standby after a few seconds (if you monitor the bluetooth packets, they're full of empty data, the node goes on standby to now waste energy, the packets are not being sent anymore, saving bandwidth too).

### The workaround

The main issue is that when a node goes on standby mode, the Dualsense will hijack the connection.
The secondary issue is that the Dualsense prevent other devices to connect, and this is unfixable without disconnecting the controller first.

So let's focus on the main issue.
The fix is very easy btw: you just have to connect the headphones to another thing too, for example: when I'm with my GF, I connect the monitor of her WF-C500 to the playback of my WH-1000XM5.

Since we're not together today, I'll show an example:
[image]
Just pretend "built-in analog stereo" (the laptop internal speakers) is "WF-C500" and you get the same thing that I do with her.
You can see how the nodes are active and they'll forever remain like that until you disconnect the monitor to the other source.
Now the Dualsense will never hijack the connection.

This is basically the fix.

gabriele2000, 2024-09-23:
Latest version fixed EVERY problem that I had!
You guys are wizards!

closed 2025-07-21 by github-actions[bot]