bluez/bluez #924: Bluetooth headset disconnect after music starts

yohad, 2024-08-13:
Hi,

I have an arch linux (dual boot with windows 11) machine which I manage to pair my senheiser 350HD headset to. Everything will be alright until about a minute after I start playing music which then the headphone will disconnect (and then reconnect again and again).

From `journalctl -u bluetooth` I see:
```
Jul 29 22:21:59 sol systemd[1]: Starting Bluetooth service...
Jul 29 22:21:59 sol bluetoothd[3855]: Bluetooth daemon 5.77
Jul 29 22:21:59 sol systemd[1]: Started Bluetooth service.
Jul 29 22:21:59 sol bluetoothd[3855]: Starting SDP server
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support bap plugin
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support bass plugin
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support mcp plugin
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support vcp plugin
Jul 29 22:21:59 sol bluetoothd[3855]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support micp plugin
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support ccp plugin
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support csip plugin
Jul 29 22:21:59 sol bluetoothd[3855]: src/plugin.c:init_plugin() System does not support asha plugin
Jul 29 22:21:59 sol bluetoothd[3855]: Bluetooth management interface 1.22 initialized
Jul 29 22:21:59 sol bluetoothd[3855]: Battery Provider Manager created
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/ldac
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/aptx_hd
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aptx_hd
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/aptx
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aptx
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/aac
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aac
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/opus_g
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/opus_g
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/sbc
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/sbc
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/faststream
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/opus_05
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/opus_05
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Jul 29 22:21:59 sol bluetoothd[3855]: Endpoint registered: sender=:1.27 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Jul 29 22:22:37 sol bluetoothd[3855]: /org/bluez/hci0/dev_00_1B_66_C7_E2_0F/sep5/fd0: fd(31) ready
Jul 29 22:25:06 sol bluetoothd[3855]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jul 29 22:25:12 sol bluetoothd[3855]: /org/bluez/hci0/dev_00_1B_66_C7_E2_0F/sep5/fd1: fd(31) ready
Jul 29 22:26:37 sol bluetoothd[3855]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jul 29 22:26:42 sol bluetoothd[3855]: /org/bluez/hci0/dev_00_1B_66_C7_E2_0F/sep5/fd2: fd(31) ready
Jul 29 22:27:35 sol bluetoothd[3855]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jul 29 22:27:40 sol bluetoothd[3855]: /org/bluez/hci0/dev_00_1B_66_C7_E2_0F/sep5/fd3: fd(31) ready
Jul 29 22:28:23 sol bluetoothd[3855]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jul 29 22:28:29 sol bluetoothd[3855]: /org/bluez/hci0/dev_00_1B_66_C7_E2_0F/sep5/fd4: fd(31) ready
Jul 29 22:29:16 sol bluetoothd[3855]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jul 29 22:29:23 sol bluetoothd[3855]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jul 29 22:29:30 sol bluetoothd[3855]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

When I pair the headphones to windows everything works, so something I guess something must be wrong with my bluez configuration.

Vudentz, 2024-08-13:
@yohad are you repairing or does your headphone supports multi-point? it looks like this is related to HFP, in any case please collect the HCI trace using btmon.

yohad, 2024-08-13:
The btmon logs are here: https://gist.github.com/yohad/7c6fe65feb3a3eff32d5757be48fbbe1

I think the earphones supports multi-point but I currently removed the pairing from windows and only pair to the linux machine.

Vudentz, 2024-08-19:
> The btmon logs are here: https://gist.github.com/yohad/7c6fe65feb3a3eff32d5757be48fbbe1 This should contain one disconnect.
> 
> I think the earphones supports multi-point but I currently removed the pairing from windows and only pair to the linux machine.

```
> HCI Event: Disconnect Complete (0x05) plen 4             #88 [hci0] 40.843931
        Status: Success (0x00)
        Handle: 3
        Reason: Connection Timeout (0x08)
```

The connection timeout normally indicates that the controller/link layer cannot keep up the connection, so this is not the Host stack initiating the disconnection. That said what perhaps is the problem seems to be that there is a discovery going on, since there are device being found, that is perhaps affecting the connection for some reason.

yohad, 2024-08-21:
Weird, as the headphones works perfectly well with windows on the same hardware.

How should I disable discovery? I tried `discovarable off` and `scan off` in `bluetoothctl` but niether had an effect.
Am I missing something?