bluez/bluez #927: [5.77] PS4 Controller instantly disconnects after connect

bajidu, 2024-08-19:
Hi there,

sadly my ps4 ctrler instantly disconnects after connecting; this problem seems to have been fixed in an older version but I can't resolve it. I can **successfully scan, find, trust and pair**, then the device connects (ctrler shortly indicates it with light) and disconnects in 1s. The usb connection works.

Bluetoothctl:
``` [bluetooth]# [CHG] Device 70:20:84:72:CF:E9 WakeAllowed: yes
[bluetooth]# hci0 70:20:84:72:CF:E9 type BR/EDR connected eir_len 5
[CHG] Device 70:20:84:72:CF:E9 Connected: yes
[Wireless Controller]# hci0 70:20:84:72:CF:E9 type BR/EDR disconnected with reason 3
[CHG] Device 70:20:84:72:CF:E9 Connected: no
```

Archlinux 6.10.5
Also tested with 5.68 packages

Is it a regression, a new bug or is my setup missing something?
Best regards

bajidu, 2024-08-19:
[bluetoothtrace.txt](https://github.com/user-attachments/files/16662008/bluetoothtrace.txt)
Here is the trace, after doing the following:
1) removing the device in question (70:20:84:72:CF:E9) via bluetoothctl remove.
2) restarting bluetooth service
3) starting a new scan in bluetoothctl and putting the device in pairing mode
TRACE BEGINS
4) trust device
5) pair device (gets disconnected)
6) connect to device (gets disconnected again)

Hope this helps

Vudentz, 2024-08-19:
> [bluetoothtrace.txt](https://github.com/user-attachments/files/16662008/bluetoothtrace.txt) Here is the trace, after doing the following:
>
>     1. removing the device in question (70:20:84:72:CF:E9) [...]

```
> HCI Event: Disconnect Complete (0x05) plen 4            #460 [hci0] 27.660652
        Status: Success (0x00)
        Handle: 69 Address: 70:20:84:72:CF:E9 (Hon Hai Precision Industry Co., Ltd.)
        Reason: Remote User Terminated Connection (0x13)
```

There doesn't seem to be a Disconnect command so I assume this is the remote side initiating the disconnection, are there any error on bluetoothd logs?

bajidu, 2024-08-19:
Thanks for the super fast reply! Turns out I did not properly charge the device. Sorry for the confusion! This is obviously solved. Best regards and thanks again!

closed 2024-08-19 by bajidu