bluez/bluez #930: Unable to use generic bluetooth wireless keyboard

Base-115, 2024-08-21:
## So...
I bought a bluetooth wireless keyboard, but I can't use it in GNU/Linux. I checked and It work as expected in Windows, iOS, Android.
## Behavior
I can pair, connect and trust a bluetooth wireless keyboard, model Jiaxi WB-8022 in GNU/Linux, but any key will not make anything. The system seems ignore the keyboard although it's connect.
## Why it is a regression?
It seems regression since the keyboard works in some review photos where (Aliexpress) I bought this keyboard. The user is using some GNU/Linux distribuition with GNOME desktop manager.

<details>
<summary>Here is the review's screenshot and the link for the keyboard's page in Aliexpress (please, I don't have mean promotion the keyboard ad)</summary>

https://pt.aliexpress.com/item/1005005826816325.html?spm=a2g0o.order_list.order_list_main.48.763fcaa4XBJUmo&gatewayAdapt=glo2bra

[image]

</details>

## Laptop settings
<details>
	<summary>Here...</summary>

```
System:
  Kernel: 6.10.5-arch1-1 arch: x86_64 bits: 64
  Desktop: GNOME v: 46.4 Distro: Arch Linux
Machine:
  Type: Laptop System: Acer product: Aspire A515-51 v: V2.02
    serial: <superuser required>
  Mobo: KBL model: Charmander_KL v: V2.02 serial: <superuser required>
    UEFI: Insyde v: 2.02 date: 01/03/2019
Battery:
  ID-1: BAT1 charge: 23.8 Wh (62.6%) condition: 38.0/48.9 Wh (77.7%)
    volts: 15.3 min: 15.2
CPU:
  Info: dual core model: Intel Core i5-7200U bits: 64 type: MT MCP cache:
    L2: 512 KiB
  Speed (MHz): avg: 606 min/max: 400/2500 cores: 1: 609 2: 600 3: 600 4: 618
Graphics:
  Device-1: Intel HD Graphics 620 driver: i915 v: kernel
  Device-2: Generic HD WebCam driver: uvcvideo type: USB
  Display: wayland server: X.org v: 1.21.1.13 with: Xwayland v: 24.1.2
    compositor: gnome-shell driver: gpu: i915 resolution: no compositor data
    resolution: 1: 2560x1080 2: 1366x768
  API: EGL Message: EGL data requires eglinfo. Check --recommends.
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel
  API: ALSA v: k6.10.5-arch1-1 status: kernel-api
  Server-1: PipeWire v: 1.2.2 status: active
Network:
  Device-1: Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet
    driver: r8169
  IF: enp1s0f1 state: up speed: 100 Mbps duplex: full mac: <filter>
  Device-2: Qualcomm Atheros QCA9377 802.11ac Wireless Network Adapter
    driver: ath10k_pci
  IF: wlan0 state: up mac: <filter>
Bluetooth:
  Device-1: N/A driver: btusb type: USB
  Report: hciconfig ID: hci0 state: up address: <filter> bt-v: 4.2
Drives:
  Local Storage: total: 238.47 GiB used: 37.05 GiB (15.5%)
  ID-1: /dev/nvme0n1 vendor: Seagate model: XPG GAMMIX S11 Pro
    size: 238.47 GiB
Partition:
  ID-1: / size: 46.95 GiB used: 17.11 GiB (36.4%) fs: ext4 dev: /dev/nvme0n1p5
  ID-2: /boot size: 511 MiB used: 143 MiB (28.0%) fs: vfat
    dev: /dev/nvme0n1p1
  ID-3: /home size: 98.83 GiB used: 19.81 GiB (20.0%) fs: ext4
    dev: /dev/nvme0n1p7
Swap:
  ID-1: swap-1 type: zram size: 3.82 GiB used: 0 KiB (0.0%) dev: /dev/zram0
  ID-2: swap-2 type: partition size: 8 GiB used: 0 KiB (0.0%)
    dev: /dev/nvme0n1p2
Sensors:
  System Temperatures: cpu: 41.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Info:
  Memory: total: 8 GiB available: 7.64 GiB used: 4.07 GiB (53.4%)
  Processes: 246 Uptime: 52m Shell: Bash inxi: 3.3.35
```
</details>

## HCI Files
<details>
	<summary>Here...</summary>

[hcitrace.txt](https://github.com/user-attachments/files/16664739/hcitrace.txt)
[hcitrace.snoop](https://drive.google.com/file/d/1liO8eWyyoi70oEqHt_Sb5nsaHxRHkrz2/view?usp=drive_link) (available in Google Drive)
</details>

## Special thanks
Special thanks to **bad_at_usernames** from [Diolinux Plus forum](https://plus.diolinux.com.br/t/gnu-linux-nao-responde-ao-pressionamento-das-teclas-do-teclado-wb-8022-bluetooth/66064) (link for the post there) - a Portuguese forum about technologic focused in GNU/Linux - for support me and to forward me here.

## Excuse
Sorry for my bad english...
</details>

Base-115, 2024-08-22:
Hi, Vudentz! :wave: 

Yes, it make all sense!

After pair and connect, the keyboard seems waiting something and some seconds later, it sleeps. So I restart the keyboard through the switch. Always I do it, the MAC Address change and I need pair and connect until it "sleeps" again.

The keyboard keeps locked in this "loop".

Vudentz, 2024-08-22:
> Hi, Vudentz! 👋
> 
> Yes, it makes all sense!
> 
> After pair and connect, the keyboard seems waiting something and some seconds later, it sleeps. So I restart the keyboard through the switch. Always I do it, the MAC Address change and I need pair and connect until it "sleeps" again.
> 
> The keyboard keeps locked in this "loop".
> 
> EDIT: While it is "waiting" state (after pair and connect), nothing happens whatever key I press.

Perhaps it needs you to enter a pincode or something? Does it say something on its manual? It could be that you need a driver for it if that does work with the generic HID driver.

Base-115, 2024-08-23:
Actually, I didn't need a pincode or something when I connected the keyboard in other devices (Windows, Android and iOS), so I thing I don't need a pin for GNU/Linux too, although the manual explains that I always would require in any plataform.

Vudentz, 2024-08-23:
> Actually, I didn't need a pincode or something when I connected the keyboard in other devices (Windows, Android and iOS), so I thing I don't need a pin for GNU/Linux too, although the manual explains that I always would require in any plataform.

And if you enter a pincode does it work?

Base-115, 2024-08-25:
Sorry for delay...

It's not provide any pincode. I recorded my screen trying pairing this keyboard through the `bluetoothctl`. [The video is available on YouTube](https://youtu.be/7pGFWSdo4r8?si=33KlfIj_K_VBOjNS).

Base-115, 2024-09-03:
Hi, again.

I tried to connect the keyboard in my parent's computer through the Ubuntu 24.0.1 LTS Live CD session. Unfortunaly, the same behevaior happened - it seems stuck in "time-out" state.

However, I booted in Windows later in the same computer (it was already installed) and tried to connect the keyboard too. I could connect the keyboard without difficulty, it doesn't ask me for a PIN.

I think that's not very helpful, but It means that there are not issue with laptop hardware and even the my parent's computer.

closed 2025-07-20 by github-actions[bot]