bluez/bluez #931: btmgmt reports different selected PHY in interactive vs non-interactive mode

jpwright, 2024-08-21:
Ubuntu 23.04
bluez/btmgmt version 5.77

I need to enable 2M PHY on my adapter. After updating to bluez 5.77 from 5.66, I can no longer reliably do this using btmgmt in non-interactive mode - for whatever reason, the Supported/Configurable/Selected phy options are different.

I _am_ able to successfully enable 2M PHY in interactive mode, but this is not ideal for scripting.

```
root@torasu:/# btmgmt --index 2 phy
Supported phys: BR1M1SLOT BR1M3SLOT BR1M5SLOT EDR2M1SLOT EDR2M3SLOT EDR2M5SLOT EDR3M1SLOT EDR3M3SLOT EDR3M5SLOT LE1MTX LE1MRX LE2MTX LE2MRX LECODEDTX LECODEDRX 
Configurable phys: BR1M3SLOT BR1M5SLOT EDR2M1SLOT EDR2M3SLOT EDR2M5SLOT EDR3M1SLOT EDR3M3SLOT EDR3M5SLOT LE2MTX LE2MRX LECODEDTX LECODEDRX 
Selected phys: BR1M1SLOT BR1M3SLOT BR1M5SLOT EDR2M1SLOT LE1MTX LE1MRX LE2MTX LE2MRX LECODEDTX LECODEDRX 
root@torasu:/# btmgmt --index 2
[hci2]# phy
[hci2]# Supported phys: LE1MTX LE1MRX LE2MTX LE2MRX LECODEDTX LECODEDRX 
[hci2]# Configurable phys: LE2MTX LE2MRX LECODEDTX LECODEDRX 
[hci2]# Selected phys: LE1MTX LE1MRX
```

Vudentz, 2024-08-22:
> Ubuntu 23.04 bluez/btmgmt version 5.77
>
> I need to enable 2M PHY on my adapter. After updating to bluez 5.77 from 5.66, I can no longer reliably do this using btmgmt in non-interactive mode - for [...]

What kernel version? One of the things that have been changed is that we enable all supported PHYs by default since:

```
commit 288c90224eec55d13e786844b7954ef060752089
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Mon Dec 19 13:37:02 2022 -0800

    Bluetooth: Enable all supported LE PHY by default
    
    This enables 2M and Coded PHY by default if they are marked as supported
    in the LE features bits.
    
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
```

jpwright, 2024-08-22:
Kernel version: 6.2.0-39-generic

I have a peripheral device that attempts to update the connection PHY to 2M, but it only works if I select 2M PHY on the adapter in btmgmt interactive mode as I mentioned, otherwise the connection stays on 1M. (For what it's worth, my adapter is a nrf52840dongle running the Zephyr `hci_usb` sample, and my peripheral is a nrf5340 running custom firmware.)

Vudentz, 2024-08-22:
Hmm, perhaps something is broken when there is --index + a command, so it could be the is running the phy command on a different controller? It seems you have more than one given that you need to use the index.

jpwright, 2024-08-22:
I think that's exactly it, and digging into it a bit, I see this commit: https://github.com/bluez/bluez/commit/f00d5546c9e989dd68ce0de0190cd0e043b0f1f5

which fixes the issue if I cherry-pick it into the 5.77 release. So looks like this is fixed in develop and I just need to wait for the next release - feel free to close.

closed 2025-07-20 by github-actions[bot]