bluez/bluez #932: Missing MGMT disconnexion Event if device turned off before Disconnect command

quentino18, 2024-08-22:
I'm using Bluez 5.64 (Also tested with Bluez 5.75).
When a LE device is connected, and turned off before sending 'Disconnect' command (up to a few seconds), the disconnection attempt never reach an end and the device is still considered 'Connected'.
Afterwards, it is always considered 'Connected' and it is never possible to connect or disconnect to the device anymore.
Bluetoothd has to be restarted to fix it.

It turns out turn off device before sending 'Disconnect' command never triggers a 'Device Disconnected' MGMT Event.

Bluetoothd logs during 'normal' disconnection : 
```
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebad700: device 00:0D:B8:01:B0:D4 profile gap-profile state changed: connected -> disconnecting (0)
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebad700: device 00:0D:B8:01:B0:D4 profile gap-profile state changed: disconnecting -> disconnected (0)
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb28f0: device 00:0D:B8:01:B0:D4 profile batt-profile state changed: connected -> disconnecting (0)
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/battery.c:btd_battery_unregister() path = /org/bluez/hci0/dev_00_0D_B8_01_B0_D4
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb28f0: device 00:0D:B8:01:B0:D4 profile batt-profile state changed: disconnecting -> disconnected (0)
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb2a00: device 00:0D:B8:01:B0:D4 profile deviceinfo state changed: connected -> disconnecting (0)
 Aug 22 11:12:27 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb2a00: device 00:0D:B8:01:B0:D4 profile deviceinfo state changed: disconnecting -> disconnected (0)
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/adapter.c:dev_disconnected() Device 00:0D:B8:01:B0:D4 disconnected, reason 2
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/adapter.c:adapter_remove_connection()
 Aug 22 11:12:29 PC958 bluetoothd[86559]: plugins/policy.c:disconnect_cb() reason 2
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:0D:B8:01:B0:D4 type 1 status 0xe
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/device.c:device_bonding_failed() status 14
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/adapter.c:resume_discovery()
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/device.c:gatt_debug() Channel 0x59094eb91380 disconnected: Software caused connection abort
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/device.c:att_disconnected_cb()
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/device.c:att_disconnected_cb() Software caused connection abort (103)
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/device.c:att_disconnected_cb() Automatic connection disabled
 Aug 22 11:12:29 PC958 bluetoothd[86559]: src/gatt-database.c:btd_gatt_database_att_disconnected()
 Aug 22 11:12:29 PC958 bluetoothd[86559]: attrib/gattrib.c:g_attrib_unref() 0x59094ebac750: g_attrib_unref=0
```

Bluetoothd logs for disconnection with device just turned off : 
```
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebad700: device 00:0D:B8:01:B0:D4 profile gap-profile state changed: connected -> disconnecting (0)
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebad700: device 00:0D:B8:01:B0:D4 profile gap-profile state changed: disconnecting -> disconnected (0)
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb28f0: device 00:0D:B8:01:B0:D4 profile batt-profile state changed: connected -> disconnecting (0)
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/battery.c:btd_battery_unregister() path = /org/bluez/hci0/dev_00_0D_B8_01_B0_D4
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb28f0: device 00:0D:B8:01:B0:D4 profile batt-profile state changed: disconnecting -> disconnected (0)
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb2a00: device 00:0D:B8:01:B0:D4 profile deviceinfo state changed: connected -> disconnecting (0)
 Aug 22 11:17:44 PC958 bluetoothd[86559]: src/service.c:change_state() 0x59094ebb2a00: device 00:0D:B8:01:B0:D4 profile deviceinfo state changed: disconnecting -> disconnected (0)
 Aug 22 11:17:48 PC958 bluetoothd[86559]: src/device.c:gatt_debug() Channel 0x59094ebb2770 disconnected: Connection reset by peer
 Aug 22 11:17:48 PC958 bluetoothd[86559]: src/device.c:att_disconnected_cb()
 Aug 22 11:17:48 PC958 bluetoothd[86559]: src/device.c:att_disconnected_cb() Connection reset by peer (104)
 Aug 22 11:17:48 PC958 bluetoothd[86559]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
 Aug 22 11:17:48 PC958 bluetoothd[86559]: src/device.c:att_disconnected_cb() Automatic connection disabled
 Aug 22 11:17:48 PC958 bluetoothd[86559]: src/gatt-database.c:btd_gatt_database_att_disconnected()
 Aug 22 11:17:48 PC958 bluetoothd[86559]: attrib/gattrib.c:g_attrib_unref() 0x59094ebb26e0: g_attrib_unref=0`
```

Btmon output during 'normal' disconnection : 
```
 @ MGMT Command: Disconnect (0x0014) plen 7                                                              {0x0001} [hci0] 11:12:29.056716
         LE Address: 00:0D:B8:01:B0:D4 (XXXXXXXX)
 < HCI Command: Disconnect (0x01|0x0006) plen 3                                                               #55 [hci0] 11:12:29.056834
         Handle: 3585
         Reason: Remote User Terminated Connection (0x13)
 > HCI Event: Command Status (0x0f) plen 4                                                                    #56 [hci0] 11:12:29.059438
       Disconnect (0x01|0x0006) ncmd 1
         Status: Success (0x00)
 > HCI Event: Disconnect Complete (0x05) plen 4                                                               #57 [hci0] 11:12:29.062402
         Status: Success (0x00)
         Handle: 3585
         Reason: Connection Terminated By Local Host (0x16)
 @ MGMT Event: Command Complete (0x0001) plen 10                                                         {0x0001} [hci0] 11:12:29.062460
       Disconnect (0x0014) plen 7
         Status: Success (0x00)
         LE Address: 00:0D:B8:01:B0:D4 (XXXXXXXX)
```

Btmon output for disconnection with device just turned off : 
```
 @ MGMT Command: Disconnect (0x0014) plen 7                                                              {0x0001} [hci0] 11:17:46.056676
         LE Address: 00:0D:B8:01:B0:D4 (XXXXXXXX)
 < HCI Command: Disconnect (0x01|0x0006) plen 3                                                               #53 [hci0] 11:17:46.056742
         Handle: 3585
         Reason: Remote User Terminated Connection (0x13)
 > HCI Event: Command Status (0x0f) plen 4                                                                    #54 [hci0] 11:17:46.059243
       Disconnect (0x01|0x0006) ncmd 1
         Status: Success (0x00)
 > HCI Event: Disconnect Complete (0x05) plen 4                                                               #55 [hci0] 11:17:53.899239
         Status: Success (0x00)
         Handle: 3585
         Reason: Connection Timeout (0x08)
```

Indeed, there is no 'Command Complete' MGMT Event for 'Disconnect' command but I would maybe expect a 'Device Disconnected' MGMT Event after a few seconds like the following, when the device is turned off without requesting a disconnection : 
```
 > HCI Event: Disconnect Complete (0x05) plen 4                                                               #56 [hci0] 11:49:42.234456
        Status: Success (0x00)
        Handle: 3585
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                       {0x0001} [hci0] 11:49:42.234570
        LE Address: 00:0D:B8:01:B0:D4 (XXXXXXXX)
        Reason: Connection timeout (0x01)
```

I guess this behaviour is not considered normal. 
Is there anything I can do about it ?

Vudentz, 2024-08-22:
@quentino18 what kernel version?

quentino18, 2024-08-22:
I'm using Linux 6.9.3 but also tested on 6.6.47 with the same behaviour.

Vudentz, 2024-08-22:
This seems to be due to mgmt_device_disconnected checking for mgmt_connected flag and just bailing out:

https://github.com/bluez/bluetooth-next/blob/master/net/bluetooth/mgmt.c#L9741

But this behavior is actually incorrect since MGMT Disconnect can be called while connection procedure is still in progress in which case it would abort the connection attempt in the process, so we probably need to do something like:

```
diff --git a/net/bluetooth/mgmt.c b/net/bluetooth/mgmt.c
index 25979f4283a6..fdfe9675535d 100644
--- a/net/bluetooth/mgmt.c
+++ b/net/bluetooth/mgmt.c
@@ -9738,9 +9738,6 @@ void mgmt_device_disconnected(struct hci_dev *hdev, bdaddr_t *bdaddr,
        struct mgmt_ev_device_disconnected ev;
        struct sock *sk = NULL;
 
-       if (!mgmt_connected)
-               return;
-
        if (link_type != ACL_LINK && link_type != LE_LINK)
                return;
 
@@ -9754,7 +9751,12 @@ void mgmt_device_disconnected(struct hci_dev *hdev, bdaddr_t *bdaddr,
        if (hdev->suspended)
                ev.reason = MGMT_DEV_DISCONN_LOCAL_HOST_SUSPEND;
 
-       mgmt_event(MGMT_EV_DEVICE_DISCONNECTED, hdev, &ev, sizeof(ev), sk);
+       /* Only report MGMT_EV_DEVICE_DISCONNECTED if device is considered
+        * connected.
+        */
+       if (mgmt_connected)
+               mgmt_event(MGMT_EV_DEVICE_DISCONNECTED, hdev, &ev, sizeof(ev),
+                          sk);
 
        if (sk)
                sock_put(sk);
```

quentino18, 2024-08-26:
I have been able to test this patch but it doesn't fix the issue (Kernel LTS 6.6.47)

We are entering  `hci_cs_disconnect` with status `0x00`, so we wait for HCI_EV_DISCONN_COMPLETE
We are then entering `hci_disconn_complete_evt` with status `0x00` but `hci_conn_hash_lookup_handle` return `NULL`, so `mgmt_device_disconnected` is never called.

Here some Bluetooth kernel debug logs during disconnection with device just turned off  : 
```
[  178.814334][   T82] sock e1bdb3c9 sk 0d16e664
[  178.814470][   T82] channel 2 len 92
[  178.815957][   T82] sock e1bdb3c9 sk 0d16e664
[  178.816091][   T82] channel 2 len 149
[  178.817174][   T82] sock e1bdb3c9 sk 0d16e664
[  178.817457][   T82] channel 2 len 152
[  178.817974][   T82] sock e1bdb3c9 sk 0d16e664
[  178.818090][   T82] channel 2 len 154
[  178.818456][   T82] sock e1bdb3c9 sk 0d16e664
[  178.818672][   T82] channel 2 len 89
[  178.819041][   T82] sock e1bdb3c9 sk 0d16e664
[  178.819154][   T82] channel 2 len 81
[  178.819501][   T82] sock e1bdb3c9 sk 0d16e664
[  178.819617][   T82] channel 2 len 81
[  178.820016][   T82] sock e1bdb3c9 sk 0d16e664
[  178.820128][   T82] channel 2 len 157
[  178.820530][   T82] sock e1bdb3c9 sk 0d16e664
[  178.820640][   T82] channel 2 len 150
[  178.821074][   T82] sock e1bdb3c9 sk 0d16e664
[  178.821185][   T82] channel 2 len 103
[  178.821818][   T82] sock e1bdb3c9 sk 0d16e664
[  178.821940][   T82] channel 2 len 153
[  178.822351][   T82] sock e1bdb3c9 sk 0d16e664
[  178.822469][   T82] channel 2 len 148
[  178.822869][   T82] sock e1bdb3c9 sk 0d16e664
[  178.822980][   T82] channel 2 len 151
[  178.823618][   T82] sock b088ce2f sk 289d3d8f
[  178.823733][   T82] got 13 bytes
[  178.823799][   T82] channel 2 len 19
[  178.823854][   T82] 0
[  178.823890][   T82] hci0 orig refcnt 10
[  178.823929][   T82] hci0: sock 289d3d8f
[  178.824006][   T82] channel 2 len 19
[  178.824100][   T82] channel 2 len 19
[  178.824161][   T82] channel 3 len 13
[  178.827030][  T308] sock ba132482, sk e4c9eee1
[  178.827259][   T82] sock 289d3d8f
[  178.827390][   T56] hci0:
[  178.827441][   T56] hci0: entry 402ed6f0
[  178.827496][   T56] hci0: ADV monitoring is off
[  178.827529][   T56] hci0: stopping background scanning
[  178.827675][   T82] channel 2 len 22
[  178.827751][   T82] hci0 orig refcnt 11
[  178.829383][   T82] sock e1bdb3c9 sk 0d16e664
[  178.829515][   T82] channel 2 len 76
[  178.829952][   T82] sock b088ce2f, sk 289d3d8f
[  178.830935][   T82] sock e1bdb3c9 sk 0d16e664
[  178.831066][   T82] channel 2 len 92
[  178.832015][   T82] sock e1bdb3c9 sk 0d16e664
[  178.832149][   T82] channel 2 len 113
[  178.832420][   T82] sock a531939d, sk 70417023
[  181.143015][   T82] sock e1bdb3c9 sk 0d16e664
[  181.143150][   T82] channel 2 len 88
[  181.143685][   T82] sock b088ce2f sk 289d3d8f
[  181.143801][   T82] got 13 bytes
[  181.143867][   T82] channel 2 len 19
[  181.143921][   T82] 0
[  181.143959][   T82] hci0 orig refcnt 10
[  181.143996][   T82] hci0: sock 289d3d8f
[  181.144055][   T82] hcon f87fc192
[  181.144095][   T82] hci0: handle 0x401 reason 0x13
[  181.144217][   T82] hci0 orig refcnt 11
[  181.144338][   T56] hci0:
[... 56 lines cut ...]
[  183.186332][   T82] parent 50e61702
[  183.186378][   T82] sock da7c16fb, sk 50e61702, how 2
[  183.186431][   T82] Handling sock shutdown
[  183.186459][   T82] chan f9d932a6 orig refcnt 2
[  183.186502][   T82] chan f9d932a6 state BT_CLOSED
[  183.186547][   T82] chan f9d932a6 state BT_CLOSED
[  183.186588][   T82] chan f9d932a6 state BT_CLOSED
[  183.186630][   T82] chan f9d932a6 orig refcnt 3
[  183.186677][   T82] Sock shutdown complete err: 0
[  183.186713][   T82] chan f9d932a6 orig refcnt 2
[  183.186754][   T82] sk 50e61702 state BT_CLOSED
[  183.186790][   T82] chan f9d932a6 orig refcnt 3
[  183.186833][   T82] sk 50e61702
[  183.186869][   T82] chan f9d932a6 orig refcnt 2
[  183.186918][   T82] chan f9d932a6 orig refcnt 1
[  183.186963][   T82] chan f9d932a6
[  188.107515][   T56] hci0
[  188.107591][   T56] hci0 Event packet
[  188.107651][   T56] opcode 0x0406 status 0x00
[  188.107717][   T56] hci0: event 0x05
[  188.107760][   T56] hci0: status 0x00
[  188.107793][   T56] [qco_debug] : hci_disconn_complete_evt line 3386 status 0x00
[  188.107833][   T56] [qco_debug] : hci_disconn_complete_evt line 3392 hci_conn_hash_lookup_handle returned NULL
[  188.107865][   T56] hci0: result 0x00
```

Vudentz, 2024-08-26:
> I have been able to test this patch but it doesn't fix the issue (Kernel LTS 6.6.47)
>
> We are entering `hci_cs_disconnect` with status `0x00`, so we wait for HCI_EV_DISCONN_COMPLETE We are then [...]

Weid, well there is something calling hci_conn_del then, that said Id try with latest to make sure it was not fixed already.

Vudentz, 2024-08-26:
@quentino18 I think I know what is going on:

`[  183.178631][   T56] hci0: end: err -110`

This is likely the command timeout after 2 seconds since we are waiting for in hci_disconnect_sync:

```
		return __hci_cmd_sync_status_sk(hdev, HCI_OP_DISCONNECT,
						sizeof(cp), &cp,
						HCI_EV_DISCONN_COMPLETE,
						HCI_CMD_TIMEOUT, NULL);
```

So it will likely cause hci_conn_failed which does not currently call mgmt_device_disconnected.

quentino18, 2024-08-27:
Great ! Theses patches seems to work !

I tried with kernel branch `linux_rolling_release` (`6.10.6`)
Without the last patch, it indeed fail to generate `Comand Complete` for `MGMT_OP_DISCONNECT` : 

<details>
<summary>Linux 6.10.6 fail logs</summary>

```
[  176.444819] hci_sock_create:2195: sock 8f385896
[  176.445034] hci_sock_bind:1190: sock 8f385896 sk a3e09e67
[  176.445178] __hci_send_to_channel:320: channel 2 len 36
[  176.445604] hci_sock_getsockopt:2127: sk a3e09e67, opt 12
[  176.445758] hci_sock_setsockopt:2014: sk a3e09e67, opt 12
[  176.721204] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.721349] __hci_send_to_channel:320: channel 2 len 150
[  176.721875] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.722006] __hci_send_to_channel:320: channel 2 len 153
[  176.722453] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.722580] __hci_send_to_channel:320: channel 2 len 149
[  176.723020] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.723146] __hci_send_to_channel:320: channel 2 len 152
[  176.723589] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.723715] __hci_send_to_channel:320: channel 2 len 151
[  176.724183] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.724319] __hci_send_to_channel:320: channel 2 len 103
[  176.724973] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.725106] __hci_send_to_channel:320: channel 2 len 154
[  176.759822] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  176.759971] hci_dev_get:72: 0
[  176.760029] hci_dev_hold:1592: hci0 orig refcnt 10
[  176.760133] __hci_send_to_channel:320: channel 2 len 61
[  176.760193] hci_dev_put:1584: hci0 orig refcnt 11
[  179.130540] hci_sock_sendmsg:1801: sock 736e1c13 sk fb2fd970
[  179.130682] hci_mgmt_cmd:1621: got 13 bytes
[  179.130757] __hci_send_to_channel:320: channel 2 len 19
[  179.130826] hci_dev_get:72: 0
[  179.130870] hci_dev_hold:1592: hci0 orig refcnt 10
[  179.130917] disconnect:3066: hci0: sock fb2fd970
[  179.130996] hci_disconnect:183: hcon 104fc662
[  179.131056] hci_abort_conn:2931: hci0: handle 0x401 reason 0x13
[  179.131180] hci_dev_put:1584: hci0 orig refcnt 11
[  179.131320] hci_cmd_sync_work:300: hci0:
[  179.131386] hci_cmd_sync_work:317: hci0: entry b77e2968
[  179.131458] __hci_cmd_sync_sk:159: hci0: Opcode 0x0406
[  179.131511] hci_cmd_sync_add:93: hci0: opcode 0x0406 plen 3
[  179.131571] hci_cmd_sync_alloc:71: hci0: skb len 6
[  179.131619] hci_cmd_sync_run:123: hci0: length 1
[  179.131771] hci_cmd_work:4092: hci0 cmd_cnt 1 cmd queued 1
[  179.131851] hci_send_cmd_sync:4061: hci0: skb 430d59d9
[  179.131946] hci_send_frame:3018: hci0 type 1 len 6
[  179.133421] hci_sock_sendmsg:1801: sock bf74d2d4 sk b2224634
[  179.133713] __hci_send_to_channel:320: channel 2 len 76
[  179.134388] hci_rx_work:3987: hci0
[  179.134473] hci_rx_work:4031: hci0 Event packet
[  179.134565] hci_event_packet:7497: hci0: event 0x0f
[  179.134685] hci_cmd_status_evt:4322: hci0: opcode 0x0406
[  179.134747] hci_cs_disconnect:2711: hci0: status 0x00
[  181.167449] __hci_cmd_sync_sk:197: hci0: end: err -110
[  181.167550] hci_conn_failed:1247: hci0: status 0x13
[  181.167614] l2cap_connect_cfm:7238: hcon 104fc662 bdaddr 00:0d:b8:01:b0:d4 status 19
[  181.167699] l2cap_conn_del:1758: hcon 104fc662 conn 6c706e42, err 104
[  181.167783] l2cap_chan_hold:500: chan a9b0fc49 orig refcnt 2
[  181.167854] l2cap_chan_del:655: chan a9b0fc49, conn 6c706e42, err 104, state BT_CONNECTED
[  181.167929] smp_teardown_cb:3072: hci0: chan a9b0fc49
[  181.167987] l2cap_chan_put:517: chan a9b0fc49 orig refcnt 3
[  181.168042] l2cap_chan_put:517: chan a9b0fc49 orig refcnt 2
[  181.168100] l2cap_chan_put:517: chan a9b0fc49 orig refcnt 1
[  181.168156] l2cap_chan_destroy:489: chan a9b0fc49
[  181.168210] l2cap_chan_hold:500: chan 1254fce6 orig refcnt 3
[... 225 lines cut ...]
<details open>
<summary>Btmon with patches success logs</summary>

```
bluetoothd[83]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                     {0x0001} 0.969482
bluetoothd[83]: @ MGMT Command: Disconnect (0x0014) plen 7                                                    {0x0001} [hci0] 30.584662
        LE Address: 00:0D:B8:01:B0:D4 (SCHILLER AG)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                      #1 [hci0] 30.592932
        Handle: 1025
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                           #2 [hci0] 30.594886
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 10                                                               {0x0001} [hci0] 32.625247
      Disconnect (0x0014) plen 7
        Status: Success (0x00)
        LE Address: 00:0D:B8:01:B0:D4 (SCHILLER AG)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                      #3 [hci0] 38.090877
        Status: Success (0x00)
        Handle: 1025
        Reason: Connection Terminated By Local Host (0x16)
```
</details>

I'm closing the issue

closed 2024-08-27 by quentino18