bluez/bluez #933: Regression: Cannot detect keyboard (dis)connect using udev rules

aperezdc, 2024-08-23:
It used to be that an `udev` rule like this would work to run a program when a Bluetooth keyboard was (re)connected:

```
ACTION=="add", KERNEL=="hidraw*", SUBSYSTEMS=="hid", KERNELS="*:046D:B342.*" RUN+="/usr/bin/k380_conf -d /dev/%k -f on"
```

I suspect this has the same root cause as issue #848 and that after ee880bee85864b0e691370197c9de8d2f7ba0af2 the UHID device not being disconnected is the cause for the `remove` and `add` events not being received anymore.

Some keyboard like the Logitech K380 have switchable role for the `Fn` key, and the setting has to be re-applied after the keyboard has been power cycled. Now that the `add` event is no longer produced, this cannot be done.

aperezdc, 2024-08-23:
Note that I would not strictly need the `add` event, and any other event that would indicate a (re)connection would work. Currently with BlueZ 5.77 this is what gets logged by `udevadmn monitor -kup` for a disconnect:

```
KERNEL[2724174.256954] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256 (bluetooth)
ACTION=remove
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=65070

UDEV  [2724174.287410] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256 (bluetooth)
ACTION=remove
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=65070
USEC_INITIALIZED=49815238
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:256
SYSTEMD_WANTS=bluetooth.target
SYSTEMD_USER_WANTS=bluetooth.target
TAGS=:systemd:
CURRENT_TAGS=:systemd:
```

And for the reconnection:

```
KERNEL[2724212.177163] add      /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256 (bluetooth)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=65071

UDEV  [2724212.231179] add      /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256 (bluetooth)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256
SUBSYSTEM=bluetooth
DEVTYPE=link
SEQNUM=65071
USEC_INITIALIZED=49815238
SYSTEMD_ALIAS=/sys/subsystem/bluetooth/devices/hci0:256
SYSTEMD_WANTS=bluetooth.target
SYSTEMD_USER_WANTS=bluetooth.target
TAGS=:systemd:
CURRENT_TAGS=:systemd:

KERNEL[2724212.415789] change   /devices/virtual/misc/uhid/0005:046D:B342.0059/power_supply/hid-34:88:5d:49:db:fb-battery (power_supply)
ACTION=change
DEVPATH=/devices/virtual/misc/uhid/0005:046D:B342.0059/power_supply/hid-34:88:5d:49:db:fb-battery
SUBSYSTEM=power_supply
DEVTYPE=power_supply
POWER_SUPPLY_NAME=hid-34:88:5d:49:db:fb-battery
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_ONLINE=1
POWER_SUPPLY_CAPACITY=70
POWER_SUPPLY_MODEL_NAME=Keyboard K380
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_SCOPE=Device
SEQNUM=65072

UDEV  [2724212.423367] change   /devices/virtual/misc/uhid/0005:046D:B342.0059/power_supply/hid-34:88:5d:49:db:fb-battery (power_supply)
ACTION=change
DEVPATH=/devices/virtual/misc/uhid/0005:046D:B342.0059/power_supply/hid-34:88:5d:49:db:fb-battery
SUBSYSTEM=power_supply
DEVTYPE=power_supply
POWER_SUPPLY_NAME=hid-34:88:5d:49:db:fb-battery
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_ONLINE=1
POWER_SUPPLY_CAPACITY=70
POWER_SUPPLY_MODEL_NAME=Keyboard K380
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_SCOPE=Device
SEQNUM=65072
USEC_INITIALIZED=2724212415880
```

Interestingly, there is a `change` event for the keyboard battery, so technically it would be possible to hook into that and end up findind out which `/dev/hidrawN` device node is associated. But I would rather not go through the extra hoops, and get instead an event for the `hidraw` device node.

Vudentz, 2024-08-23:
> It used to be that an `udev` rule like this would work to run a program when a Bluetooth keyboard was (re)connected:
> 
> ```
> ACTION=="add", KERNEL=="hidraw*", SUBSYSTEMS=="hid", KERNELS="*:046D:B342.*" RUN+="/usr/bin/k380_conf -d /dev/%k -f on"
> ```
> 
> I suspect this has the same root cause as issue #848 and that after [ee880be](https://github.com/bluez/bluez/commit/ee880bee85864b0e691370197c9de8d2f7ba0af2) the UHID device not being disconnected is the cause for the `remove` and `add` events not being received anymore.
> 
> Some keyboard like the Logitech K380 have switchable role for the `Fn` key, and the setting has to be re-applied after the keyboard has been power cycled. Now that the `add` event is no longer produced, this cannot be done.

Well if you need it to be disconnected you can either disable uhid or set a timeout:

https://github.com/bluez/bluez/blob/master/profiles/input/input.conf#L10

aperezdc, 2024-08-26:
> Well if you need it to be disconnected you can either disable uhid or set a timeout:
> 
> https://github.com/bluez/bluez/blob/master/profiles/input/input.conf#L10

Disabling UHID worked in my case, thanks for the idea.

Nevertheless, if there are devices which need (or work better with) UHID, the issue remains and people may still want to get udev notified when a device is reconnected.

Of course, feel free to close the issue if supporting (re)connect events with UHID enabled is a non-goal.

willywonka-aaa, 2024-09-04:
/etc/bluetooth/input.conf --> UserspaceHID=false and 
/usr/lib/udev/rules.d/80-k380.rules  --> SUBSYSTEMS=="hid"

This worked for me,
The bad thing is that he no longer saves energy,
But now I can use my shortcuts, something strange that happened was that at 6 in the afternoon I changed and forced me to use the FN key but with this modification no longer.

Vudentz, 2024-10-28:
Fixed by https://github.com/bluez/bluez/commit/1782bfd791f76317a3f3ded2884606c1b5b164eb

closed 2024-10-28 by Vudentz