bluez/bluez #937: Bluetooth audio transport disconnection after 15 minutes

jipodine, 2024-08-27:
Hello,

I am trying to use a Raspberry Pi (3B+ or 4B) to play audio via a bluetooth speaker, but it stops after 15 minutes. I would welcome any help to track down the problem.

I connected, paired and trusted the device, once. Now, each time I turn the speaker on, it works for 15 minutes and then the transport disconnects. This is very reproducible.

> pi@dotpi-asterismes-001:~ $ bluetoothctl 
> [DEL] Transport /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1/fd2 
> [DEL] Endpoint /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1 
> [CHG] Device 10:94:97:24:D0:13 Connected: no

There are also pipewire and dbus errors, although I experienced the same disconnection with jackd.

> Aug 27 14:29:24 dotpi-asterismes-001 pipewire[704]: pw.node: (bluez_output.10_94_97_24_D0_13.1-65) running -> error (Received error eve>
> Aug 27 14:29:24 dotpi-asterismes-001 dbus-daemon[437]: [system] Rejected send message, 0 matched rules; type="method_return", sender=":>
> Aug 27 14:29:24 dotpi-asterismes-001 wireplumber[705]: (bluez_output.10_94_97_24_D0_13.1-19) running -> error (Received error event)

I am able to reconnect a variable number of times (5-10)

> [CHG] Device 10:94:97:24:D0:13 Connected: yes
> [NEW] Endpoint /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1
> [NEW] Transport /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1/fd3 [CHG] Transport /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1/fd3 Delay: 0x05dc (1500)
> [CHG] Transport /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1/fd3 Volume: 0x002f (47)
> [CHG] Transport /org/bluez/hci0/dev_10_94_97_24_D0_13/sep1/fd3 Volume: 0x0032 (50)
> [Ast-01]# 

And after a while the controller seems down.

> Aug 27 13:42:54 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
> Aug 27 13:42:54 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
> Aug 27 13:42:52 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
> Aug 27 13:42:52 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
> Aug 27 13:42:50 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
> Aug 27 13:42:50 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
> Aug 27 13:42:48 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
> Aug 27 13:42:48 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
> Aug 27 13:42:45 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
> Aug 27 13:42:45 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110

Then, it is not possible to do a `sudo hciconfig hci0 reset`, nothing is functional until a reboot.

Here is my configuration.

> pi@dotpi-asterismes-001:~ $ uname -a Linux dotpi-asterismes-001 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
> pi@dotpi-asterismes-001:~ $ pipewire --version
> pipewire
> Compiled with libpipewire 0.3.65
> Linked with libpipewire 0.3.65
> pi@dotpi-asterismes-001:~ $ wireplumber --version
> wireplumber
> Compiled with libwireplumber 0.4.13
> Linked with libwireplumber 0.4.13
> pi@dotpi-asterismes-001:~ $ apt list --installed '*blue*'
> Listing... Done
> bluez-firmware/stable,stable,now 1.2-9+rpt3 all [installed,automatic]
> bluez/stable,now 5.66-1+rpt1+deb12u1 arm64 [installed,automatic]
> libbluetooth3/stable,now 5.66-1+rpt1+deb12u1 arm64 [installed,automatic]
> libspa-0.2-bluetooth/stable,now 0.3.65-3+rpt8+deb12u1 arm64 [installed,automatic]
> pi-bluetooth/stable,stable,now 0.1.20 all [installed]
Speaker : UE Megaboom 3

I only changed one bluetooth property, for wireplumber to start without login:

> -- load this before /usr/share/wireplumber/bluetooth.lua.d/90-enable-all.lua
>
> -- do not start and stop bluez with login: always run
> bluez_monitor.properties["with-logind"] = false 

Does anybody experienced this issue?

Vudentz, 2024-08-28:
```
Aug 27 13:42:54 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
Aug 27 13:42:54 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
Aug 27 13:42:52 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
Aug 27 13:42:52 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
Aug 27 13:42:50 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
Aug 27 13:42:50 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
Aug 27 13:42:48 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
Aug 27 13:42:48 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
Aug 27 13:42:45 dotpi-asterismes-001 kernel: Bluetooth: hci0: hardware error 0x00
Aug 27 13:42:45 dotpi-asterismes-001 kernel: Bluetooth: hci0: Opcode 0x0c03 failed: -110
```

Ok, this is really ad since opcode 0x0c03 is HCI_Reset, so it looks like the controller is failing/crashing and even if we attempt to reset it doesn't and it fails again in a loop.

If Im not mistake the rpi uses broadcom controller, so you might want to get in contact with them, perhaps this has been fixed in the firmware or something.

jipodine, 2024-08-29:
Thank you for you hint. 

As far as I understand, however, the firmware version depends on the `bluez-firmware` package of Raspberry Pi OS, which is up-to-date. I reported the issue to bluez-firmware maintainer at debian.

I did not find a link to broadcom, for an other firmware or to report an issue. Do you know any place I could find or report issues?

closed 2025-07-21 by github-actions[bot]