bluez/bluez #939: L2CAP/LE/REJ/BI-02-c

Manju560, 2024-08-29:
Hi ,

I am facing issue to pass below PTS test case, Can any one provide solution?

L2CAP/LE/REJ/BI-02-C i getting failed because the device did not correctly handle unsupported L2CAP cammands  codes.Instead  of sending a "Command not understood" response,the device sent command Reject with "Invalid CID in request" reason for Disconnect Request. All other commands sent(including connection request, information request, Create channel request)  were rejected with a Command Reject response

Vudentz, 2024-08-30:
```

static inline int l2cap_le_sig_cmd(struct l2cap_conn *conn,
				   struct l2cap_cmd_hdr *cmd, u16 cmd_len,
				   u8 *data)
{
	int err = 0;

	switch (cmd->code) {
	case L2CAP_COMMAND_REJ:
		l2cap_le_command_rej(conn, cmd, cmd_len, data);
		break;

	case L2CAP_CONN_PARAM_UPDATE_REQ:
		err = l2cap_conn_param_update_req(conn, cmd, cmd_len, data);
		break;

	case L2CAP_CONN_PARAM_UPDATE_RSP:
		break;

	case L2CAP_LE_CONN_RSP:
		l2cap_le_connect_rsp(conn, cmd, cmd_len, data);
		break;

	case L2CAP_LE_CONN_REQ:
		err = l2cap_le_connect_req(conn, cmd, cmd_len, data);
		break;

	case L2CAP_LE_CREDITS:
		err = l2cap_le_credits(conn, cmd, cmd_len, data);
		break;

	case L2CAP_ECRED_CONN_REQ:
		err = l2cap_ecred_conn_req(conn, cmd, cmd_len, data);
		break;

	case L2CAP_ECRED_CONN_RSP:
		err = l2cap_ecred_conn_rsp(conn, cmd, cmd_len, data);
		break;

	case L2CAP_ECRED_RECONF_REQ:
		err = l2cap_ecred_reconf_req(conn, cmd, cmd_len, data);
		break;

	case L2CAP_ECRED_RECONF_RSP:
		err = l2cap_ecred_reconf_rsp(conn, cmd, cmd_len, data);
		break;

	case L2CAP_DISCONN_REQ:
		err = l2cap_disconnect_req(conn, cmd, cmd_len, data);
		break;

	case L2CAP_DISCONN_RSP:
		l2cap_disconnect_rsp(conn, cmd, cmd_len, data);
		break;

	default:
		BT_ERR("Unknown LE signaling command 0x%2.2x", cmd->code);
		err = -EINVAL;
		break;
	}

	return err;
}
```

```
	err = l2cap_le_sig_cmd(conn, cmd, len, skb->data);
	if (err) {
		struct l2cap_cmd_rej_unk rej;

		BT_ERR("Wrong link type (%d)", err);

		rej.reason = cpu_to_le16(L2CAP_REJ_NOT_UNDERSTOOD);
		l2cap_send_cmd(conn, cmd->ident, L2CAP_COMMAND_REJ,
			       sizeof(rej), &rej);
	}
```

We send that respond PTS is asking on if the command opcode is not understood, but from your comment it seems the Disconnect Request is sent which doesn't actually constitute an unsupported command and will be handled by:

```
static inline int l2cap_disconnect_req(struct l2cap_conn *conn,
				       struct l2cap_cmd_hdr *cmd, u16 cmd_len,
				       u8 *data)
{
	struct l2cap_disconn_req *req = (struct l2cap_disconn_req *) data;
	struct l2cap_disconn_rsp rsp;
	u16 dcid, scid;
	struct l2cap_chan *chan;

	if (cmd_len != sizeof(*req))
		return -EPROTO;

	scid = __le16_to_cpu(req->scid);
	dcid = __le16_to_cpu(req->dcid);

	BT_DBG("scid 0x%4.4x dcid 0x%4.4x", scid, dcid);

	chan = l2cap_get_chan_by_scid(conn, dcid);
	if (!chan) {
		cmd_reject_invalid_cid(conn, cmd->ident, dcid, scid);
		return 0;
	}
```

So we would only fail to parse with -EPROTO if the PDU length don't match, which doesn't seems to be the case.

closed 2025-07-21 by github-actions[bot]

Vudentz, 2026-01-15:
> [@Vudentz](https://github.com/Vudentz) I am also seeing L2CAP/LE/REJ/BI-02-C PTS test failure with BlueZ 5.78 on Linux Kernel 6.6.48, which is not allowing us to qualify. Do you have any recommended change to BlueZ or the Linux Kernel that would allow this to pass qualification?
> 
> Here is the portion of the PTS log relevant to this failure:
> 
> ```
> +8234 ms
> Verdict Description:     Sent=Disconnection Request: 
>     Code: [6 (0x06)] Code
>     Identifier: [14 (0x0E)]
>     Length: [4 (0x0004)]
>     Destination CID: [64 (0x0040)]
>     Source CID: [0 (0x0000)]
> 
> 
> 
> +8234 ms
> Send Event               HCI!ACL_DATA_REQ{
>                            connectionHandle=0x0000,
>                            boundary=0x00,
>                            broadcasting=0x00,
>                            length=0x000C,
>                            p_data[0]=0x08
>                          }
> 
> 
> +8359 ms
> Receive Event            HCI?ACL_DATA{
>                            connHandle=0x0000,
>                            length=14,
>                            boundaryFlag=0x02,
>                            broadcastFlag=0x00
>                          }
> 
> 
> +8359 ms
> Verdict Description:     ReceivePacket!
> receive=0A000500010E0600020040000000, CID=5
> 
> 
> 
> +8359 ms
> Verdict Description:     Received CMD:Command Reject: 
>     Code: [1 (0x01)] Code
>     Identifier: [14 (0x0E)]
>     Length: [6 (0x0006)]
>     Reason Field: [2 (0x0002)] Invalid CID in request
>     Requested Local CID: [64 (0x0040)]
>     Requested Remote CID: [0 (0x0000)]
> 
> 
> 
> +8359 ms
> Verdict Description:     expected:Command Reject: 
>     Code: [1 (0x01)] Code
>     Identifier: (lt)WildCard: Exists(gt)
>     Length: (lt)WildCard: Exists(gt)
>     Reason Field: [0 (0x0000)] Command not understood,
> but received:Command Reject: 
>     Code: [1 (0x01)] Code
>     Identifier: [14 (0x0E)]
>     Length: [6 (0x0006)]
>     Reason Field: [2 (0x0002)] Invalid CID in request
>     Requested Local CID: [64 (0x0040)]
>     Requested Remote CID: [0 (0x0000)]
> ```
> 
> I did make an attempt to patch the kernel based on what I thought you meant in your above response by changing
> 
> ```
> 	chan = l2cap_get_chan_by_scid(conn, dcid);
> 	if (!chan) {
> 		cmd_reject_invalid_cid(conn, cmd->ident, dcid, scid);
> 		return 0;
> 	}
> ```
> 
> to
> 
> ```
> 	chan = l2cap_get_chan_by_scid(conn, dcid);
> 	if (!chan) {
> 		return -EPROTO;
> 	}
> ```
> 
> But that appears to cause the test to fail even sooner than before. I would appreciate any input on other things I could try to resolve this. Thanks!

[image]

So we need to check what is L2CAP 2a/45? Perhaps that is L2CAP CoC that we do support since the other codes refer to codes used in classic, otherwise we would need a way to disable L2CAP CoC.

ceggers-arri, 2026-01-16:
I highly assume that `L2CAP 2/45a` (Disconnect command) must be enabled (together with `L2CAP 2/46` "credit based flow control", a.k.a CoC), simply because Linux implements it. If this option is disabled, the `L2CAP/LE/REJ/BI-02-C` (LE Command Reject) test would expect to see something like "not implemented" rather than "invalid parameter". In other words: The `L2CAP/LE/REJ/BI-02-C` test detects if somebody tries to cheat by checking fewer features than actually supported.