bluez/bluez #940: Provide a way to annotate vendor-specific UUIDs with bluetoothctl

vulpes2, 2024-08-31:
It would be very useful to have `bluetoothctl` display some more helpful descriptions for common vendor-specific UUIDs when they're available, instead of just "vendor specific". For example, `00001107-d102-11e1-9b23-00025b00a5a5` is the UUID for the GAIA protocol, and `df21fe2c-2515-4fdb-8886-f12c4d67927c` is Google's Fast Pair protocol.

While I don't think it's a good idea to include potentially inaccurate descriptions in `uuid.h`, would it be acceptable to create a separate file to provide descriptions for these common UUIDs for use in `bluetoothctl`? I'm happy to send a patch for this, just want to make sure it's not unwanted effort before I start working on it.

Vudentz, 2024-09-04:
@vulpes2 there is a table already for decoding vendor specific UUIDs:

https://github.com/bluez/bluez/blob/master/src/shared/util.c#L1602

vulpes2, 2024-09-04:
Somehow I missed that, thanks. I'll add a couple more and send in a patch at some point.

closed 2024-09-04 by vulpes2