bluez/bluez #942: $ bluetoothctl not working as expected, i.e., not functioning the same way as [bluetooth]

iconoclasthero, 2024-09-02:
**tl;dr: why is it not doing the same thing with the same commands regardless of whether i'm inside the bluetoothctl utility or making discrete calls from the cli/script?**

I'm trying to get away from having to log into bluetoothctl every time I want to connect to my speaker, i.e., I want to script the things I'm doing over and over so I can stop doing them over and over (and do something else _simpler_ over and over).

When I try this from the command line, i.e., a precursor to scripting the commands, I do not get the same functionality as when `$ bluethoothctl` is called and I "log into it."

```
$ bluetoothctl list; hr; bluetoothctl select DC:FB:48:62:F8:53 ; hr; bluetoothctl devices; hr; bluetoothctl show
Controller 00:15:83:45:34:B6 usb-chip [default]
Controller DC:FB:48:62:F8:53 internal bluetooth 
----------------------------------------------------------------------------------------------------------------------
Controller DC:FB:48:62:F8:53 internal bluetooth [default]
----------------------------------------------------------------------------------------------------------------------
Device C8:2B:6B:49:A6:A3 JBL TUNE510BT
----------------------------------------------------------------------------------------------------------------------
Controller 00:15:83:45:34:B6 (public)
	Manufacturer: 0x000a (10)
	Version: 0x04 (4)
	Name: latitude #1
	Alias: usb-chip
	Class: 0x007c0000 (8126464)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: SIM Access                (0000112d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	Modalias: usb:v1D6Bp0246d054D
	Discovering: no

$ bluetoothctl
Agent registeredct to bluetoothd...[bluetooth]#         JBL Xtreme 2
[JBL Xtreme 2]# list
Controller 00:15:83:45:34:B6 usb-chip [default]
Controller DC:FB:48:62:F8:53 internal bluetooth 
[JBL Xtreme 2]# select DC:FB:48:62:F8:53 
Controller DC:FB:48:62:F8:53 internal bluetooth [default]
[JBL Xtreme 2]# devices
Device 0C:7F:B2:59:AC:46 Living Room
Device 80:C5:F2:3E:51:6A Everywhere speaker
Device C8:2B:6B:49:A6:A3 JBL TUNE510BT
Device 00:42:79:D9:B6:AF JBL Xtreme 2
Device E8:07:B7:21:07:91 Vidonn F1
[JBL Xtreme 2]# show
Controller DC:FB:48:62:F8:53 (public)
	Manufacturer: 0x0002 (2)
	Version: 0x08 (8)
	Name: latitude
	Alias: internal bluetooth
	Class: 0x007c0000 (8126464)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (03b80e5a-ede8-4b33-a751-6ce34ec4c700)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: SIM Access                (0000112d-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d054D
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x05 (5)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedCapabilities.MaxAdvLen: 0x1f (31)
	SupportedCapabilities.MaxScnRspLen: 0x1f (31)
[JBL Xtreme 2]# 
quit
[JBL Xtreme 2]# 
```

why is it not doing the same thing with the same commands regardless of whether i'm inside the bluetoothctl utility or making discrete calls from the cli/script?    I don't see why this would be a preferred way to use the utility; it certainly is confusing.

iconoclasthero, 2024-09-02:
So, as further explanation, what I want to do is select the DC:... controller, then connect it to the 00:... JBL Extreme 2.
It works when I call `bluetoothctl`.
Why doesn't it work when I do it on the cli?

Vudentz, 2024-09-04:
@iconoclasthero if you run a command and exit the next one may not continue from where it last stopped, you should instead create a script and run with bluetoothctl -s <script>

closed 2024-09-04 by Vudentz

iconoclasthero, 2024-09-09:
Could use some help here:

```
BLUETOOTHCTL(1)                   General Commands Manual                   BLUETOOTHCTL(1)

NAME
       bluetoothctl: - interactive bluetooth control tool

DESCRIPTION
        bluetoothctl ver 5.49 Usage:

               bluetoothctl [options]

OPTIONS
       --agent
              Register agent handler: <capability>

       --timeout
              Timeout in seconds for non-interactive mode

       --version
              Display version

       --help Display help

```

```
$ bluetoothctl -s
bluetoothctl: option requires an argument -- 's'
bluetoothctl ver 5.72
Usage:
	bluetoothctl [--options] [commands]
Options:
	--agent 	Register agent handler: <capability>
	--endpoints 	Register Media endpoints
	--monitor 	Enable monitor output
	--timeout 	Timeout in seconds for non-interactive mode
	--version 	Display version
	--init-script 	Init script file
	--help 		Display help
```