bluez/bluez #949: Regression: ThinkPad TrackPoint Keyboard II is not reinitialized since 5.72

ValdikSS, 2024-09-10:
This seems like an issue initially caused by https://github.com/bluez/bluez/commit/ee880bee85864b0e691370197c9de8d2f7ba0af2
However neither setting `UserspaceHID=false` nor setting `IdleTimeout=5` helps on 5.77 and 5.78. The keyboard is always "reused" (no new input devices appear in dmesg even when keyboard is powered off for a long time), and this keyboard requires re initialization by the kernel module.

Managed to "fix" it by adding unconditional `force=true` here:

https://github.com/bluez/bluez/blob/615fc3592a57e32b42691a16b5d5cc88e378cbc1/profiles/input/hog-lib.c#L829-L838

See https://lore.kernel.org/linux-input/8e4a1afd-f82c-4262-b05b-ff5a1c18bee6@valdikss.org.ru/#t by @laf0rge

ValdikSS, 2024-09-11:
I should add that this keyboard is not only keyboard, but a **keyboard+mouse (trackpoint) combo** device.
This may be the reason why the checks for keyboard/non-keyboard are not working (e4f6c68f2a4dfbbdbc24379a5242564c65aa9ae2), because this device is both.

ValdikSS, 2024-09-20:
>you mention it is not reinitialized the replay logic was suppose to record and replay initialization messages

Hrm, how does it work? This keyboard requires initialization to switch it to "native" mode, this is done by the driver (`hid-lenovo`). I don't think BlueZ know anything about what the driver does. Or maybe the driver uses just a generic packet sending way, not a special proper initialization routine, if there's any.
I'm not very savvy in USB or Bluetooth or kernel modules, but I added this keyboard support to existing kernel module, and it stopped working properly since 5.72.

Vudentz, 2024-09-20:
> Hrm, how does it work? This keyboard requires initialization to switch it to "native" mode, this is done by the driver (`hid-lenovo`). I don't think BlueZ know anything about what the driver does. [...]

Because the input node is not destroyed the driver will be keep in the last state so bt_uhid will records the packets (put them on a list/queue) when it is first connected then it will replay (dequeue the messages) whenever reconnected. Perhaps this logic doesn't work for some reason e.g. the peripheral retain some state when disconnected then the packet recorded during the first connection does not make much sense.

ValdikSS, 2024-09-20:
The packets are the same every time. I don't know how does this feature work in bluez, but if it capture everything the driver sends upon connection and then replays everything, it should work. But for some reason it does not right now.

Vudentz, 2024-09-20:
> The packets are the same every time. I don't know how does this feature work in bluez, but if it capture everything the driver sends upon connection and then replays everything, it should work. But for some reason it does not right now.

Then perhaps there is a bug or something, you can capture the HCI trace using btmon and check if there is something different upon re-connection.

ValdikSS, 2024-09-20:
[tpIIkeyboard-bluez-bug.zip](https://github.com/user-attachments/files/17078063/tpIIkeyboard-bluez-bug.zip)

* Packets 8-271 - first connection, proper initialization
* Packets 272-280 - keyboard powered off
* Packets 273-329 - keyboard powered on, improper initialization.

ValdikSS, 2024-09-20:
[tpIIkeyboard-bluez-bug.btmon.zip](https://github.com/user-attachments/files/17078190/tpIIkeyboard-bluez-bug.btmon.zip)

Here's btmon dump, just to be sure and the encryption key is inside the dump.

* 1 - 272 - first connection, proper initialization
* 273 - 281 - power off
* 282 - 332 - power on

218 is the main initialization packet (switch to native mode), 221 and 224 are the additional parameters.

Vudentz, 2024-09-20:
> [tpIIkeyboard-bluez-bug.btmon.zip](https://github.com/user-attachments/files/17078190/tpIIkeyboard-bluez-bug.btmon.zip)
> 
> Here's btmon dump, just to be sure and the encryption key is inside the dump.
> 
>     * 1 - 272 - first connection, proper initialization
> 
>     * 273 - 281 - power off
> 
>     * 282 - 332 - power on
> 
> 
> 218 is the main initialization packet (switch to native mode, `btatt.value==0901`), 221 and 224 are the additional parameters.

Perhaps put them side by side or somehow to do a diff removing the packet/frame number and timing information.

ValdikSS, 2024-09-20:
There's no configuration packets on second connection.

Vudentz, 2024-09-23:
> There's no configuration packets on second connection.

We may need to debug why is that then, perhaps the configuration is not done via GET_REPORT/SET_REPORT? We only record/replay those reports currently:

```
	switch (ev.type) {
	case UHID_GET_REPORT:
	case UHID_SET_REPORT:
		bt_uhid_record(uhid, false, &ev);
		break;
	}
```

ValdikSS, 2024-09-23:
The keyboard is configured with output reports:

```
		ret = hid_hw_raw_request(hdev, 0x13, buf, 3,
					HID_FEATURE_REPORT, HID_REQ_SET_REPORT);
```

Vudentz, 2024-09-23:
> The keyboard is configured with output reports:
> 
> ```
>     /*
>      * Feature report 0x13 is used for USB,
>      * output report 0x18 is used for Bluetooth.
>      * buf[0] is ignored by hid_hw_raw_request.
>      */
> 	buf[0] = 0x18;
> 	buf[1] = byte2;
> 	buf[2] = byte3;
> 
> 
> 		ret = hid_hw_raw_request(hdev, 0x13, buf, 3,
> 					HID_FEATURE_REPORT, HID_REQ_SET_REPORT);
> ```

I would expect HID_REQ_SET_REPORT to be translated to UHID_SET_REPORT:

https://github.com/torvalds/linux/blob/master/include/uapi/linux/uhid.h#L41

That should be recorded though, UHID_OUTPUT is not so perhaps we need to do the following:

```
diff --git a/src/shared/uhid.c b/src/shared/uhid.c
index 20bd26781799..e8fc1f13e909 100644
--- a/src/shared/uhid.c
+++ b/src/shared/uhid.c
@@ -181,6 +181,7 @@ static bool uhid_read_handler(struct io *io, void *user_data)
        switch (ev.type) {
        case UHID_GET_REPORT:
        case UHID_SET_REPORT:
+       case UHID_OUTPUT:
                bt_uhid_record(uhid, false, &ev);
                break;
        }
```

ValdikSS, 2024-10-06:
> That should be recorded though, UHID_OUTPUT is not so perhaps we need to do the following:

Unfortunately, this patch did not help.

Vudentz, 2024-10-25:
@ValdikSS  could you please check with the above changes?

ValdikSS, 2024-10-26:
@Vudentz works fine now with this patch, thanks!
FYI, there's a typo in a commit text: `adds support supportfor persist`

closed 2024-10-28 by BluezTestBot