bluez/bluez #950: Build error: Pygments package not found

tot0rokr, 2024-09-11:
Hi,
An issue that was not present in previous builds of 3.65 is now present in the latest commit build.

I tried build on over 4 core of Arm64, Debian 10.

```
doc/l2cap.rst:19: (WARNING/2) Cannot analyze code. Pygments package not found.
Exiting due to level-2 (WARNING) system message.
make[1]: *** [Makefile:13897: doc/l2cap.7] Error 1
make[1]: *** Waiting for unfinished jobs....
doc/rfcomm.rst:19: (WARNING/2) Cannot analyze code. Pygments package not found.
Exiting due to level-2 (WARNING) system message.
make[1]: *** [Makefile:13897: doc/rfcomm.7] Error 1
make: *** [Makefile:4681: all] Error 2
```

It says that I don't have the Pygment package, and I've tried downloading it with pip3 or even trying it in a virtual environment to solve it.

I use the following build settings, and I'm wondering where the problem is occurring.

```
./bootstrap
./configure --enable-mesh --enable-testing --enable-tools --prefix=/usr --mandir=/usr/share/man  --sysconfdir=/etc --localstatedir=/var
make -j4
```

Additionally, there is a warning log from the configure run. The relevant apt packages are also attached.

```
checking cups directory... Package cups was not found in the pkg-config search path.                                                                                                                                                                                                                                                                Perhaps you should add the directory containing `cups.pc'
to the PKG_CONFIG_PATH environment variable
No package 'cups' found
```
```
root@e:/bluez# apt list | grep cups
                                                                                                                                                                                                                                                                                                                                                    WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

apcupsd-cgi/oldoldstable 3.14.14-2 arm64
apcupsd-doc/oldoldstable 3.14.14-2 all
apcupsd/oldoldstable 3.14.14-2 arm64
bluez-cups/oldoldstable 5.50-1.2~deb10u4 arm64
cups-backend-bjnp/oldoldstable 2.0.1-1 arm64
cups-browsed/oldoldstable 1.21.6-5+deb10u1 arm64
cups-bsd/oldoldstable 2.2.10-6+deb10u9 arm64
cups-client/oldoldstable 2.2.10-6+deb10u9 arm64
cups-common/oldoldstable 2.2.10-6+deb10u9 all
cups-core-drivers/oldoldstable 2.2.10-6+deb10u9 arm64
cups-daemon/oldoldstable 2.2.10-6+deb10u9 arm64
cups-filters-core-drivers/oldoldstable 1.21.6-5+deb10u1 arm64
cups-filters/oldoldstable 1.21.6-5+deb10u1 arm64
cups-ipp-utils/oldoldstable 2.2.10-6+deb10u9 arm64
cups-pk-helper/oldoldstable 0.2.6-1+b1 arm64
cups-ppdc/oldoldstable 2.2.10-6+deb10u9 arm64
cups-server-common/oldoldstable 2.2.10-6+deb10u9 all
cups-tea4cups/oldoldstable 3.14~alpha0+svn3576-1 all
cups-x2go/oldoldstable 3.0.1.4-1 all
cups/oldoldstable 2.2.10-6+deb10u9 arm64
libcups2-dev/oldoldstable 2.2.10-6+deb10u9 arm64
libcups2/oldoldstable,now 2.2.10-6+deb10u9 arm64 [installed,automatic]
libcupsfilters-dev/oldoldstable 1.21.6-5+deb10u1 arm64
libcupsfilters1/oldoldstable 1.21.6-5+deb10u1 arm64
libcupsimage2-dev/oldoldstable 2.2.10-6+deb10u9 arm64
libcupsimage2/oldoldstable 2.2.10-6+deb10u9 arm64
libnet-cups-perl/oldoldstable 0.64-1+b1 arm64
printer-driver-cups-pdf/oldoldstable 3.0.1-5 arm64
printer-driver-hpcups/oldoldstable 3.18.12+dfsg0-2 arm64
python-cups/oldoldstable 1.9.73-2+b1 arm64
python3-cups/oldoldstable 1.9.73-2+b1 arm64
python3-cupshelpers/oldoldstable 1.5.11-4 all
```

Vudentz, 2024-09-18:
@tot0rokr try with the following changes:

```
diff --git a/doc/l2cap.rst b/doc/l2cap.rst
index 2486f7c6f55a..f90f88222ae8 100644
--- a/doc/l2cap.rst
+++ b/doc/l2cap.rst
@@ -16,7 +16,7 @@ L2CAP protocol
 SYNOPSIS
 ========
 
-.. code-block:: c
+.. code-block::
 
     #include <sys/socket.h>
     #include <bluetooth/bluetooth.h>
@@ -40,7 +40,7 @@ connection, allowing for concurrent data transmission using different protocols.
 SOCKET ADDRESS
 ==============
 
-.. code-block:: c
+.. code-block::
 
     struct sockaddr_l2 {
         sa_family_t    l2_family;
@@ -52,7 +52,7 @@ SOCKET ADDRESS
 
 Example:
 
-.. code-block:: c
+.. code-block::
 
     struct sockaddr_l2 addr;
 
@@ -90,7 +90,7 @@ Channel security level, possible values:
 
 Example:
 
-.. code-block:: c
+.. code-block::
 
     int level = BT_SECURITY_HIGH;
     int err = setsockopt(l2cap_socket, SOL_BLUETOOTH, BT_SECURITY, &level,
@@ -116,7 +116,7 @@ authorization at profile level, possible values:
 
 Example:
 
-.. code-block:: c
+.. code-block::
 
     int defer_setup = 1;
     int err = setsockopt(l2cap_socket, SOL_BLUETOOTH, BT_DEFER_SETUP,
diff --git a/doc/rfcomm.rst b/doc/rfcomm.rst
index 4e43094b797d..b20c7544b64f 100644
--- a/doc/rfcomm.rst
+++ b/doc/rfcomm.rst
@@ -16,7 +16,7 @@ RFCOMM protocol
 SYNOPSIS
 ========
 
-.. code-block:: c
+.. code-block::
 
     #include <sys/socket.h>
     #include <bluetooth/bluetooth.h>
@@ -36,7 +36,7 @@ the 9 circuits of RS-232 (EIATIA-232-E) serial ports.
 SOCKET ADDRESS
 ==============
 
-.. code-block:: c
+.. code-block::
 
     struct sockaddr_rc {
         sa_family_t rc_family;
@@ -46,7 +46,7 @@ SOCKET ADDRESS
 
 Example:
 
-.. code-block:: c
+.. code-block::
 
     struct sockaddr_rc addr;
 
@@ -78,7 +78,7 @@ Channel security level, possible values:
 
 Example:
 
-.. code-block:: c
+.. code-block::
 
     int level = BT_SECURITY_HIGH;
     int err = setsockopt(rfcomm_socket, SOL_BLUETOOTH, BT_SECURITY, &level,
@@ -104,7 +104,7 @@ authorization at profile level, possible values:
 
 Example:
 
-.. code-block:: c
+.. code-block::
 
     int defer_setup = 1;
     int err = setsockopt(rfcomm_socket, SOL_BLUETOOTH, BT_DEFER_SETUP,
```

tot0rokr, 2024-09-19:
@Vudentz Hi, your patch is adapt to me well. I tested 615fc3592a57e32b42691a16b5d5cc88e378cbc1. Thank you.

closed 2024-09-19 by tot0rokr