bluez/bluez #953: `bluetoothd: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down`

paulmenzel, 2024-09-14:
Debian sid/unstable with *bluez* 5.77-1

I had a case, where I was unable to connect to (an already paired) Bluetooth speaker.

```
ep 14 10:27:11 abreu sudo[46970]:  pmenzel : TTY=pts/4 ; PWD=/home/pmenzel ; USER=root ; COMMAND=/usr/bin/systemctl start bluetooth
Sep 14 10:27:11 abreu sudo[46970]: pam_unix(sudo:session): session opened for user root(uid=0) by pmenzel(uid=5272)
Sep 14 10:27:11 abreu systemd[1]: Starting bluetooth.service - Bluetooth service...
Sep 14 10:27:11 abreu (uetoothd)[46975]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Sep 14 10:27:11 abreu bluetoothd[46975]: Bluetooth daemon 5.77
Sep 14 10:27:11 abreu systemd[1]: Started bluetooth.service - Bluetooth service.
Sep 14 10:27:11 abreu bluetoothd[46975]: Starting SDP server
Sep 14 10:27:11 abreu sudo[46970]: pam_unix(sudo:session): session closed for user root
Sep 14 10:27:11 abreu dbus-daemon[584]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.504' (uid=0 pid=46975 comm="/usr/l
ibexec/bluetooth/bluetoothd")
Sep 14 10:27:11 abreu bluetoothd[46975]: profiles/network/bnep.c:bnep_init() kernel lacks bnep-protocol support
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support network plugin
Sep 14 10:27:11 abreu systemd[1]: Starting systemd-hostnamed.service - Hostname Service...
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support bap plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support bass plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support mcp plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support vcp plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support micp plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support ccp plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support csip plugin
Sep 14 10:27:11 abreu bluetoothd[46975]: src/plugin.c:init_plugin() System does not support asha plugin
Sep 14 10:27:11 abreu kernel: Bluetooth: MGMT ver 1.23
Sep 14 10:27:11 abreu bluetoothd[46975]: Bluetooth management interface 1.23 initialized
Sep 14 10:27:11 abreu bluetoothd[46975]: Battery Provider Manager created
Sep 14 10:27:11 abreu kernel: Bluetooth: RFCOMM TTY layer initialized
Sep 14 10:27:11 abreu kernel: Bluetooth: RFCOMM socket layer initialized
Sep 14 10:27:11 abreu kernel: Bluetooth: RFCOMM ver 1.11
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/ldac
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/aptx_hd
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_hd
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/aptx
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/opus_g
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/opus_g
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/sbc
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/sbc
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/faststream
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/faststream_duplex
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/opus_05
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/opus_05
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Sep 14 10:27:11 abreu bluetoothd[46975]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Sep 14 10:27:11 abreu systemd[1]: Started systemd-hostnamed.service - Hostname Service.
Sep 14 10:27:11 abreu dbus-daemon[584]: [system] Successfully activated service 'org.freedesktop.hostname1'
Sep 14 10:27:27 abreu bluetoothd[46975]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Sep 14 10:27:27 abreu gnome-shell[1658]: Failed to connect device "JBL GO 2": GDBus.Error:org.bluez.Error.Failed: br-connection-page-timeout
Sep 14 10:27:39 abreu bluetoothd[46975]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Sep 14 10:27:39 abreu gnome-shell[1658]: Failed to connect device "JBL GO 2": GDBus.Error:org.bluez.Error.Failed: br-connection-page-timeout
Sep 14 10:27:41 abreu systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Sep 14 10:27:48 abreu bluetoothd[46975]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Sep 14 10:27:48 abreu gnome-shell[1658]: Failed to connect device "JBL GO 2": GDBus.Error:org.bluez.Error.Failed: br-connection-page-timeout
```

I tried several Linux kernel version, but in the end after several restarts it also worked correctly with the once problematic Linux kernel version 6.11.0-rc7-00097-g196145c606d0.

[Output `sudo btmon`](https://github.com/user-attachments/files/17003515/20240914--bluetooth--6.11.0-rc7-00097.txt)

nyabinary, 2024-09-17:
same here :p 
[image]
nixos tho

bobvanderlinden, 2024-10-09:
I also ran into this error while debugging my headset not being able to connect a second time. I'm quite confused by this problem. I've also tried reverting to an older version of my system (one where I knew it worked), but the same problem occurs there too now. Could it be new firmware of my headset?

To reproduce the issue I've done the following steps:

1 `systemctl stop bluetooth.service`
2. (Attempt to) Remove all state from bluez: `sudo rm -rf /var/lib/bluetooth/*`
3. `systemctl start bluetooth.service`
4. Start `blueberry`
5. Bring my headset into pairing mode.
6. Connect to headset in `blueberry`
7. Notice that headset works fine. Also disconnecting and connecting from the laptop and from the headset works fine.
8. Turn off the headset
9. `systemctl restart bluetooth.service`
10. (for the logs below) Start `btmon`
11. Turn on the headset.
12. Notice the headset fails to connect to the laptop. The headset removes the laptop from its paired devices.
13. Stop `btmon`

<details>
<summary>btmon log</summary>

```
Bluetooth monitor ver 5.76
btmon[16782]: = Note: Linux version 6.6.54 (x86_64)                                              0.671997
btmon[16782]: = Note: Bluetooth subsystem version 2.22                                           0.672000
= New Index: 90:65:84:C6:1B:40 (Primary,USB,hci0)                                         [hci0] 0.672001
= Open Index: 90:65:84:C6:1B:40                                                           [hci0] 0.672002
= Index Info: 90:65:84:C6:1B:40 (Intel Corp.)                                             [hci0] 0.672002
bluetoothd[16591]: @ MGMT Open: bluetoothd (privileged) version 1.22                    {0x0001} 0.672004
> HCI Event: Connect Request (0x04) plen 10                                            #1 [hci0] 9.744671
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Class: 0x240418
          Major class: Audio/Video (headset, speaker, stereo, video, vcr)
          Minor class: Headphones
          Rendering (Printing, Speaker)
          Audio (Speaker, Microphone, Headset)
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                          #2 [hci0] 9.744897
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                                              #3 [hci0] 9.745525
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                 #4 [hci0] 9.923506
        Status: Success (0x00)
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Role: Central (0x00)
> HCI Event: Vendor (0xff) plen 4                                                     #5 [hci0] 10.109708
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                          #6 [hci0] 10.112500
        Status: Success (0x00)
        Handle: 256
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                    #7 [hci0] 10.112815
        Handle: 256 Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
btmon[16782]: @ RAW Open: btmon (privileged) version 2.22                              {0x0002} 10.113232
btmon[16782]: @ RAW Close: btmon                                                       {0x0002} 10.113243
> HCI Event: Command Status (0x0f) plen 4                                             #8 [hci0] 10.113454
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                 #9 [hci0] 10.113560
        Scan enable: No Scans (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                          #10 [hci0] 10.114501
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
> ACL Data RX: Handle 256 flags 0x02 dlen 12                                         #11 [hci0] 10.121500
[... 285 lines cut ...]
        Status: Success (0x00)
```

</details>

[btsnoop.log](https://github.com/user-attachments/files/17318435/btsnoop.log)

After this I also tried connecting from laptop to headset:

14. Start `blueberry`
15. Run `journalctl --unit bluetooth --follow`
16. Connect to headset
17. Connecting fails
18. Notice the `Hands-Free Voice gateway failed connect` in the bluetooth logs

<details>
<summary>systemctl --unit bluetooth --follow</summary>

```
Oct 09 23:09:53 nac44250 bluetoothd[22751]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 2C:41:A1:C8:E5:04: Connection refused (111)
Oct 09 23:09:53 nac44250 bluetoothd[22751]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 2C:41:A1:C8:E5:04: Connection refused (111)
Oct 09 23:09:55 nac44250 bluetoothd[22751]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 2C:41:A1:C8:E5:04: Connection refused (111)
```

</details>

LautiLosio, 2025-04-17:
+1 to this. Used to connect to my phone without any hassle. Now i can't get it to connect no matter what i do

row-huh, 2025-10-07:
I'm facing the same problem. Has anyone figured out anything?

TheRealFame, 2025-11-05:
I have been having alot of issues where sometimes this error pops up, Hopefully the package can dive into more cards and hardware presets that may experience issues like these through funding and donations.

gtambara, 2025-11-22:
Had the same problem. With:

 src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 08:F0:B6:DD:40:09: Connection reset by peer (104)
profiles/audio/avdtp.c:avdtp_connect_cb() connect to 08:F0:B6:DD:40:09: Permission denied (13)
profiles/audio/avdtp.c:avdtp_connect_cb() connect to 08:F0:B6:DD:40:09: Permission denied (13)

I actively removed the bluetooth headset from paired devices, turned off and then on again, and then the system automatically detected the deviced trying to reconnect. Then i normally connected.

Maybe something to do with the device trying to reconnect at the same time as the system trying to communicate with the device and then the on-off cycle begins?

TheRealFame, 2025-11-22:
I believe hands free has always been one of the reasons the service felt unstable on my system? Although I'm not too sure, Because I always actively attempted to disable it in favor of just leaving my headsets the option for SBC, LDAC and AAC

[Tagged Unrelated].