bluez/bluez #958: Build error from obexd on 1a681aa0143e1b8f4e5d4ba7a74d26af398c3f75

tot0rokr, 2024-09-19:
The build will fail as of commit 1a681aa0143e1b8f4e5d4ba7a74d26af398c3f75. I found this bug commit using `git bisect`.
The build error is as follows:

```
/usr/bin/ld: obexd/client/obexd-manager.o:(.data.rel.ro.modules+0x98): undefined reference to `bip_init'
/usr/bin/ld: obexd/client/obexd-manager.o:(.data.rel.ro.modules+0xa0): undefined reference to `bip_exit'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:6057: obexd/src/obexd] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:4681: all] Error 2
```

### Whole logs

```
root@e:/bluez# ./configure --enable-mesh --enable-testing --enable-tools --prefix=/usr --mandir=/usr/share/man  --sysconfdir=/etc --localstatedir=/var
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for C/C++ restrict keyword... __restrict
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking whether gcc accepts -fPIE... yes
checking whether gcc accepts -fsanitize=address... yes
checking whether gcc accepts -fsanitize=leak... yes
checking whether gcc accepts -fsanitize=undefined... yes
checking build system type... aarch64-unknown-linux-gnu
checking host system type... aarch64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert aarch64-unknown-linux-gnu file names to aarch64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert aarch64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
[... 659 lines cut ...]
  CCLD     tools/oobtest
  CCLD     tools/advtest
  CCLD     tools/btgatt-client
  CCLD     tools/btgatt-server
  CCLD     tools/gatt-service
  CCLD     profiles/iap/iapd
  CCLD     tools/btmgmt
  CCLD     tools/obex-client-tool
  CCLD     tools/obex-server-tool
  CCLD     tools/bluetooth-player
  CCLD     tools/obexctl
  CCLD     src/bluetoothd
  CCLD     obexd/src/obexd
  CCLD     client/bluetoothctl
  CCLD     monitor/btmon
  CCLD     tools/rctest
  CCLD     tools/l2test
  CCLD     tools/l2ping
/usr/bin/ld: obexd/client/obexd-manager.o:(.data.rel.ro.modules+0x98): undefined reference to `bip_init'
/usr/bin/ld: obexd/client/obexd-manager.o:(.data.rel.ro.modules+0xa0): undefined reference to `bip_exit'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:6057: obexd/src/obexd] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:4681: all] Error 2
```

fdanis-oss, 2024-09-23:
Can you check if `obexd/client/bip.c` should be built on your system:
```
$ grep 'bip\.c' Makefile.obexd
$ grep 'bip\.c' Makefile.in
```

tot0rokr, 2024-09-24:
@fdanis-oss Hi. There is no output. But, the old revisions before mentioned commit build fine.

tot0rokr, 2024-09-24:
@fdanis-oss Hi, this is output based on the mentioned commit.
```
root@em:/bluez# grep 'bip\.c' Makefile.obexd
                        obexd/client/bip.h obexd/client/bip.c \
root@em:/bluez# grep 'bip\.c' Makefile.in
        obexd/client/bip.h obexd/client/bip.c obexd/client/map-event.c \
@OBEX_TRUE@                     obexd/client/bip.h obexd/client/bip.c \
obexd/client/obexd-bip.o: obexd/client/bip.c
@am__fastdepCC_TRUE@    $(AM_V_CC)$(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(obexd_src_obexd_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -MT obexd/client/obexd-bip.o -MD -MP -MF obexd/client/$(DEPDIR)/obexd-bip.Tpo -c -o obexd/client/obexd-bip.o `test -f 'obexd/client/bip.c' || echo '$(srcdir)/'`obexd/client/bip.c
@AMDEP_TRUE@@am__fastdepCC_FALSE@       $(AM_V_CC)source='obexd/client/bip.c' object='obexd/client/obexd-bip.o' libtool=no @AMDEPBACKSLASH@
@am__fastdepCC_FALSE@   $(AM_V_CC@am__nodep@)$(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(obexd_src_obexd_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -c -o obexd/client/obexd-bip.o `test -f 'obexd/client/bip.c' || echo '$(srcdir)/'`obexd/client/bip.c
obexd/client/obexd-bip.obj: obexd/client/bip.c
@am__fastdepCC_TRUE@    $(AM_V_CC)$(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(obexd_src_obexd_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -MT obexd/client/obexd-bip.obj -MD -MP -MF obexd/client/$(DEPDIR)/obexd-bip.Tpo -c -o obexd/client/obexd-bip.obj `if test -f 'obexd/client/bip.c'; then $(CYGPATH_W) 'obexd/client/bip.c'; else $(CYGPATH_W) '$(srcdir)/obexd/client/bip.c'; fi`
@AMDEP_TRUE@@am__fastdepCC_FALSE@       $(AM_V_CC)source='obexd/client/bip.c' object='obexd/client/obexd-bip.obj' libtool=no @AMDEPBACKSLASH@
@am__fastdepCC_FALSE@   $(AM_V_CC@am__nodep@)$(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(obexd_src_obexd_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -c -o obexd/client/obexd-bip.obj `if test -f 'obexd/client/bip.c'; then $(CYGPATH_W) 'obexd/client/bip.c'; else $(CYGPATH_W) '$(srcdir)/obexd/client/bip.c'; fi`
```

And, There is no output based on old commit 615fc3592a57e32b42691a16b5d5cc88e378cbc1.

I used below configuration options:

```
./bootstrap
./configure --enable-mesh --enable-testing --enable-tools --prefix=/usr --mandir=/usr/share/man  --sysconfdir=/etc --localstatedir=/var
```

Thank you.

fdanis-oss, 2024-09-24:
I just tried to build using your commands configuration and doesn't reproduce the build error.

What is strange is that there's no `CC       obexd/client/obexd-bip.o` in your original logs while `obexd/client/bip.c` is present in the Makefile.in.
Did you retry to build after the bootstrap and configure commands?

tot0rokr, 2024-09-24:
@fdanis-oss  I'm pretty sure I've done the bootstrap and configuration, but I think I'm missing something. I apologize. My mistake.

closed 2024-09-24 by tot0rokr