bluez/bluez #959: After CIS disconnects, Bluez will turn on scan automatically.

carlchenzekun, 2024-09-19:
The current logic is whenever Bluez receives a Disconnect Complete Event, it will turn the scan enable on to re-connect.
It does not check the connection handle got disconnected is a ACL handle or ISO handle. In the case below, the ACL connection is still alive but user turns off streaming. Bluez will enter scan mode which will interfere when users starts streaming again in the future.
[image]
[jabra1_BluezTurnsOnScan.zip](https://github.com/user-attachments/files/17063819/jabra1_BluezTurnsOnScan.zip)

Vudentz, 2024-09-19:
https://patchwork.kernel.org/project/bluetooth/patch/20240919165253.847475-1-luiz.dentz@gmail.com/

carlchenzekun, 2024-09-19:
Thanks a lot Luiz

Vudentz, 2024-09-19:
> Thanks a lot Luiz

Have you confirm the above changes fixed the problem?

carlchenzekun, 2024-09-19:
> > Thanks a lot Luiz
> 
> Have you confirm the above changes fixed the problem?

I need to have qa to verify that since I don't have enough bandwidth. But looking at the change it is what I meant... checking the connection type at hci_disconnect. I will keep it open and let you know once have some results. It will need a kernel rebuilt.

Vudentz, 2024-09-19:
> I need to have qa to verify that since I don't have enough bandwidth. But looking at the change it is what I meant... checking the connection type at hci_disconnect. I will keep it open and let you [...]

Yeah, the change is pretty straight forward, what I more concern about is if there are some other use case that maybe need to be addressed thus having some testing is probably a good idea.

Vudentz, 2024-09-27:
@carlchenzekun any update regarding this?

carlchenzekun, 2024-09-30:
> @carlchenzekun any update regarding this?

Hi Luiz, gathering the log and will discuss on Thursday's sync up as well as summerize here.

carlchenzekun, 2024-10-03:
Hi Luiz,

Applying the patch, I don't see the scan turned on after CIS got disconnected anymore. Normal LE reconnection procedure is not affected.

One thing I noticed is CIG get removed and immediately reconfigured (in screenshot) after CIS disconnection. It does not affect normal operation so I will close this issue for now. 
[image]

closed 2024-10-03 by carlchenzekun