bluez/bluez #960: mpris-proxy not working

lolpowerluke, 2024-09-19:
If I start mpris-proxy (in Konsole)
it says
```
org.bluez appeared
Bluetooth Adapter /org/bluez/hci0 found
Bluetooth Transport /org/bluez/hci0/dev_D4_3A_2C_76_CC_BE/sep4/fd1 found
```
which looks normal to me, its just that from observing logs from my autostart script
```
#! /bin/bash
mpris-proxy
```
it should also say 
```
Bluetooth Player /org/bluez/hci0/dev_D4_3A_2C_76_CC_BE/player0 found
Player org.mpris.MediaPlayer2.Pixel_7 created
```
and sometimes this happens, but since 2 days ago, it also says 
```
Bluetooth Player /org/bluez/hci0/dev_D4_3A_2C_76_CC_BE/player0 removed
player org.mpris.MediaPlayer2.Pixel_7 at :1.65 disappear
```
right after the player created lines

fyi, I use mpris-proxy with wireplumber and pipewire
and I use it to control my phone's music playback using my pc

I dont remember changing any configs related to bluez/mpris-proxy
I have tried reinstalling packages related to mpris-proxy

if I forgot any info you need, just lmk and I'll do my best to give it
also just lmk if this is the wrong repo to report this issue, I will move this to the appropriate repo in that case

lolpowerluke, 2024-09-19:
well after restarting my phone to apply a google play system update, it started working again, any ideas as to why this happens and how I can keep it from happening in the future?

Vudentz, 2024-09-19:
> well after restarting my phone to apply a google play system update, it started working again, any ideas as to why this happens and how I can keep it from happening in the future?

Well it is a proxy, perhaps the actual player application was not running on the phone?

lolpowerluke, 2024-09-19:
It was running, I have a macro (with macrodroid) that opens yt music when connecting to my PC
I also could still play music on my PC, it was just the control that didn't work

lolpowerluke, 2024-09-21:
well I guess you dont know either then, closing since fixed

closed 2024-09-21 by lolpowerluke