bluez/bluez #961: keeps disconnecting

skarpinis, 2024-09-27:
HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                     #1963 521.167520
        Status: Success (0x00)
        Handle: 256 Address: 17:15:10:03:67:48 (OUI 17-15-10)
        Reason: Remote User Terminated Connection (0x13)

Vudentz, 2024-09-27:
@skarpinis logs? Please collect HCI trace using btmon.

skarpinis, 2024-09-28:
[kodikdump.log](https://github.com/user-attachments/files/17173886/kodikdump.log)
thanks

benlc, 2024-10-10:
Also experienced tons of disconnects on 5.78-1.fc40 with bluetooth headphones. Reverting to 5.73-3.fc40 seems to be solving frequent disconnects

Vudentz, 2024-10-10:
```
> ACL Data RX: Handle 256 flags 0x02 dlen 18                                                                                              #45 [hci0] 29.692552
      Channel: 64 len 14 [PSM 25 mode Basic (0x00)] {chan 0}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 3 nosp 0
        ACP SEID: 1
        INT SEID: 1
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: SBC (0x00)
            Frequency: Unknown (0x00)
            Channel Mode: Unknown (0x00)
            Block Length: Unknown (0x00)
            Subbands: Unknown (0x00)
            Allocation Method: Unknown (0x00)
            Minimum Bitpool: 0
            Maximum Bitpool: 0
< ACL Data TX: Handle 256 flags 0x00 dlen 8                                                                                               #46 [hci0] 29.692588
      Channel: 64 len 4 [PSM 25 mode Basic (0x00)] {chan 0}
      AVDTP: Set Configuration (0x03) Response Reject (0x03) type 0x00 label 3 nosp 0
        Service Category: Reserved (0x29)
        Error code: UNSUPPORTED_CONFIGURATION (0x29)

```

Looks like the remote end is trying to configure some unsupported/invalid parameters, btw there quite many codecs enabled which in the past was causing headsets to get confused, I assume they come from pipewire/pulseaudio so you may want to turn some of them off to see if that helps.

skarpinis, 2024-10-20:
Did not help sadly

closed 2025-07-19 by github-actions[bot]