bluez/bluez #963: Document more vendor UUIDs

mirh, 2024-09-28:
I see there's already a short list [here](https://github.com/bluez/bluez/blob/5.78/src/shared/util.c#L1602), but much more could be done:

| UUID | Tentative explanation|
| ------------- | ------------- |
| [00000000-deca-fade-deca-deafdecacafe](https://github.com/bluez/bluez/blob/master/profiles/iap/main.c) | Iphone-side MFi/iAP/IAP2 protocol |
| [00000000-deca-fade-deca-deafdecacaff](https://wiomoc.de/misc/posts/mfi_iap.html) | Accessory-side MFi/iAP/IAP2 protocol |
| [d31fbf50-5d57-2797-a240-41cd484388ec](https://scholar.tecnico.ulisboa.pt/api/records/z5zF-ZFmJAPUjTaGOwIctI50IT9EcgTGBMtM/file/0ceb5c5e35cb81acd1c5fdbc843a90ff6d2782231f7f096f686bc6d1a47be1e2.pdf#page=76) | CarPlay vehicle-side |
| [1a29eaab-0173-bc88-1c45-4de166248d2d](https://www.programmersought.com/article/96301003550/)? | CarPlay iphone-side |
| [74ec2172-0bad-4d01-8f77-997b2be0722a](https://github.com/adolfintel/OpenPods/issues/58#issuecomment-2282312559) | Airpods advanced controls service or something |
| [2a72e02b-7b99-778f-014d-ad0b7221ec74](https://github.com/d4rken-org/capod/blob/main/app-common/src/main/java/eu/darken/capod/pods/core/apple/protocol/ContinuityProtocol.kt) | Always hand-in-hand with the previous one (guy calls it continuity protocol?) |
| [96cc203e-5068-46ad-b32d-e316f5e069ba](https://github.com/Keller18306/SonyHeadphonesProtocol/tree/main/src/protocol/v1_deprecated) | Sony advanced controls protocol v1 (broadcom specific?) |
| [74ec2172-0bad-4d01-8f77-997b2be0722a](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/sony/headphones/SonyHeadphonesIoThread.java) | Sony advanced controls protocol v2 (aka MDR table set 2) |
| [4de17a00-52cb-11e6-bdf4-0800200c9a66](https://github.com/openDsh/openauto/blob/develop/btservice/AndroidBluetoothService.cpp) | Android Auto protocol |
| [669a0c20-0008-f4bd-e611-cb52007ae14d](https://github.com/Romern/gms_decompiled/blob/master/sources/p000/pck.java#L17C29-L17C30) | Android Auto.. but "3"? Totally unclear |
| [7b265b0e-2232-4d45-bef4-bb8ae62f813d](https://github.com/egaebel/sony-headphones-hack/blob/c350ccf1201eb12224c0fb0a43569f6a2e16adbe/base-dex2jar.jar.src--files/com/csr/gaia/library/GaiaLink.java#L17) | Sony calls it "Gaia link" while Windows prefers "CSR Gaia" |
| [fe59bfa8-7fe3-4a05-9d94-99fadc69faff](https://github.com/DyLanCao/2300_audio_open_sdk/blob/main/services/ble_profiles/voicepath/gsound/gsound_gatt_server.c) | Gsound/bisto([?](https://www.androidauthority.com/google-assistant-bose-quietcomfort-35-ii-800543/)) communication service.. Might be google assistant? |
| [f8d1fbe4-7966-4334-8024-ff96c9330e15](https://github.com/qqq5127/BES_2300/blob/BES_2300/services/voicepath/gsound/gsound_target_api_read_only/gsound_target_bt.h#L103) | Gsound data (SDP for control channel), Windows names it "AudioWear" |
| [81c2e72a-0591-443e-a1ff-05f988593351](https://github.com/qqq5127/BES_2300/blob/BES_2300/services/voicepath/gsound/gsound_custom/src/gsound_custom_bt.cpp#L101) | Gsound voice (SDP for audio), Windows names it "AudioWear Voice" |
| [0000fe03-0000-1000-8000-00805f9b34fb](https://docs.nordicsemi.com/bundle/ncs-1.7.0/page/nrf/libraries/bluetooth_services/services/gadgets.html) | Alexa Gadgets Service |

I also found [this](https://github.com/egaebel/sony-headphones-hack/blob/main/base-dex2jar.jar.src--files/com/sony/songpal/ble/client/ServiceUuid.java), but idk wtf.

| UUID | Service |
| ------------- | ------------- |
| 5b833e01-6bc7-4802-8e9a-723ceca4bd8f | BT_CONNECTION_SERVICE |
| 5b833e02-6bc7-4802-8e9a-723ceca4bd8f | BT_MULTI_AUDIO_SERVICE |
| 5b833e03-6bc7-4802-8e9a-723ceca4bd8f | BT_BROADCAST_AUDIO_SERVICE |
| 5b833e04-6bc7-4802-8e9a-723ceca4bd8f | NW_SETTING_WITH_EXT_APP_SERVICE |
| 5b833e05-6bc7-4802-8e9a-723ceca4bd8f | BLUETOOTH_PAIRING_SERVICE |
| 5b833e06-6bc7-4802-8e9a-723ceca4bd8f | BLUETOOTH_IAP_CONNECTION_SERVICE |
| 5b833e07-6bc7-4802-8e9a-723ceca4bd8f | MODEL_INFORMATION_SERVICE |
| 5b833e08-6bc7-4802-8e9a-723ceca4bd8f | WIFI_CONNECTION2_SERVICE |
| 5b833e09-6bc7-4802-8e9a-723ceca4bd8f | PARTY_CONNECT_SERVICE |
| 5b833e0a-6bc7-4802-8e9a-723ceca4bd8f | BLUETOOTH_IAP_CONNECTION_MC_SERVICE |
| 5b833e20-6bc7-4802-8e9a-723ceca4bd8f | TANDEM_OVER_BLE_HPC_SERVICE |
| 5b833e21-6bc7-4802-8e9a-723ceca4bd8f | TANDEM_OVER_BLE_MC_SERVICE |
| 5b833e30-6bc7-4802-8e9a-723ceca4bd8f | BLANC_FOTA_SERVICE |
| 5b833e40-6bc7-4802-8e9a-723ceca4bd8f | CONFIG_SERVICE |
| 5b830000-6bc7-4802-8e9a-723ceca4bd8f | UNKNOWN_SERVICE |

vulpes2, 2024-09-30:
https://github.com/bluez/bluez/issues/940 I've also got a list of stuff that I haven't had time to clean up and format into a patch yet, feel free to do so if you can.

mirh, 2024-09-30:
Finding out all this stuff was frankly already far too tiresome. 
*Besides* while I only put here the identifiers that I had some "more likely than not" explanation for, if somebody with access to the appropriate documentation could improve on the names it'd be much neater.

mirh, 2025-05-03:
Definitively still actual.

rkyurpan256, 2025-05-09:
Hello here some user dev experience:

when I `sdptool browse <BT ADDR>` to browse the BT services, here what I observe for _iPhone 14 Pro_ and _iPhone 5_. (The following are just snippets from the command execution):
- UUIDs for [iPhone 14 Pro](#iphone-14-pro)
- UUIDs for [iPhone 5](#iphone-5)

_PS: Indeed a rabbit hole it is... I feel your pain 😅_

---
### iPhone 14 Pro

 > **Service Name: AAP Client** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Service RecHandle: 0x4f493aa0
> Service Class ID List:
>   **UUID 128: 1ff31936-572e-4b36-a2bf-b2409b1aa6f4** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>     PSM: 4107
> Language Base Attr List:
>   code_ISO639: 0x656e
>   encoding:    0x6a
>   base_offset: 0x100
>   code_ISO639: 0x6672
>   encoding:    0x6a
>   base_offset: 0x110
>   code_ISO639: 0x6465
>   encoding:    0x6a
>   base_offset: 0x120
>   code_ISO639: 0x6a61
>   encoding:    0x6a
>   base_offset: 0x130
> Profile Descriptor List:
>   "" (0x4b6f7c74-07f4-49de-b0b9-ab4304728f29)
>     Version: 0x0100
> 
> Service Name: MAP MAS-iOS
> Service RecHandle: 0x4f491132
> Service Class ID List:
>   "Message Access - MAS" (0x1132)
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 2
>   "OBEX" (0x0008)
> Language Base Attr List:
>   code_ISO639: 0x656e
>   encoding:    0x6a
>   base_offset: 0x100
>   code_ISO639: 0x6672
>   encoding:    0x6a
>   base_offset: 0x110
>   code_ISO639: 0x6465
>   encoding:    0x6a
>   base_offset: 0x120
>   code_ISO639: 0x6a61
>   encoding:    0x6a
>   base_offset: 0x130
> Profile Descriptor List:
>   "Message Access" (0x1134)
>     Version: 0x0104
> 
> **Service Name: Wireless iAP v2** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Service RecHandle: 0x4f49acb0
> Service Class ID List:
>   **UUID 128: 02030302-1d19-415f-86f2-22a2106a0a77** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 1
> Language Base Attr List:
>   code_ISO639: 0x656e
>   encoding:    0x6a
>   base_offset: 0x100
>   code_ISO639: 0x6672
>   encoding:    0x6a
>   base_offset: 0x110
>   code_ISO639: 0x6465
>   encoding:    0x6a
>   base_offset: 0x120
>   code_ISO639: 0x6a61
>   encoding:    0x6a
>   base_offset: 0x130
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0102
> 
> **Service Name: Wireless iAP** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Service RecHandle: 0x4f49ac90
> Service Class ID List:
>   **UUID 128: 00000000-deca-fade-deca-deafdecacafe** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "#" (0x0003)
>     Channel: 1
> Language Base Attr List:
>   code_ISO639: 0x656e
>   encoding:    0x6a
>   base_offset: 0x100
>   code_ISO639: 0x6672
>   encoding:    0x6a
>   base_offset: 0x110
>   code_ISO639: 0x6465
>   encoding:    0x6a
>   base_offset: 0x120
>   code_ISO639: 0x6a61
>   encoding:    0x6a
>   base_offset: 0x130
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0102

---
### iPhone 5

> Service Name: MAP MAS-iOS
> Service RecHandle: 0x4f491132
> Service Class ID List:
>   "Message Access - MAS" (0x1132)
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 2
>   "OBEX" (0x0008)
> Language Base Attr List:
>   code_ISO639: 0x656e
>   encoding:    0x6a
>   base_offset: 0x100
>   code_ISO639: 0x6672
>   encoding:    0x6a
>   base_offset: 0x110
>   code_ISO639: 0x6465
>   encoding:    0x6a
>   base_offset: 0x120
>   code_ISO639: 0x6a61
>   encoding:    0x6a
>   base_offset: 0x130
> Profile Descriptor List:
>   "Message Access" (0x1134)
>     Version: 0x0100
> 
> **Service Name: Wireless iAP** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Service RecHandle: 0x4f4964e0
> Service Class ID List:
>   **UUID 128: 00000000-deca-fade-deca-deafdecacafe** &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<<========
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 1
> Language Base Attr List:
>   code_ISO639: 0x656e
>   encoding:    0x6a
>   base_offset: 0x100
>   code_ISO639: 0x6672
>   encoding:    0x6a
>   base_offset: 0x110
>   code_ISO639: 0x6465
>   encoding:    0x6a
>   base_offset: 0x120
>   code_ISO639: 0x6a61
>   encoding:    0x6a
>   base_offset: 0x130
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0100

mirh, 2025-09-27:
Because I'm just a random doofus, and if somebody with actual expertise (and even better, documentation!) chimed in I'd be glad.

closed 2025-11-05 by github-actions[bot]