bluez/bluez #964: Issue with building bluez from source

aramcpp, 2024-09-30:
I'm trying to build bluez-5.7.4 from source. First of all I think README needs fixing because it says that you need to run `./configure` but the file doesn't exists, but there is `configure.ac` that means I need to run `autoconf` beforehand.
But running `autoconf` I get this lines.
```
configure.ac:5: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:11: error: possibly undefined macro: AM_MAINTAINER_MODE
configure.ac:24: error: possibly undefined macro: AM_PROG_CC_C_O
configure.ac:25: error: possibly undefined macro: AC_PROG_CC_PIE
configure.ac:26: error: possibly undefined macro: AC_PROG_CC_ASAN
configure.ac:27: error: possibly undefined macro: AC_PROG_CC_LSAN
configure.ac:28: error: possibly undefined macro: AC_PROG_CC_UBSAN
configure.ac:44: error: possibly undefined macro: AM_CONDITIONAL
```

Vudentz, 2024-09-30:
> I'm trying to build bluez-5.7.4 from source. First of all I think README needs fixing because it says that you need to run `./configure` but the file doesn't exists, but there is `configure.ac` that [...]

Are you talking about building from source tree or from a tarball? From source you probably want to take a [bootstrap](https://github.com/bluez/bluez/blob/master/bootstrap) or [bootstrap-configure](https://github.com/bluez/bluez/blob/master/bootstrap-configure)

aramcpp, 2024-10-01:
Thanks running bootstrap fixed the issue

aramcpp, 2024-10-03:
what about updating the README with this information?

falbrechtskirchinger, 2024-10-03:
> what about updating the README with this information?

https://github.com/bluez/bluez/blob/8146d8f7dd675d7b0f8e997f39add23b1cc4e578/README#L21

closed 2024-10-10 by Vudentz