bluez/bluez #968: Manufacturer-specific data bigger than 53 bytes - removed from D-BUS message

dmitryk-nca, 2024-10-05:
Bluez version 5.64
Kernel 6.8.0-40-generic #40~22.04.3-Ubuntu x86_64
BT card is AX201 by Intel

Advertisement message with a manufacturer data longer than 53 bytes - is sent to a D-BUS with the data removed. I use Bleak Python library and it cannot pick the data from D-BUS.

Example of a **correct** D-BUS message
org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_D9_39_4A_AC_6E_5A): ['org.bluez.Device1', {'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {****: <dbus_fast.signature.Variant ('ay', bytearray(b'\x02\xfc6\xff\xfc\x00e\x00\x01\x00\x00\x00\x01\n\xa0E\x96\xda\x0cpl\x1c'))>})>}, []]

Example of an **incorrect** D-BUS message (for a longer data)
org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DA_E3_B0_41_31_20): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -75)>}, []]

Log of a **correct** message size from btmon and Bleak (sorry, I had to remove the manufacturer ID):
> HCI Event: LE Meta Event (0x3e) plen 79                                                                                                                                                    #111 [hci0] 72.242701
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0001
            Props: 0x0001
              Connectable
            Data status: Complete
          Address type: Random (0x01)
          Address: DA:E3:B0:41:31:20 (Static)
          Primary PHY: LE Coded
          Secondary PHY: LE Coded
          SID: 0x00
          TX power: 127 dBm
          RSSI: -82 dBm (0xae)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x35
        02 01 06 15 09 41 41 41 5f 61 36 64 36 35 39 38  .....AAA_a6d6598
        65 38 30 39 30 37 66 39 30 1b ff ** ** 10 00 65  e80907f90..9...e
        00 43 02 2a 00 00 00 00 00 00 0a e2 3e 00 00 00  .C.*........>...
        00 00 00 0c d0                                   .....           
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): AAA_a6d6598e80907f90
        Company: ********
          Data: 1000650043022a0000000000000ae23e0000000000000cd0
@ MGMT Event: Unknown (0x002f) plen 69                                                                                                                                                   {0x0001} [hci0] 72.242763
        00 00 20 31 41 b0 e3 da 02 ae 00 00 00 00 35 00  .. 1A.........5.
        02 01 06 15 09 41 41 41 5f 61 36 64 36 35 39 38  .....AAA_a6d6598
        65 38 30 39 30 37 66 39 30 1b ff ** ** 10 00 65  e80907f90..9...e
        00 43 02 2a 00 00 00 00 00 00 0a e2 3e 00 00 00  .C.*........>...
        00 00 00 0c d0            

scan_callback DA:E3:B0:41:31:20: AAA_a6d6598e80907f90 AdvertisementData(local_name='AAA_a6d6598e80907f90', manufacturer_data={****: b'\x10\x00e\x00C\x02*\x00\x00\x00\x00\x00\x00\n\xe2>\x00\x00\x00\x00\x00\x00\x0c\xd0'}, rssi=-82)

Example of an **incorrect** kind of a message:
> HCI Event: LE Meta Event (0x3e) plen 89                                                                                                                                                     #95 [hci0] 65.315350
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0001
            Props: 0x0001
              Connectable
            Data status: Complete
          Address type: Random (0x01)
          Address: DA:E3:B0:41:31:20 (Static)
          Primary PHY: LE Coded
          Secondary PHY: LE Coded
          SID: 0x00
          TX power: 127 dBm
          RSSI: -75 dBm (0xb5)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x3f
        02 01 06 15 09 41 41 41 5f 61 36 64 36 35 39 38  .....AAA_a6d6598
        65 38 30 39 30 37 66 39 30 25 ff ** ** 10 00 7a  e80907f90%.9...z
        00 29 02 2c 00 00 00 00 00 00 0a 99 45 00 16 00  .).,........E...
        00 00 00 0c ce 00 00 00 00 00 00 00 00 00 b8     ............... 
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): AAA_a6d6598e80907f90
        Company: ***
          Data: 10007a0029022c0000000000000a99450016000000000cce000000000000000000b8
@ MGMT Event: Unknown (0x002f) plen 79                                                                                                                                                   {0x0001} [hci0] 65.315366
        00 00 20 31 41 b0 e3 da 02 b5 00 00 00 00 3f 00  .. 1A.........?.
        02 01 06 15 09 41 41 41 5f 61 36 64 36 35 39 38  .....AAA_a6d6598
        65 38 30 39 30 37 66 39 30 25 ff ** ** 10 00 7a  e80907f90%.9...z
        00 29 02 2c 00 00 00 00 00 00 0a 99 45 00 16 00  .).,........E...
        00 00 00 0c ce 00 00 00 00 00 00 00 00 00 b8     ............... 

scan_callback DA:E3:B0:41:31:20: AAA_a6d6598e80907f90 AdvertisementData(local_name='AAA_a6d6598e80907f90', rssi=-75)

Vudentz, 2024-10-07:
@dmitryk-nca if you want to advertise big chunks data then you probably gonna need Extended Advertising feature and make sure [this](ea242fbf860284c13cf17f46c9731b148da4a6ac) is included.

dmitryk-nca, 2024-10-07:
In my case, the advertising device is set to use an extended advertising - as it can be seen from the report:

LE **Extended Advertising** Report (0x0d)

The issue is on the receiving side (BlueZ) - and looks like it is similar to #832. Honestly - I am a bit struggling to understand in which version it is fixed.

Vudentz, 2024-10-07:
It has been merged upstream: https://github.com/bluez/bluez/commit/f40c5857d70a91e417e22da9da20f4193414b8a3

Vudentz, 2024-10-07:
Check if the kernel has the following:

https://github.com/bluez/bluetooth-next/commit/db08722fc7d46168fe31d9b8a7b29229dd959f9f
https://github.com/bluez/bluetooth-next/commit/112b5090c21905531314fee41f691f0317bbf4f6

dmitryk-nca, 2024-10-07:
Thanks, I will check this tomorrow.
But if the issue is in the kernel - how comes that btmon shows the data
correctly?

Vudentz, 2024-10-07:
> Thanks, I will check this tomorrow. But if the issue is in the kernel - how comes that btmon shows the data correctly?

It shows correctly over HCI not over MGMT.

Vudentz, 2024-10-10:
> Not sure I understand. In the log of btmon - MGMT event has same data as the original message: Data length: 0x3f 02 01 06 15 09 41 41 41 5f 61 36 64 36 35 39 38 .....AAA_a6d6598 65 38 30 39 30 37 66 39 30 25 ff ** ** 10 00 7a e80907f90%.9...z 00 29 02 2c 00 00 00 00 00 00 0a 99 45 00 16 00 .).,........E... 00 00 00 0c ce 00 00 00 00 00 00 00 00 00 b8 ............... Flags: 0x06 LE General Discoverable Mode BR/EDR Not Supported Name (complete): AAA_a6d6598e80907f90 Company: *** Data: 10007a0029022c0000000000000a99450016000000000cce000000000000000000b8 @ MGMT Event: Unknown (0x002f) plen 79 {0x0001} [hci0] 65.315366 00 00 20 31 41 b0 e3 da 02 b5 00 00 00 00 3f 00 .. 1A.........?. 02 01 06 15 09 41 41 41 5f 61 36 64 36 35 39 38 .....AAA_a6d6598 65 38 30 39 30 37 66 39 30 25 ff ** ** 10 00 7a e80907f90%.9...z 00 29 02 2c 00 00 00 00 00 00 0a 99 45 00 16 00 .).,........E... 00 00 00 0c ce 00 00 00 00 00 00 00 00 00 b8 ...............
> […](#)

It is an unknown event it is generating, so it really seems like it is not being processed as a Device Found event, that perhaps explain why the daemon doesn't understand it.

Vudentz, 2024-10-10:
@dmitryk-nca can you attack the raw/binariy HCI trace using btmon -w <filename>?

closed 2025-07-19 by github-actions[bot]