bluez/bluez #969: LEAdvertisingManager1 - Handling of invalid instances

kdewald, 2024-10-05:
Hi! I've got a simple feature request (or maybe a bug report)

Could we add a property to `LEAdvertisingManager1` that lists all paths that have been registered and/or allow a mechanism to deregister them if the paths are unknown?

I got into a situation where I had a test program calling `RegisterAdvertisement` on a path that would be handled by my application but would ignore all incoming calls to `ObjectManager` and `Properties` before exiting.

After calling this program multiple times, the number of `SupportedInstances` became 0 and would stay there indefinitely. I had to restart the Bluetooth service to clear it.

Vudentz, 2024-10-07:
@kdewald Sounds like a bug if the instances are not cleanup after the process that register them is terminated, or the processes are still alive? If they are then that is probably what is taken ownership of the instances, and with respect to having a way to unregister them Id say that would break ownership so Im not inclined to that without a very strong reason.

kdewald, 2024-10-07:
Nope, processes did gracefully exit on both cases, just forgot to call `deregister` before returning.

Is there some data collection I can do with btmon that will help?

Vudentz, 2024-10-07:
```
bluetoothctl[351116]: @ MGMT Close: bluetoothctl                                                                      {0x0002} 14:34:17.833051
memcheck-amd64-[350917]: = src/agent.c:agent_disconnect() Agent :1.37429 disconnected                                          14:34:17.841144
memcheck-amd64-[350917]: = src/agent.c:remove_default_agent() Default agent cleared                                            14:34:17.841379
memcheck-amd64-[350917]: = src/adapter.c:set_mode() sending set mode command for index 0                                       14:34:17.841425
memcheck-amd64-[350917]: = src/agent.c:agent_destroy() agent :1.37429                                                          14:34:17.841763
memcheck-amd64-[350917]: = src/agent.c:agent_unref() 0x54981f0: ref=0                                                          14:34:17.841845
memcheck-amd64-[350917]: = Adv Monitor app :1.37429 disconnected from D-Bus                                             [hci0] 14:34:17.842126
memcheck-amd64-[350917]: = src/adv_monitor.c:app_destroy() Destroy Adv Monitor app :1.37429 at path /                          14:34:17.842193
memcheck-amd64-[350917]: = src/advertising.c:client_disconnect_cb() Client disconnected                                        14:34:17.849874
memcheck-amd64-[350917]: @ MGMT Command: Set Bondable (0x0009) plen 1                                          {0x0001} [hci0] 14:34:17.850792
        Bondable: Disabled (0x00)
@ MGMT Event: Command Complete (0x0001) plen 7                                                                 {0x0001} [hci0] 14:34:17.850797
      Set Bondable (0x0009) plen 4
        Status: Success (0x00)
        Current settings: 0x003e0ac1
          Powered
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
          Wideband Speech
          CIS Central
          CIS Peripheral
          ISO Broadcaster
          Sync Receiver
memcheck-amd64-[350917]: = src/shared/mgmt.c:send_request() [0x0000] command 0x0009                                            14:34:17.850958
memcheck-amd64-[350917]: = src/shared/mgmt.c:can_read_data() [0x0000] command 0x0009 complete: 0x00                            14:34:17.851417
memcheck-amd64-[350917]: = src/adapter.c:new_settings_callback() Settings: 0x003e0ac1                                          14:34:17.851523
memcheck-amd64-[350917]: = src/adapter.c:settings_changed() Changed settings: 0x00000010                                       14:34:17.851612
memcheck-amd64-[350917]: = src/adapter.c:settings_changed() Pending settings: 0x00000000                                       14:34:17.851696
memcheck-amd64-[350917]: @ MGMT Command: Remove Advertising (0x003f) plen 1                                    {0x0001} [hci0] 14:34:17.852160
        Instance: 1
< HCI Command: LE Set Extended Advertising Enable (0x08|0x0039) plen 6                                               #7 [hci0] 14:34:17.852199
        Extended advertising: Disabled (0x00)
        Number of sets: 1 (0x01)
        Entry 0
          Handle: 0x01
          Duration: 0 ms (0x00)
          Max ext adv events: 0
memcheck-amd64-[350917]: = src/shared/mgmt.c:send_request() [0x0000] command 0x003f                                            14:34:17.852273
> HCI Event: Command Complete (0x0e) plen 4                                                                          #8 [hci0] 14:34:17.852899
      LE Set Extended Advertising Enable (0x08|0x0039) ncmd 1
        Status: Success (0x00)
[3667241984]: < HCI Command: LE Remove Advertising Set (0x08|0x003c) plen 1                                          #9 [hci0] 14:34:17.852940
        Handle: 1
> HCI Event: Command Complete (0x0e) plen 4                                                                         #10 [hci0] 14:34:17.853891
      LE Remove Advertising Set (0x08|0x003c) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                                 {0x0001} [hci0] 14:34:17.853912
      Remove Advertising (0x003f) plen 1
        Status: Success (0x00)
        Instance: 1
memcheck-amd64-[350917]: = src/shared/mgmt.c:can_read_data() [0x0000] command 0x003f complete: 0x00                            14:34:17.863905
```

With bluetoothctl at least it works just fine as it detects the client has disconnected from D-Bus and cleanup properly.

kdewald, 2024-10-10:
@Vudentz, thanks for the reply.

> with respect to having a way to unregister them Id say that would break ownership so Im not inclined to that without a very strong reason.

100% agreed. I'd maybe change my request to having some sort of introspection on what processes are holding each path, not so much for production use, but to know if one is making a mistake and how to address it.

From what I've been able to test, it seems the issue could be a race condition or some error handling on the Bluez side if the client is not behaving as intended, which can have it miss the case when the client process ends unexpectedly. It seems to happen quite reliably in my environment, but it's hard to know where to look to find out what is wrong.

If it helps, I can cook an example on my project that should get Bluez to that state and share it with you, or gather some diagnostic data. Feel free to request whatever info you need and I'll do my best to get it.

Vudentz, 2024-10-10:
> @Vudentz, thanks for the reply.
>
>
> 100% agreed. I'd maybe change my request to having some sort of introspection on what processes are holding each path, not so much for production use, but to know [...]

Hmm I think the process is most likely no crashing, but perhaps getting unresponsive, otherwise the kernel would detect that and cleanup its sockets including the D-Bus one which is then causes a D-Bus disconnection/name owner changed signal.

Anyway if you find a way to crash a process and that don't generate a disconnect signal then the bug would be in the D-Bus daemon itself, so it wouldn't be up to bluetoothd to detect that a process has crashed it is the D-Bus daemon.

Regarding the option to list instances, perhaps that could be arranged if perhaps we have something like a testing/developers option which enables features not really meant for application, which we can also break APIs, etc.

kdewald, 2024-10-11:
When I'm running these tests, the process is exiting successfully, it doesn't crash or become unresponsible, but it's possible that some cleanup might not be effective.

Is there some additional cleanup that is required after calling `RegisterAdvertisement` aside from calling `UnregisterAdvertisement`? What happens if the process exits normally but fails to perform that cleanup?

If all depends on DBus properly detecting this issue, can you share some pointers on how I might be able to figure this out?

kdewald, 2024-10-11:
@Vudentz I just noticed something that could be useful.

When my program finished, I saw the following message in `dbus-monitor`:

```
error time=1728625139.208778 sender=org.freedesktop.DBus -> destination=:1.974 error_name=org.freedesktop.DBus.Error.NoReply reply_serial=565
   string "Message recipient disconnected from message bus without replying"
```

That method call corresponds to the following:

```
method call time=1728625117.818864 sender=:1.974 -> destination=:1.1012 serial=565 path=/custom/advertisements/potato; interface=org.freedesktop.DBus.Properties; member=Set
   string "org.bluez.LEAdvertisement1"
   string "TxPower"
   variant       int16 7
```

I noticed there was a bug in my code that was causing the response to this method call to never be sent. Could it be possible that the LEAdvertisingManager is still waiting for that response even in the case where the child process exits?

Vudentz, 2024-10-11:
@kdewald don't think that would matter if there is an ongoing message or not, well except if that does affect the D-Bus daemon to not generate the disconnected signal e.g:

`memcheck-amd64-[350917]: = src/advertising.c:client_disconnect_cb() Client disconnected `

kdewald, 2024-10-11:
Do you know someone from the D-Bus team that we can ask?

Vudentz, 2024-10-11:
You can probably report issues to https://github.com/bus1/dbus-broker/issues but perhaps this is a binding issue, what D-Bus binding are you using?

closed 2025-07-18 by github-actions[bot]