bluez/bluez #970: systemd: enable restart on failure

myaaaaaaaaa, 2024-10-08:
When `bluetoothd` crashes, it becomes impossible to use my bluetooth-only keyboard[^1] until I restart my computer. In those situations, it would be very helpful if systemd could automatically restart `bluetoothd`.
[^1]: Yes, this setup was a mistake.

Fixing this should be as simple as uncommenting the `#Restart=on-failure` line in `bluetooth.service` (which I've confirmed to work by editing my locally installed files):

https://github.com/bluez/bluez/blob/31e5b8f206791f0f96c20d9ea358208b88e44dfe/src/bluetooth.service.in#L12

Vudentz, 2024-10-08:
> When `bluetoothd` crashes, it becomes impossible to use my bluetooth-only keyboard[1](#user-content-fn-1-e8648153c58775e0d1b9ff8e7f6279be) until I restart my computer. In those situations, it would be very helpful if systemd could automatically restart `bluetoothd`.
> 
> Fixing this should be as simple as uncommenting the `#Restart=on-failure` line in `bluetooth.service` (which I've confirmed to work by editing my locally installed files):
> 
> https://github.com/bluez/bluez/blob/31e5b8f206791f0f96c20d9ea358208b88e44dfe/src/bluetooth.service.in#L12
> 
> ## Footnotes
> 
>     1. Yes, this setup was a mistake. [↩](#user-content-fnref-1-e8648153c58775e0d1b9ff8e7f6279be)

Well I guess that is more of a workaround if you ask me, the real fix is to fix the crash though.

myaaaaaaaaa, 2024-10-09:
> Well I guess that is more of a workaround if you ask me, the real fix is to fix the crash though.

That would of course be ideal, but investigating and fixing bugs takes time and energy. In the meantime, `Restart=on-failure` is a one-line change that also applies to future crashes.

myaaaaaaaaa, 2025-05-04:
Friendly ping

myaaaaaaaaa, 2025-10-20:
Friendly ping

myaaaaaaaaa, 2025-11-20:
Friendly ping