bluez/bluez #972: Invalid MTU request?!

danergo, 2024-10-08:
Folks, I hope you read this forum.

There is machine with latest (compiled from scratch) bluez.

Sometimes, we're facing with weird MTU requests, which is only visible in "btmon", but not our higher-level C++ application.

Please see this screenshot directly from btmon's output examined in wireshark:

[image]

You can see (in 2nd and 3rd line) a standard MTU request / MTU response sequence. That's visible within our application too.

However sometimes, we see these invalid MTU requests: client sends MTU request of 247, and bluez immediately sends back a new larger MTU request 517. We don't know where this number is coming from but in the logs it's clearly visible it's coming from localhost (i.e., not from the remote client).

Is this intended/known?

danergo, 2024-10-10:
ping

danergo, 2024-10-10:
pong

Vudentz, 2024-10-10:
@danergo ATT socket is under bluetoothd control, so you really can't have your implementation of that on top of the BlueZ daemon. GAP/GATT are exclusive to bluetoothd since there are very tied to the whole functioning of the Bluetooth LE stack, which is why we don't allow application to mess up with MTU negotiation.

IF you don't intend to use the ATT implementation from BlueZ then don't run the bluetoothd to begin with.

danergo, 2024-10-10:
Thank you!

Do you have any idea why is Bluetoothd requesting 517 MTU when it receives a request of 247 from att client?

Vudentz, 2024-10-11:
> Thank you!
> 
> Do you have any idea why is Bluetoothd requesting 517 MTU when it receives a request of 247 from att client?

MTU can be different on each direction, so what is happening is that remote ATT client exchange MTU of 247 and the local ATT client exchange MTU of 517, that said it should have been responding with 517 so I think you got something other than BlueZ responding with 247:

```
BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 3, Part F
page 1417:

If a device is both a client and a server, the following rules shall apply:
1. A device's ATT_EXCHANGE_MTU_REQ PDU shall contain the same MTU
as the device's ATT_EXCHANGE_MTU_RSP PDU (i.e. the MTU shall be
symmetric).
2. If MTU is exchanged in one direction, that is sufficient for both directions.
3. It is permitted, (but not necessary - see 2.) to exchange MTU in both
directions, but the MTUs shall be the same in each direction (see 1.)
```

Anyway the default in BlueZ is 517 so I don't think it is BlueZ that is replying with 247, well except if you change it via main.conf:

https://github.com/bluez/bluez/blob/master/src/main.conf#L258

danergo, 2024-10-11:
Remote client has a default MTU of 247. From 10.000 connections, ~9995 times my app (in the host, through BlueZ) receives the MTU request of this client, and sends back a confirmation accepting 247. However, about 5 times, BlueZ sends back immediately 517 (not a reply, but a new request), and in this case my app doesn't even get the MTU request from the client.

So that's my question: why is BlueZ sending MTU requests in return for another MTU request from the client?

See, 139874 in the above screenshot: client sends MTU request (247) to localhost (BlueZ).
Then BlueZ sends back a new MTU request (517) to this client. Why?

Normally, (you can see it in 139827) after client sends this MTU request, my app receives it, and sends back an accept message: in 139829, MTU response (that's sent by my app through BlueZ).

Vudentz, 2024-10-11:
> Remote client has a default MTU of 247. From 10.000 connections, ~9995 times my app (in the host, through BlueZ) receives the MTU request of this client, and sends back a confirmation accepting 247. [...]

Don't think we offer any API for handling ATT requests, or are you opening a socket with CID set to ATT CID using root? That is not meant as API for applications, ATT shall be handled by bluetoothd otherwise don't use it and don't bother upstream with custom solution that just fragment the stack.

danergo, 2024-10-11:
We are opening the HCI interface directly. Isn't it going through BlueZ?

We have similar app than "gatttool": it allows setting mtu. But our app also allows rejecting or accepting the remote mtu requests.

Vudentz, 2024-10-11:
> We are opening the HCI interface directly. Isn't it going through BlueZ?
> 
> We have similar app than "gatttool": it allows setting mtu. But our app also allows rejecting or accepting the remote mtu requests.

Even worst, HCI_RAW sockets are known to interfere with bluetoothd, why not use bluetoothd and bluetoothctl-gatt menu:

https://github.com/bluez/bluez/blob/master/client/bluetoothctl-gatt.rst

danergo, 2024-10-11:
We use standard C calls, like bind and socket,  setsockopt (with BTPROTO_L2CAP), to communicate with gatt clients. In this case we are not using BlueZ at all? Sorry for n00b question!

Vudentz, 2024-10-11:
> We use standard C calls, like bind and socket, setsockopt (with BTPROTO_L2CAP), to communicate with gatt clients. In this case we are not using BlueZ at all? Sorry for n00b question!

Socket interface is part of Linux kernel/linux-bluetooth subsystem, that said SIG reserved PSM requires extra capabilities:

https://github.com/bluez/bluetooth-next/blob/master/net/bluetooth/l2cap_sock.c#L76

The shall be the same with fixed CIDs but it doesn't seem we are checking the process capabilities for some reason, anyway the reasoning is the same since CIG reserved CIDs are subject to qualification we can't just simply have any application open a socket to fixed CIDs.

danergo, 2024-10-12:
Let me rephrase this question. In case we are using native C methods to control bluetooth over HCI; *and* bluetooth.service (BlueZ) is also running, can they disturb each other somehow? Maybe they are competing for the communication, and sometimes BlueZ wins and sends the max MTU?

We needed BlueZ, because we are using "btattach" to get a HCI interface from a serial device.

Vudentz, 2024-10-15:
> Let me rephrase this question. In case we are using native C methods to control bluetooth over HCI; _and_ bluetooth.service (BlueZ) is also running, can they disturb each other somehow? Maybe they [...]

btattach is only need for the kernel driver to pickup the serial device as HCI interface after that if you don't intend to use bluetoothd APIs then you can use HCI_CHANNEL_USER (see ) which disables the processing of HCI command/events in the kernel after that you have full control over the HCI interface and nothing else will touch it.

closed 2024-10-23 by danergo