bluez/bluez #973: Unable to connect bluetooth headset after restarting bluez or rebooting

bobvanderlinden, 2024-10-09:
I also ran into this error while debugging my headset not being able to connect a second time. I'm quite confused by this problem. I've also tried reverting to an older version of my system (one where I knew it worked), but the same problem occurs there too now. Could it be new firmware of my headset?

To reproduce the issue I've done the following steps:

1 `systemctl stop bluetooth.service`
2. (Attempt to) Remove all state from bluez: `sudo rm -rf /var/lib/bluetooth/*`
3. `systemctl start bluetooth.service`
4. Start `blueberry`
5. Bring my headset into pairing mode.
6. Connect to headset in `blueberry`
7. Notice that headset works fine. Also disconnecting and connecting from the laptop and from the headset works fine.
8. Turn off the headset
9. `systemctl restart bluetooth.service`
10. (for the logs below) Start `btmon`
11. Turn on the headset.
12. Notice the headset fails to connect to the laptop. The headset removes the laptop from its paired devices.
13. Stop `btmon`

<details>
<summary>btmon log</summary>

```
Bluetooth monitor ver 5.76
btmon[16782]: = Note: Linux version 6.6.54 (x86_64)                                              0.671997
btmon[16782]: = Note: Bluetooth subsystem version 2.22                                           0.672000
= New Index: 90:65:84:C6:1B:40 (Primary,USB,hci0)                                         [hci0] 0.672001
= Open Index: 90:65:84:C6:1B:40                                                           [hci0] 0.672002
= Index Info: 90:65:84:C6:1B:40 (Intel Corp.)                                             [hci0] 0.672002
bluetoothd[16591]: @ MGMT Open: bluetoothd (privileged) version 1.22                    {0x0001} 0.672004
> HCI Event: Connect Request (0x04) plen 10                                            #1 [hci0] 9.744671
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Class: 0x240418
          Major class: Audio/Video (headset, speaker, stereo, video, vcr)
          Minor class: Headphones
          Rendering (Printing, Speaker)
          Audio (Speaker, Microphone, Headset)
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                          #2 [hci0] 9.744897
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                                              #3 [hci0] 9.745525
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                 #4 [hci0] 9.923506
        Status: Success (0x00)
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Role: Central (0x00)
> HCI Event: Vendor (0xff) plen 4                                                     #5 [hci0] 10.109708
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                          #6 [hci0] 10.112500
        Status: Success (0x00)
        Handle: 256
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                    #7 [hci0] 10.112815
        Handle: 256 Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
btmon[16782]: @ RAW Open: btmon (privileged) version 2.22                              {0x0002} 10.113232
btmon[16782]: @ RAW Close: btmon                                                       {0x0002} 10.113243
> HCI Event: Command Status (0x0f) plen 4                                             #8 [hci0] 10.113454
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                 #9 [hci0] 10.113560
        Scan enable: No Scans (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                          #10 [hci0] 10.114501
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
> ACL Data RX: Handle 256 flags 0x02 dlen 12                                         #11 [hci0] 10.121500
[... 296 lines cut ...]
16. Connect to headset
17. Connecting fails
18. Notice the `Hands-Free Voice gateway failed connect` in the bluetooth logs

<details>
<summary>systemctl --unit bluetooth --follow</summary>

```
Oct 09 23:09:53 nac44250 bluetoothd[22751]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 2C:41:A1:C8:E5:04: Connection refused (111)
Oct 09 23:09:53 nac44250 bluetoothd[22751]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 2C:41:A1:C8:E5:04: Connection refused (111)
Oct 09 23:09:55 nac44250 bluetoothd[22751]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 2C:41:A1:C8:E5:04: Connection refused (111)
```

</details>

Instead of `systemctl restart bluetooth`, a reboot will have the same effect(!):

1. Re-pair headset and connect
2. Notice it being able to connect and reconnect
3. Turn off headset
4. Reboot
5. Turn on headset
6. Notice it not being able to connect anymore

Also worth noting: the bluetooth mouse keeps working.

Vudentz, 2024-10-15:
```
< HCI Command: Link Key Request Negative Reply (0x01|0x000c) plen 6                  #46 [hci0] 10.222647
        Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
```

Looks like the link-keys cannot be found, is /var/lib/bluetooth being wiped out on service restart? Does that has the same result when running bluetoothctl as pairing agent instead of blueberry?

Balssh, 2024-10-15:
> ```
>
> < HCI Command: Link Key Request Negative Reply (0x01|0x000c) plen 6                  #46 [hci0] 10.222647
>
>         Address: 2C:41:A1:C8:E5:04 (Bose Corporation)
>
> ```
>
>
>
> Looks like the link-keys [...]

I'll have to double check tomorrow at my laptop, but I'm positive I'm using bluetoothctl and have same problem. After every second reboot, bluetooth stops working. I also use nixos, but don't think it's related.

ProYourName, 2024-10-19:
I have a same issue. Pairing using bluetoothctl, blueman and bluedevil had no effect.

Vudentz, 2024-10-25:
Are you all using nixos? Does it has some special handling over /var/lib/bluetooth? Or perhaps something with systemd .service file, how does it look like? Here are the defaults:

https://github.com/bluez/bluez/blob/master/src/bluetooth.service.in

ProYourName, 2024-10-25:
> Are you all using nixos?
No, i use arch. Journalctl show many errors like this:
Bluetooth: hci0: ACL packet for unknown connection handle 3804

Balssh, 2024-10-25:
A little update: got bluez updated to 5.78 and now it seems to be working fine.

bobvanderlinden, 2024-10-26:
I tried to bisect to older versions of nixos, but it never seemed to work no matter which bluez version. However, I didn't reboot between switching of versions, so it was always with the same kernel. Other NixOS users have also found that switching kernels did fix the issue (https://github.com/NixOS/nixpkgs/issues/346443#issuecomment-2433370903).

It has been resolved for me now.

maru-sama, 2024-10-30:
Hello, I have the same issue with kernel 6.10.13 and 6.11.5. After an initial pairing I can connect my headset but after a restart of the machine it fails to connect. If I boot up with an older 6.8 kernel I am able to connect immediately without re-pairing. 
My mouse works without any issues with all kernels.
I am currently compiling 6.10.14 to check if it fixes the issue. Attached you will find the btmon log

The machine is running Ubuntu 24.04.
[btmon.log](https://github.com/user-attachments/files/17578888/btmon.log)

Vudentz, 2024-10-30:
@maru-sama is the link-key present in /var/lib/bluetooth? Does it show as paired in the likes of bluetoothctl?

It would also be great if you guys can collect the btmon logs when bluetoothd is restarted so we can check if this is not due something like:

commit 532f8bcd1c2c4e8112f62e1922fd1703bc0ffce0
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Aug 27 14:37:22 2024 -0400

    Revert "Bluetooth: MGMT/SMP: Fix address type when using SMP over BREDR/LE"
    
    This reverts commit 59b047bc98084f8af2c41483e4d68a5adf2fa7f7 which
    breaks compatibility with commands like:
    
    bluetoothd[46328]: @ MGMT Command: Load.. (0x0013) plen 74  {0x0001} [hci0]
            Keys: 2
            BR/EDR Address: C0:DC:DA:A5:E5:47 (Samsung Electronics Co.,Ltd)
            Key type: Authenticated key from P-256 (0x03)
            Central: 0x00
            Encryption size: 16
            Diversifier[2]: 0000
            Randomizer[8]: 0000000000000000
            Key[16]: 6ed96089bd9765be2f2c971b0b95f624
            LE Address: D7:2A:DE:1E:73:A2 (Static)
            Key type: Unauthenticated key from P-256 (0x02)
            Central: 0x00
            Encryption size: 16
            Diversifier[2]: 0000
            Randomizer[8]: 0000000000000000
            Key[16]: 87dd2546ededda380ffcdc0a8faa4597
    @ MGMT Event: Command Status (0x0002) plen 3                {0x0001} [hci0]
          Load Long Term Keys (0x0013)
            Status: Invalid Parameters (0x0d)
    
    Cc: stable@vger.kernel.org
    Link: https://github.com/bluez/bluez/issues/875
    Fixes: 59b047bc9808 ("Bluetooth: MGMT/SMP: Fix address type when using SMP over BREDR/LE")
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>

maru-sama, 2024-10-30:
I see the following for the headphone in the info file

```
[LinkKey]
Key=675E8BA2930E28EF88306A0F2D5XXXX <-- modified by me
Type=4
PINLength=0

```
and this is during a restart of bluetooth service

[restart.log](https://github.com/user-attachments/files/17579291/restart.log)

I do not see the errors during the key loading though.

maru-sama, 2024-10-30:
6.10.14 does not fix the issue I still get the same error
```
 ACL Data TX: Handle 256 flags 0x00 dlen 10                                                                                                                              #24 [hci0] 7.562387
      L2CAP: Information Request (0x0a) ident 2 len 2
        Type: Fixed channels supported (0x0003)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                    #25 [hci0] 7.568017
        Num handles: 1
        Handle: 256 Address: 04:5D:4B:80:CA:CD (Sony Corporation)
        Count: 1
        #24: len 10 (16 Kb/s)
        Latency: 5 msec (5-10 msec ~8 msec)
> HCI Event: Link Key Request (0x17) plen 6                                                                                                                               #26 [hci0] 7.569074
        Address: 04:5D:4B:80:CA:CD (Sony Corporation)
< HCI Command: Link Key Request Negative Reply (0x01|0x000c) plen 6                                                                                                       #27 [hci0] 7.569275
        Address: 04:5D:4B:80:CA:CD (Sony Corporation)
```

maru-sama, 2024-10-30:
Booting with the standard 6.8.0 without changing anything works immediately

```
 HCI Event: Link Key Request (0x17) plen 6                                                                                                                              #25 [hci0] 10.426378
        Address: 04:5D:4B:80:CA:CD (Sony Corporation)
< HCI Command: Link Key Request Reply (0x01|0x000b) plen 22                                                                                                              #26 [hci0] 10.426441
        Address: 04:5D:4B:80:CA:CD (Sony Corporation)
        Link key: 675e8ba2930e28ef88306a0f2d5ccafc
> HCI Event: Command Complete (0x0e) plen 10                                                                                                                             #27 [hci0] 10.427354
      Link Key Request Reply (0x01|0x000b) ncmd 1
        Status: Success (0x00)
        Address: 04:5D:4B:80:CA:CD (Sony Corporation)
```

maru-sama, 2024-10-30:
I can confirm that this fixes the issue. I re-build the ubuntu bluez package with this patch added and the headphone connected immediately after a reboot to 6.11.5. Thank you for your support.

maru-sama, 2024-10-31:
Does this mean that for newer kernels either bluez 5.78 or a backport of this patch  is required for the BT devices to work correctly, or has this been fixed in the 6.12 release cycle in the kernel?

icegood, 2024-10-31:
is it the same I have in the latest ubuntu?
https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/2084963

Vudentz, 2024-10-31:
> Does this mean that for newer kernels either bluez 5.78 or a backport of this patch is required for the BT devices to work correctly, or has this been fixed in the 6.12 release cycle in the kernel?

I think you always gonna need the userspace because there could be invalid keys in the storage and the latest kernel just ignore them which is probably what is happening here, anyway I incline to close this as duplicate of https://github.com/bluez/bluez/issues/875.

maru-sama, 2024-10-31:
> is it the same I have in the latest ubuntu? https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/2084963

I just updated my notebook and see the same issues. Either updating the package to 5.78 or adding the one patch mentioned earlier should fix this issue.

bobvanderlinden, 2024-11-03:
Thanks for all the feedback. I think the conclusion is to upgrade or patch. Since this has already been fixed in bluez 5.78 I'll close this issue.

closed 2024-11-03 by bobvanderlinden