bluez/bluez #974: how to set scan window and scan interval in main.conf

eiSouthBoy, 2024-10-11:
Hi

First I started the bluetoothd service. When I used bluetoothctl to send the "scan on" command, I could see the device being discovered in the window at first, but after a while, the scan stopped, and then resumed after a while.
But I have an application scenario where I want the adapter to keep scanning, so what should I do? Of course, I understand that this may be related to "scan window" and "scan interval", and I also see some related parameters in the configuration file "main.conf". I hope to set "scan window" = 0x0012 and "scan interval" = 0x0012, but I don't know how to set it specifically?

Environment introduction:
OS: ubuntu24.04
bluez version: 5.72

Hope to get your reply, thank you.

eiSouthBoy, 2024-10-11:
Of course, I saw the following discovery-related parameters in main.conf:
[BR]
# BR/EDR Inquiry scan activity configuration
#InquiryScanType=
#InquiryScanInterval=
#InquiryScanWindow=

[LE]
# LE scanning parameters used for active scanning supporting discovery
# ​​proceedure
#ScanIntervalDiscovery=
#ScanWindowDiscovery=

But I don't know which one or more to set, and whether there is any connection between them.

eiSouthBoy, 2024-10-12:
my controller is dual-mode.

I tried to send the "scan le" command in the bluetoothctl app, and found that the scan kept going, which is a bit like "hcitool lescan". Of course, this is what I want.
But I still have some confusion, are bredr inquiry and le scans mutually exclusive?
Looking forward to your answer, thank you.

Vudentz, 2024-10-15:
As far I known BR/EDR Inquiry and LE Scan procedures cannot be run in parallel, at least I haven't encounter any controller that can do that.

closed 2024-10-15 by Vudentz