bluez/bluez #975: Distinguish Central and Peripheral devices

kdewald, 2024-10-11:
I hope this is a quick one.

On the list of devices that appear under `/org/bluez/hci0`, for example. Is there a way of knowing if the device has been added as a result of scanning vs one that connected after advertising?

I checked the the following documentation https://github.com/bluez/bluez/blob/master/doc/org.bluez.Device.rst but haven't been able to identify if there's a field I can use to distinguish one or the other.

Thanks!

Vudentz, 2024-10-11:
> I hope this is a quick one.
>
> On the list of devices that appear under `/org/bluez/hci0`, for example. Is there a way of knowing if the device has been added as a result of scanning vs one that [...]

Well if you discovering then it and the device is not connected/paired then it is the result of the discovery. Why does that matters though?

kdewald, 2024-10-13:
Sure, I also came up with the solution to the trivial case, but this doesn't really solve the problem when you have more than one thing interacting with Bluez at the same time, as would happen with a user having the Bluetooth Settings page open.

Your approach requires the developer to know the sequence of events that caused a device to appear in the first place or try to infer it based on some other information, but it doesn't seem to be straightforward.

weliem, 2024-10-13:
Since my app acts as a peripheral, I want to ignore all other connected peripherals and only deal with centrals

Vudentz, 2024-10-14:
Well the problem is then that devices may not be lock into a single role though, specially the likes of Phones, Laptops, etc they are dual-mode, also assumptions to central role being GATT client, etc, are not correct either and there is also role change procedures just to make matters worse.

I think the best approach is not actually based on role but use a service/profile procedures.

kdewald, 2024-10-14:
It's not about devices being locked into a certain role, but to leave some trace of how the device was added. This is important in use cases where the operation of that device is stateful in some manner, otherwise the user has to make assumptions about this or responsibility is pushed to the device itself, which can't always be relied upon.

Vudentz, 2024-10-15:
> It's not about devices being locked into a certain role, but to leave some trace of how the device was added. This is important in use cases where the operation of that device is stateful in some [...]

Right, then we must persist the method in which it was added but that is not available for devices already in the storage which are loaded at startup. Is there anything similar on a different OS to begin with? Or you are trying to implement some device specific policy or a per-device state?

abqjln, 2025-01-18:
I have the same issue and raised/closed in #1073 (should have searched more diligently). 

One use case is when device appears/connects in a dual role system, it needs to know whether to locally stop advertising (remote device in a central role) or to locally stop scanning (remote device is peripheral role).

The "initiator" I need is part of the link layer state machine, like the "connected" state and only valid for the duration of that connection. Different from bonded and paired which are usually persistent. Agreed that initiator and acceptor roles could indeed change for each new connection, for example if two dual-role devices are communicating. It is ephemeral. This is different layer than GATT client/server roles

I am new to diving into bluez source, but I've been looking at the device.c bearer_state.initiator. It is set in device_connect_le, but I can't see how that would work--seems like a different concept.

Also looking in mgmt-api.txt at the Device Connected Event Bit3 Initiated Connection, but alas not sure how to call or use that. Or even if it the right flag to separate central and peripheral? Thanks for all your help.

abqjln, 2025-01-18:
@Vudentz I'm now realizing that there are initiators in mesh that may be confusing the conversation.

Here is a clearer example using the Silicon Labs stack. Their sl_bt_evt_connection_opened has the role (central/peripheral) info that I'd ideally have accessible similarly to the device Connected property.

[https://docs.silabs.com/bluetooth/6.2.0/bluetooth-fundamentals-connections/connection-process](url)

closed 2025-07-18 by github-actions[bot]