bluez/bluez #977: writes to /dev/hidrawN for bluetooth keyboard not working after reconnect

pfps, 2024-10-11:
I have two Bluetooth devices, a Logitech MX Master 3 mouse and a Logitech Craft Keyboard.

The hidraw interface to the two devices act differently since the recent changes to the bluez software.

The mouse acts normally.  When I first connect the mouse reads and writes to the hidraw node for it work as expected.  In particular commands sent elicit the expected responses.  When the mouse is turned off the hidraw node for it disappears.  When the mouse is turned back on an hidraw node for the mouse appears and reads and writes again work as expected.

The keyboard does not act normally.  When I first connect the keyboard reads and writes to the hidraw node for it work as expected.  In particular commands sent elicit the expected responses.  But when the keyboard is turned off the hidraw node for it does not disappear.  When the keyboard is turned back on no new hidraw node appears.  Spontaneous input from the keyboard can be read from this node.  But commands to this node to not elicit the expected response, or indeed any response at all.  It is as if writes to the node are not being sent to the device.

I am puzzled as why there should be a difference between the behaviour of the two devices.  The only difference that I know of between the two devices that is relevant to Linux, besides their device types, is that the Logitech device driver sends commands to the mouse at connect time to change its behaviour but none to the keyboard.

What can I do to find out just what is happening?  I have programs that just read and write to the devices using their hidraw node.

Vudentz, 2024-10-12:
@pfps what version? There has been quite a few changes that try to address this sort of problem, and there are also quite a few issues already open with similar problem.

pfps, 2024-10-12:
Sorry, I should have included that information.

I am running Fedora 39 / XFCE with bluez 5.77.

I see a couple of issues that might be related but nothing about writes to hidraw not working.   Is there an issue with this problem?

pfps, 2024-10-12:
I upgraded to Fedora 40 with bluez 5.78 and the problem is still occurring.

Vudentz, 2024-10-25:
@pfps could you please check with the above changes?

pfps, 2024-10-25:
Will do, but how do I install and run the PR?

closed 2024-10-28 by BluezTestBot

pfps, 2024-10-29:
@Vudentz Will do, but how do I install and run the new version and do I need to do anything to set up the device?