bluez/bluez #979: The client segfaults when the dbus connection cannot be obtained and libdbus is compiled with DBUS_DISABLE_CHECKS

AeonianDusk, 2024-10-13:
In the `main` function of `client/main.c`, we have these two lines:

```c
  dbus_conn = g_dbus_setup_bus(DBUS_BUS_SYSTEM, NULL, NULL);
  g_dbus_attach_object_manager(dbus_conn);
```

On error, `g_dbus_setup_bus` returns `NULL`, which is passed to `g_dbus_attach_object_manager`, then `object_path_ref`, and finally `dbus_connection_get_object_path_data`. Normally, this function returns `FALSE` if `connection` is `NULL`, but this behavior cannot be relied on. If the user has compiled libdbus with `DBUS_DISABLE_CHECKS`, then libdbus will skip this check and cause a`NULL` pointer dereference.

I'm not familiar with this codebase, but I think the case where `dbus_conn` is `NULL` should be handled before calling `g_dbus_attach_object_manager`, because this is what already happens in the `connect_dbus` function found in `src/main.c`.

Vudentz, 2024-10-15:
@AeonianDusk that might fix the segfault but we would have to exit anyway since we can't connect to system bus there is no point on continuing after that, so I guess the real issue is that there is isn't a system bus to begin with so are you sure your environment is configured properly?

AeonianDusk, 2024-10-16:
Yes, my system was misconfigured, and I already fixed that. To be clear, I'm saying that the program should have reported an error and quit cleanly instead of crashing without explanation.