bluez/bluez #981: Observing A2DP disconnect from ubuntu for incoming connection

rbujala, 2024-10-14:
Hi,

we tried connecting rpi and ubuntu device from iphone.
observing a2dp disconnect from ubuntu device sometimes.

connect and disconnect ubuntu device with iphone multiple times.
after 3-4 repeats, we were observing a2dp disconnection in ubuntu device
due to disconnect timer timeout of 1 second.

same scenario we have verified on raspberry pi device, tried several
times and it didn't failed in any iteration.

observations:
pulseaudio versions on ubuntu and rpi varies as below
ubuntu v15.99.1
rpi    v14.2 

there is a difference in pulseaudio with respect to sbc codec.
In 15.99.1, it supports different types of sbc codec settings like
below
https://github.com/pulseaudio/pulseaudio/blob/v15.99.1/src/modules/bluetooth/a2dp-codec-util.c#L68
    &pa_a2dp_endpoint_conf_sbc,
    &pa_a2dp_endpoint_conf_sbc_xq_453,
    &pa_a2dp_endpoint_conf_sbc_xq_512,
    &pa_a2dp_endpoint_conf_sbc_xq_552,

In 14.2 it supports only single sbc codec setting
https://github.com/pulseaudio/pulseaudio/blob/v14.2/src/modules/bluetooth/a2dp-codec-util.c#L33
const pa_a2dp_codec *pa_a2dp_codecs[] = {
    &pa_a2dp_codec_sbc,
};

as number of capabilities increased in pulseaudio v15.99.1, it is taking more time than disconnect timer(1sec)
in processing getcapabilities and due to this timeout a2dp signalling starting again from ubuntu device, 
at somepoint both setconfiguration collides and leading to a2dp connection failure.

logs:
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from DC:B5:4F:A0:78:7F
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_DC_B5_4F_A0_78_7F: SOURCE_STATE_DISCONNECTED -> SOURCE_STATE_CONNECTING
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to DC:B5:4F:A0:78:7F
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=1004
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:set_disconnect_timer() timeout 1
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received DISCOVER_CMD
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x562abb4990c0: Get_Capability_Ind
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x562abb492ec0: Get_Capability_Ind
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x562abb45c1d0: Get_Capability_Ind
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
Oct 14 11:42:19 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x562abb474890: Get_Capability_Ind
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x562abb46acd0: Get_Capability_Ind
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x562abb488cc0: Get_Capability_Ind
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_resp() DISCOVER request succeeded
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 1 type 0 media 0 in use 0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 2 type 0 media 0 in use 0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 3 type 0 media 0 in use 0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 4 type 0 media 0 in use 0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 5 type 0 media 0 in use 0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 6 type 0 media 0 in use 0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:discover_cb() version 0x0103 err (nil)
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/source.c:discovery_complete() Discovery complete
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/media.c:media_endpoint_async_call() Calling SelectConfiguration: name = :1.2921 path = /MediaEndpoint/A2DPSink/sbc
Oct 14 11:42:20 ubuntu pulseaudio[3923256]: D: [pulseaudio] bluez5-util.c: dbus: path=/MediaEndpoint/A2DPSink/sbc, interface=org.bluez.MediaEndpoint1, member=SelectConfiguration
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:a2dp_config() a2dp_config: selected SEP 0x562abb45c1d0
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_set_configuration() 0x562abb4bc510: int_seid=8, acp_seid=1
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received SET_CONFIGURATION_CMD
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:session_cb()
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Configuration not supported (41)
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:setconf_cfm() Sink 0x562abb45c1d0: Set_Configuration_Cfm
Oct 14 11:42:20 ubuntu bluetoothd[3654452]: profiles/audio/a2dp.c:invalidate_remote_cache() Invalidating Remote SEP from cache

version details:
bluez       5.65
kernel      5.15.0-91-generic
pulseaudio  15.99.1

Is there any workaround or fix for this issue?

Thanks & Regards,
Raghavender

Vudentz, 2024-10-15:
@rbujala check with the patch above if the issue can still be reproduced.

rbujala, 2024-10-17:
@Vudentz  thanks for the patch
issue didn't observed with this patch. Tried 10-15 iterations.