bluez/bluez #986: Sony WH-CH700N only shows as audio device when connecting twice.

ToxicMushroom, 2024-10-20:
Below are my reproduction steps, terminal on the left using bluetoothctl to connect (same behavior occurs with blueman or gnome bluetooths ui thing), and pavucontrol on the right to show available audio devices.

1st connect:
In this state headphones don't give audible feedback of this connection 
And other devices like my android phone can just connect to it while linux thinks it's connected:
[image]

2nd connect:
[image]

I have this issue both on my desktop and on my laptop.
Issue didn't happen on windows or on android.

Please let me know what extra information I can provide to help figure out what's causing this :c and how I can provide it if not trivial.

Vudentz, 2024-10-21:
@ToxicMushroom can you attach the btmon trace? What version are you running?

ToxicMushroom, 2024-10-21:
Here is my version info, my bluetooth is currently acting up so I'll get you the trace soon.
```
Bluetooth monitor ver 5.78
btmon[75570]: = Note: Linux version 6.11.4-arch1-1 (x86_64)                                                                          0.068246
btmon[75570]: = Note: Bluetooth subsystem version 2.22                                                                               0.068247
= New Index: 00:1A:7D:DA:71:13 (Primary,USB,hci0)                                                                             [hci0] 0.068248
= Open Index: 00:1A:7D:DA:71:13                                                                                               [hci0] 0.068248
= Index Info: 00:1A:7D:DA:71:13 (Cambridge Silicon Radio)                                                                     [hci0] 0.068249
bluetoothd[592]: @ MGMT Open: bluetoothd (privileged) version 1.23                                                          {0x0001} 0.068250
bluetoothctl[75576]: @ MGMT Open: bluetoothctl version 1.23                                                                 {0x0002} 1.476923
```

Vudentz, 2024-10-21:
> Here is my version info, my bluetooth is currently acting up so I'll get you the trace soon.
>
> ```
> Bluetooth monitor ver 5.78
> btmon[75570]: = Note: Linux version 6.11.4-arch1-1 [...]

You will have to leave btmon collecting the logs when you connect so we are able to see what is going on.

ToxicMushroom, 2024-10-21:
I rebooted a couple times but btmon no longer wants to run ;-; 
```
1 merlijn@arch ~ % sudo btmon
Bluetooth monitor ver 5.78
*** buffer overflow detected ***: terminated
[1]    2537 IOT instruction  sudo btmon
```
No coredump, just SIGABRT

ToxicMushroom, 2024-10-21:
```
merlijn@arch ~ % sudo valgrind btmon
==7457== Memcheck, a memory error detector
==7457== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==7457== Using Valgrind-3.23.0 and LibVEX; rerun with -h for copyright info
==7457== Command: btmon
==7457==
Bluetooth monitor ver 5.78
==7457== Syscall param socketcall.bind(my_addr.rc_bdaddr) points to uninitialised byte(s)
==7457==    at 0x4B6F12B: bind (syscall-template.S:120)
==7457==    by 0x10D2F4: UnknownInlinedFun (control.c:996)
==7457==    by 0x10D2F4: open_channel (control.c:1065)
==7457==    by 0x10C6E8: UnknownInlinedFun (control.c:1548)
==7457==    by 0x10C6E8: main (main.c:289)
==7457==  Address 0x1ffefffe06 is on thread 1's stack
==7457==  in frame #1, created by open_channel (control.c:1055)
==7457==
==7457== Syscall param socketcall.bind(my_addr.rc_channel) points to uninitialised byte(s)
==7457==    at 0x4B6F12B: bind (syscall-template.S:120)
==7457==    by 0x10D2F4: UnknownInlinedFun (control.c:996)
==7457==    by 0x10D2F4: open_channel (control.c:1065)
==7457==    by 0x10C6E8: UnknownInlinedFun (control.c:1548)
==7457==    by 0x10C6E8: main (main.c:289)
==7457==  Address 0x1ffefffe08 is on thread 1's stack
==7457==  in frame #1, created by open_channel (control.c:1055)
==7457==
memcheck-amd64-[7457]: = Note: Linux version 6.11.4-arch1-1 (x86_64)                                                                                                                                        0.361824
memcheck-amd64-[7457]: = Note: Bluetooth subsystem version 2.22                                                                                                                                             0.361826
= New Index: 00:1A:7D:DA:71:13 (Primary,USB,hci0)                                                                                                                                                    [hci0] 0.361826
= Open Index: 00:1A:7D:DA:71:13                                                                                                                                                                      [hci0] 0.361827
= Index Info: 00:1A:7D:DA:71:13 (Cambridge Silicon Radio)                                                                                                                                            [hci0] 0.361827
bluetoothd[1769]: @ MGMT Open: bluetoothd (privileged) version 1.23                                                                                                                                {0x0001} 0.361828
bluetoothctl[7473]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.550752
[7473]: @ MGMT Close: bluetoothctl                                                                                                                                                                 {0x0002} 0.551939
bluetoothctl[7476]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.554374
bluetoothctl[7476]: @ MGMT Close: bluetoothctl                                                                                                                                                     {0x0002} 0.555378
bluetoothctl[7479]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.557926
bluetoothctl[7479]: @ MGMT Close: bluetoothctl                                                                                                                                                     {0x0002} 0.558830
bluetoothctl[7483]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.561526
bluetoothctl[7483]: @ MGMT Close: bluetoothctl                                                                                                                                                     {0x0002} 0.562644
bluetoothctl[7490]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.577311
bluetoothctl[7490]: @ MGMT Close: bluetoothctl                                                                                                                                                     {0x0002} 0.578485
bluetoothctl[7493]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.580797
bluetoothctl[7493]: @ MGMT Close: bluetoothctl                                                                                                                                                     {0x0002} 0.581742
bluetoothctl[7496]: @ MGMT Open: bluetoothctl version 1.23                                                                                                                                         {0x0002} 0.583970
[... cut ...]
==7457==    indirectly lost: 0 bytes in 0 blocks
==7457==      possibly lost: 0 bytes in 0 blocks
==7457==    still reachable: 16,768 bytes in 7 blocks
==7457==         suppressed: 0 bytes in 0 blocks
==7457== Rerun with --leak-check=full to see details of leaked memory
==7457==
==7457== Use --track-origins=yes to see where uninitialised values come from
==7457== For lists of detected and suppressed errors, rerun with: -s
==7457== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

Vudentz, 2024-10-21:
> ```
> merlijn@arch ~ % sudo valgrind btmon
> ==7457== Memcheck, a memory error detector
> ==7457== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
> ==7457== Using Valgrind-3.23.0 and [...]

If you don't ctrl+c it doesn't crash? Well perhaps you can keep it running with valgrind then just to collect the logs.

ToxicMushroom, 2024-10-21:
I couldn't get my device to connect anymore so I repaired my device and now I can no longer reproduce it. 

If it occurs again I'll try to get logs of it to post here, thanks for your time and help <3

ToxicMushroom, 2024-11-12:
It started happening again today here's the log of when I pressed connect and got into a ghost-connection state.

<details><summary>btmon log</summary>

````
merlijn@statice ~ % sudo btmon
[sudo] password for merlijn:
Bluetooth monitor ver 5.79
btmon[5994]: = Note: Linux version 6.11.6-arch1-1 (x86_64)                                       0.477090
btmon[5994]: = Note: Bluetooth subsystem version 2.22                                            0.477092
= New Index: 58:96:1D:FF:1D:05 (Primary,USB,hci0)                                         [hci0] 0.477093
= Open Index: 58:96:1D:FF:1D:05                                                           [hci0] 0.477093
= Index Info: 58:96:1D:FF:1D:05 (Intel Corp.)                                             [hci0] 0.477094
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                  {0x0002} [hci0] 0.477094
bluetoothd[753]: @ MGMT Open: bluetoothd (privileged) version 1.23                      {0x0001} 0.477095
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0003} 0.495270
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0003} 0.495278
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0003} 1.495345
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0003} 1.495358
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0003} 2.495697
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0003} 2.495716
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0003} 3.496081
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0003} 3.496101
< HCI Command: LE Add Device To Accept List (0x08|0x0011) plen 7                       #1 [hci0] 3.599986
        Address type: Public (0x00)
        Address: CC:98:8B:AF:45:A7 (SONY Visual Products Inc.)
> HCI Event: Command Complete (0x0e) plen 4                                            #2 [hci0] 3.710575
      LE Add Device To Accept List (0x08|0x0011) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                    #3 [hci0] 3.710758
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 60.000 msec (0x0060)
> HCI Event: Command Complete (0x0e) plen 4                                            #4 [hci0] 3.712557
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                        #5 [hci0] 3.712713
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                            #6 [hci0] 3.714608
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0003} 4.496245
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0003} 4.496265
bluetoothctl[6012]: @ MGMT Open: bluetoothctl version 1.23                              {0x0003} 5.150403
bluetoothctl[6012]: @ MGMT Close: bluetoothctl                                          {0x0003} 5.164156
bluetoothctl[6015]: @ MGMT Open: bluetoothctl version 1.23                              {0x0003} 5.169393
bluetoothctl[6015]: @ MGMT Close: bluetoothctl                                          {0x0003} 5.175444
bluetoothctl[6018]: @ MGMT Open: bluetoothctl version 1.23                              {0x0003} 5.180408
bluetoothctl[6018]: @ MGMT Close: bluetoothctl                                          {0x0003} 5.186390
bluetoothctl[6022]: @ MGMT Open: bluetoothctl version 1.23                              {0x0003} 5.191616
bluetoothctl[6022]: @ MGMT Close: bluetoothctl                                          {0x0003} 5.197497
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0003} 5.496431
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0003} 5.496450
> HCI Event: LE Meta Event (0x3e) plen 54                                              #7 [hci0] 5.720202
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
[... 276 lines cut ...]
> ACL Data RX: Handle 3585 flags 0x02 dlen 5                                          #54 [hci0] 7.288630
      ATT: Write Response (0x13) len 0
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0004} 7.496799
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0004} 7.496818
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                         {0x0004} 8.496984
blueman-manager[5909]: @ RAW Close: blueman-manager                                     {0x0004} 8.497003
[... cut ...]
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                        {0x0004} 11.498278
blueman-manager[5909]: @ RAW Close: blueman-manager                                    {0x0004} 11.498286
blueman-manager[5909]: @ RAW Open: blueman-manager version 2.22                        {0x0004} 12.498551
blueman-manager[5909]: @ RAW Close: blueman-manager                                    {0x0004} 12.498586
^C%                                                                                                       merlijn@statice ~ %
````
</details>

ToxicMushroom, 2024-11-21:
@Vudentz Did you miss my last comment ? Or is the trace not useful ?

Vudentz, 2024-11-21:
> It started happening again today here's the log of when I pressed connect and got into a ghost-connection state.
> btmon log

Looks like it is connecting over LE instead of over BR/EDR, it seems it is programmed to auto-connect once it sees it is advertising over LE, that said Im not sure it doesn't trigger BR/EDR connection, at least that how normally headset behave when they are turned on they automatically connect over BR/EDR but perhaps that has changed to use LE to indicate its presence and they wait the central to start connecting over classic.

ToxicMushroom, 2024-11-24:
Can I force it to always connect one way or another ? 
Can I monitor how my phone connects to it (without special hw) ?

keekkenen, 2025-01-02:
I have the same situation on three headsets. At the first time, they are correctly paired (all Bluetooth codecs available for choice), but after pc or Bluetooth restart I got HSP/HFP mode (CVSD or mSBC) only.

ToxicMushroom, 2025-01-02:
That doesn't sound like the exact same situation, mine doesn't show up as audio output compared to yours having less codecs available. 

Maybe they are caused by the same underlying issue but I couldn't say.

Vudentz, 2025-05-02:
> remains unsolved, here's updated logs on arch-6.14.4
> 
> My repro steps:
> 
>     1. Clicked connect in blueman-manager: around the 21s in btmon.log
> 
>     2. `bluetoothctl connect mac-addr`: around the 50s in btmon.log
> 
> 
> btmon log

Have you tried with:

bluetoothctl> bearer <dev> bredr

ToxicMushroom, 2025-05-16:
```
[bluetoothctl]> bearer --help
Get/Set preferred bearer
Usage:
	 bearer <dev> [last-seen/bredr/le]
[bluetoothctl]> bearer CC:98:8B:AF:45:A7 bredr

Failed to set bredr: org.freedesktop.DBus.Error.UnknownProperty
[bluetoothctl]>
```
does not look successfull

Vudentz, 2025-05-16:
> It did log something about br/edr becoming available here ?
> 
> ```
> [bluetoothctl]> connect CC:98:8B:AF:45:A7
> 
> Attempting to connect to CC:98:8B:AF:45:A7
> Failed to connect: org.bluez.Error.Failed Input/output error
> [CHG] Controller 00:1A:7D:DA:71:13 PowerState: off-enabling
> hci0 class of device changed: 0x6c0104
> [CHG] Controller 00:1A:7D:DA:71:13 Class: 0x006c0104 (7078148)
> hci0 new_settings: powered bondable ssp br/edr le secure-conn
> [CHG] Controller 00:1A:7D:DA:71:13 PowerState: on
> [CHG] Controller 00:1A:7D:DA:71:13 Powered: yes
> [bluetoothctl]> bearer CC:98:8B:AF:45:A7 bredr
> 
> Failed to set bredr: org.freedesktop.DBus.Error.UnknownProperty
> ```

Are you sure you are running with the latest bluetoothd? Otherwise this would indicate that your device is not dual-mode but the logs seems to indicate it is.

ToxicMushroom, 2025-05-16:
I'm running the `bluetoothctl: 5.82` I could try aur/bluez-git ? but its marked out of data for some reason

Vudentz, 2025-05-16:
```
	Name: WH-CH720N
	Alias: WH-CH720N
	Class: 0x00240404 (2360324)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	CablePairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (0000ff01-0000-1000-8000-00805f9b34ff)
	UUID: Vendor specific           (0000ff01-0000-1000-8000-504759592966)
	UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
	UUID: Vendor specific           (8901dfa8-5c7e-4d8f-9f0c-c2b70683f5f0)
	UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
	UUID: Vendor specific           (956c7b26-d49a-4ba8-b03f-b17d393cb6e2)
	UUID: Vendor specific           (df21fe2c-2515-4fdb-8886-f12c4d67927c)
	UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
	UUID: Vendor specific           (fa349b5f-8050-0030-0010-00001bbb231d)
	Modalias: usb:v054Cp0EAFd0109
```

These are not dual-mode, at least not with the firmware right out of the box.

closed 2025-05-20 by BluezTestBot

joakim-tjernlund, 2025-05-21:
> > I have the same situation on three headsets. At the first time, they are correctly paired (all Bluetooth codecs available for choice), but after pc or Bluetooth restart I got HSP/HFP mode (CVSD or mSBC) only.
> 
> I have something similar: Power on HS and there is only HSP/HSF. Disconnect and connect again, then there is A2DP too.

FYI, latest pipewire, wireplumber and bluez master did not help.

Vudentz, 2025-05-21:
> FYI, latest pipewire, wireplumber and bluez master did not help.

Does sound like the same problem though, this was about dual-mode headset and connecting over LE, anyway feel free to create a new issue.

joakim-tjernlund, 2025-05-21:
> Does sound like the same problem though, this was about dual-mode headset and connecting over LE, anyway feel free to create a new issue.

https://github.com/bluez/bluez/issues/1283