bluez/bluez #988: 5.78 (or .77?) breaks DualSense connection

LuNeder, 2024-10-21:
Hey!

Just updated from 5.76 to 5.78

DualSense is broken again:

- Press PS button, DualSense will blink until give up and won't connect
- Remove DualSense from bluetooth list
- Rescan and add DualSense. Connects, controller blue right/left lights turn on but middle white doesn't and inputs have no effect (tho this has been a problem on 5.76 too btw, it would only work properly when the connection started from the DualSense side. pressing the connect button from Linux would make this happen.)
- Press PS button to attempt connecting starting from the gamepad's side, DualSense will blink until give up and won't connect
- Attempt connect button from blueman, errors, Input/Output error

When attempting to connect from the DualSense's side (PS button), nothing shows up in dmesg or systemd journal at all. 

Attempting to connect from Linux's side shows this on systemd journal: 

```
out 21 14:40:10 Luana-X670E panel-7-systray[2577]: Unable to replace properties on 0: Error getting properties for ID
out 21 14:40:15 Luana-X670E bluetoothd[1258]: profiles/input/device.c:control_connect_cb() connect to 88:03:4C:8A:EB:BD: Host is down (112)
out 21 14:40:15 Luana-X670E blueman-manager[3571]: blueman-manager 14.40.15 WARNING  ManagerDeviceMenu:145 fail      : fail g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusFailedError:  br-connection-create-socket
out 21 14:40:15 Luana-X670E blueman-manager[3571]:  (0)
out 21 14:40:15 Luana-X670E blueman-manager[3571]: blueman-manager 14.40.15 WARNING  ManagerDeviceMenu:197 _handle_error_message: bluetoothd reported input/output error. Check its logs for context.
out 21 14:40:15 Luana-X670E bluetoothd[1258]: Controller resume with wake event 0x0
```

DualSense works fine when rebooted to a previous NixOS generation with bluez 5.76. 5.77 never came to NixOS so idk if it worked. 

I'll try to use bluetoothctl to see if anything interesting shows up as soon as I remember how to use it and update this issue (submitting it rn bc I need to reboot first and I don't wana lose what I've already written)

LuNeder, 2024-10-21:
Update: Actually it does show up on bluetoothctl, but it has the same problems as mentioned above (pressing PS button does nothing, running connect from bluetoothctl fails). But: pressing the PS button and then immediately runnning `connect [MAC}` makes it work!

LuNeder, 2024-10-21:
Here's me using bluetoothctl on the three versions

[bluetoothctl-576-working.txt](https://github.com/user-attachments/files/17467463/bluetoothctl-576-working.txt)
[bluetoothctl-577-mostlyBrokenButWorksIfRunConnectExactlyAtTheRightMoment.txt](https://github.com/user-attachments/files/17467465/bluetoothctl-577-mostlyBrokenButWorksIfRunConnectExactlyAtTheRightMoment.txt)
[bluetoothctl-578-SameAs577worksbutkindadoesnt.txt](https://github.com/user-attachments/files/17467466/bluetoothctl-578-SameAs577worksbutkindadoesnt.txt)
[bluetoothctl-578-welltimedconnectneeded.txt](https://github.com/user-attachments/files/17467468/bluetoothctl-578-welltimedconnectneeded.txt)

It actually seems that after a well-timed connect as I mentioned, the problem gets "fixed" until a reboot, I can then connect and disconnect the controller without needing to mess with anything on the PC side. But after a reboot, attempting to start the connection from the controller does nothing. I have to once again do the very well-timed connect after pressing the PS button (fun fact is that the well-timed connect can also be done from blueman, but that's harder to get the timing (another fun fact is that when you do get the timing correctly it'll say error input/output first before connecting shortly after). I tried setting discoverable on and pairable on without doing thew well-timed connect just to see if that'd change anything, but nope: the well-timed connect is always needed at least once.

LuNeder, 2024-10-21:
Uhhh hold on, the problem seems to be unrelated to the bluez version, but related to kernel version in use.

Bluez 5.76 works fine on kernel 6.11.0.  
Bluez 5.76, 5.77 and 5.78 all act up as mentioned above when on kernel 6.11.4.

I’ll try 5.78 on a different kernel version next.

LuNeder, 2024-10-21:
Yeah, the problem is the kernel version 

- Kernel 6.11.4: 5.76, 5.77 and 5.78 are all broken on the way mentioned above
- Kernel 6.11.0: 5.76 works fine (others untested)
- Kernel 6.10.11: 5.78 works fine (others untested)

I’m so sorry for all the confusion, I’ve updated the title to reflect the correct problem: DualSense connection is broken on kernel 6.11.4, on all the three tested bluez versions (5.76, 5.77 and 5.78).

LuNeder, 2024-10-22:
After some troubleshooting (as you can see on the commits linked above), the problem seems to indeed only happen on Kernel 6.11.4. 6.11.3 works fine, and using an older linux-firmware on 6.11.4 does not fix it. 6.11.3 with the latest linux-firmware works.

I'll add some information that might (or not lol) be useful

DS5 firmware or something:

```
[ 1221.530029] hid-generic 0005:054C:0CE6.000C: unknown main item tag 0x0
[ 1221.530115] input: DualSense Wireless Controller as /devices/virtual/misc/uhid/0005:054C:0CE6.000C/input/input26
[ 1221.530244] hid-generic 0005:054C:0CE6.000C: input,hidraw11: BLUETOOTH HID v1.00 Gamepad [DualSense Wireless Controller] on 00:91:9e:55:4e:ce
[ 1221.583870] playstation 0005:054C:0CE6.000C: unknown main item tag 0x0
[ 1221.583922] playstation 0005:054C:0CE6.000C: hidraw11: BLUETOOTH HID v1.00 Gamepad [DualSense Wireless Controller] on 00:91:9e:55:4e:ce
[ 1221.602776] input: DualSense Wireless Controller as /devices/virtual/misc/uhid/0005:054C:0CE6.000C/input/input27
[ 1221.602874] input: DualSense Wireless Controller Motion Sensors as /devices/virtual/misc/uhid/0005:054C:0CE6.000C/input/input28
[ 1221.602943] input: DualSense Wireless Controller Touchpad as /devices/virtual/misc/uhid/0005:054C:0CE6.000C/input/input29
[ 1221.603150] playstation 0005:054C:0CE6.000C: Registered DualSense controller hw_version=0x00000514 fw_version=0x010c000a
```

Even tho the issue does not depend on wether I'm using linux-firmware 20240909 or 20241017 (both firmwares have bluez/dualsense working on 6.11.3 and both have it broken on 6.11.4), I'll add this here in case it's useful:
```diff
diff ./Documentos/firmware-20241017 ./Documentos/firmware-0909
1,15c1,15
< [    4.104178] systemd[1]: Clear Stale Hibernate Storage Info was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/HibernateLocation-8cf2644b-4b0b-428f-9387-6d876050dc67).
< [    4.383685] iwlwifi 0000:4c:00.0: loaded firmware version 89.6b44fa0b.0 ty-a0-gf-a0-89.ucode op_mode iwlmvm
< [    5.818622] [drm] Loading DMUB firmware via PSP: version=0x05001C00
< [    5.818980] [drm] Found VCN firmware Version ENC: 1.31 DEC: 3 VEP: 0 Revision: 3
< [    5.849463] Bluetooth: hci0: Firmware timestamp 2024.18 buildtype 1 build 81528
< [    5.849467] Bluetooth: hci0: Firmware SHA1: 0xa8bb3f39
< [    5.856165] Bluetooth: hci0: Found device firmware: intel/ibt-0041-0041.sfi
< [    5.856179] Bluetooth: hci0: Firmware Version: 91-33.24
< [    6.751440] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
< [    6.752689] Bluetooth: hci0: Found device firmware: intel/ibt-0041-0041.sfi
< [    6.752701] Bluetooth: hci0: Firmware Version: 91-33.24
< [    9.640354] Bluetooth: hci0: Waiting for firmware download to complete
< [    9.640371] Bluetooth: hci0: Firmware loaded in 2820001 usecs
< [    9.694402] Bluetooth: hci0: Firmware timestamp 2024.33 buildtype 1 build 81755
< [    9.694404] Bluetooth: hci0: Firmware SHA1: 0xd028ffe4
---
> [    4.211044] systemd[1]: Clear Stale Hibernate Storage Info was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/HibernateLocation-8cf2644b-4b0b-428f-9387-6d876050dc67).
> [    4.484371] iwlwifi 0000:4c:00.0: loaded firmware version 89.202a2f7b.0 ty-a0-gf-a0-89.ucode op_mode iwlmvm
> [    5.871931] [drm] Loading DMUB firmware via PSP: version=0x05001C00
> [    5.872298] [drm] Found VCN firmware Version ENC: 1.31 DEC: 3 VEP: 0 Revision: 2
> [    5.960785] Bluetooth: hci0: Firmware timestamp 2024.33 buildtype 1 build 81755
> [    5.960787] Bluetooth: hci0: Firmware SHA1: 0xd028ffe4
> [    5.966922] Bluetooth: hci0: Found device firmware: intel/ibt-0041-0041.sfi
> [    5.966937] Bluetooth: hci0: Firmware Version: 120-18.24
> [    6.857781] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
> [    6.859137] Bluetooth: hci0: Found device firmware: intel/ibt-0041-0041.sfi
> [    6.859149] Bluetooth: hci0: Firmware Version: 120-18.24
> [    9.749701] Bluetooth: hci0: Waiting for firmware download to complete
> [    9.749706] Bluetooth: hci0: Firmware loaded in 2822820 usecs
> [    9.802708] Bluetooth: hci0: Firmware timestamp 2024.18 buildtype 1 build 81528
> [    9.802711] Bluetooth: hci0: Firmware SHA1: 0xa8bb3f39
```

`sudo lspci -v | grep -i -C16 Network` (cropped to only the network part):

```
4c:00.0 Network controller: Intel Corporation Wi-Fi 6E(802.11ax) AX210/AX1675* 2x2 [Typhoon Peak] (rev 1a)
	Subsystem: Rivet Networks Killer Wi-Fi 6E AX1675x 160MHz
	Flags: bus master, fast devsel, latency 0, IRQ 45, IOMMU group 24
	Memory at a0100000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [c8] Power Management version 3
	Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [40] Express Endpoint, IntMsgNum 0
	Capabilities: [80] MSI-X: Enable+ Count=16 Masked-
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [14c] Latency Tolerance Reporting
	Capabilities: [154] L1 PM Substates
	Kernel driver in use: iwlwifi
	Kernel modules: iwlwifi
```

yretenai, 2024-10-22:
Seems to be specific to Intel bluetooth controllers (tested non-working on AX210,) non-reproduceable on MediaTek controllers (tested working on MT7922.)

NyaomiDEV, 2024-10-22:
Linux 6.11.4 here too. It gets worse: it also affects my Xbox Series controller! A workaround is `sudo modprobe -r btusb && sudo modprobe btusb`. The first connection works, all the subsequent ones are dropped. (is this even just related to gamepads though?)

Oh yeah, I should note I have an Intel AX200 chip.

LuNeder, 2024-10-27:
Fixed on [Linux](https://lwn.net/Articles/995131/) [6.11.5](https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.11.5)!

closed 2024-10-27 by LuNeder