bluez/bluez #989: `lib/*` should be licensed with `LGPL` (or perhaps even `GPL-2.0-only WITH GCC-exception-2.0 AND Linux-syscall-note`) instead of `GPL-2.0-or-later`.

tysonclugg, 2024-10-22:
**Summary:**
`lib/*` should be licensed with `LGPL` (or perhaps even `GPL-2.0-only WITH GCC-exception-2.0 AND Linux-syscall-note`) instead of `GPL-2.0-or-later`.

As per [my comment against commit c0d5244](https://github.com/bluez/bluez/commit/c0d524486a50e8366c12c5ebea1a4441e9db46aa#r148198289):
> Removing the `$Id$` CVS relic seems to be the intended change here, but it looks like the entire license was replaced by mistake (from "LGPL" to "GPL v2 or later"). 

Here's the diff for `lib/bluetooth.h` from the commit referenced above, on a commit which is meant to be removing "CVS relics" (such as `$Id$`):
```diff
diff --git a/include/bluetooth.h b/include/bluetooth.h
index b5d056df6..8ced10247 100644
--- a/include/bluetooth.h
+++ b/include/bluetooth.h
@@ -8,24 +8,19 @@
  *
  *
  *  This program is free software; you can redistribute it and/or modify
- *  it under the terms of the GNU General Public License version 2 as
- *  published by the Free Software Foundation;
+ *  it under the terms of the GNU General Public License as published by
+ *  the Free Software Foundation; either version 2 of the License, or
+ *  (at your option) any later version.
  *
- *  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS
- *  OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
- *  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT OF THIRD PARTY RIGHTS.
- *  IN NO EVENT SHALL THE COPYRIGHT HOLDER(S) AND AUTHOR(S) BE LIABLE FOR ANY
- *  CLAIM, OR ANY SPECIAL INDIRECT OR CONSEQUENTIAL DAMAGES, OR ANY DAMAGES 
- *  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN 
- *  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF 
- *  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.
+ *  This program is distributed in the hope that it will be useful,
+ *  but WITHOUT ANY WARRANTY; without even the implied warranty of
+ *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
+ *  GNU General Public License for more details.
  *
- *  ALL LIABILITY, INCLUDING LIABILITY FOR INFRINGEMENT OF ANY PATENTS, 
- *  COPYRIGHTS, TRADEMARKS OR OTHER RIGHTS, RELATING TO USE OF THIS 
- *  SOFTWARE IS DISCLAIMED.
+ *  You should have received a copy of the GNU General Public License
+ *  along with this program; if not, write to the Free Software
+ *  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  *
- *
- *  $Id$
  */
 
 #ifndef __BLUETOOTH_H
```
Clearly, this commit changed much more than simply removing the CVS relics.

The fact that `COPYING.LIB` exists (note the reference to `LIB` in the file name) and is all about the LGPL, suggests to me that the LGPL was the intended license for all files in `lib/` - this also aligns with the license before the erroneous commit referenced above.

This is an issue for any project that wishes to link against BlueZ libs, such as [Python](https://github.com/python/cpython/blob/283ea5f3b2b6a18605b8598a979afe263b0f21ce/Modules/socketmodule.h#L117) (and its downstream projects such as [Python Standalone Builds](https://github.com/indygreg/python-build-standalone) where issues such as [this](https://github.com/indygreg/python-build-standalone/issues/331) have been logged). The implication here is that these projects have included GPL code in non-GPL licensed software (ie: in violation of the license), even though the BlueZ license was never meant to be restricted in this manner.

I appreciate that you will want to double (triple?) check all of the above before making any official comments or taking corrective actions (ie: fixing license headers in `lib/*`, and possibly coordinating with downstream packagers to fix fallout such as https://salsa.debian.org/bluetooth-team/bluez/-/commit/ce4adb4d60368ae1fa4a03cd2dcc34071fba0ac8). You're welcome to reach out to me for assistance, I'm happy to help chase whatever loose ends need chasing in order to close this out - IANAL but I'll do what I can in any case, I'm watching comments here on GitHub.

tysonclugg, 2024-10-22:
The [Linux kernel licensing rules](https://www.kernel.org/doc/html/latest/process/license-rules.html) docs are very clear about what license is expected:
> The Linux Kernel is provided under the terms of the GNU General Public License version 2 only (GPL-2.0), as provided in LICENSES/preferred/GPL-2.0, with an explicit syscall exception described in LICENSES/exceptions/Linux-syscall-note, as described in the COPYING file.

Given that the whole point of BlueZ project is to be the "Official Linux Bluetooth protocol stack", it would be very strange if Linux itself disregarded the syscall exception. Indeed the kernel bluetooth sources [include both the `GCC-exception-2.0` and `Linux-syscall-note` exceptions](https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth.git/tree/LICENSES/exceptions).

tysonclugg, 2024-10-22:
I've written up the timeline below which might be easier to follow than my ramblings above, and hopefully helps to clear things up.

## 1st February 1992
[Linux relicensed under "GNU copyleft"](https://kernel.googlesource.com/pub/scm/linux/kernel/git/nico/archive/+/refs/tags/v0.12), aka: [GNU GENERAL PUBLIC LICENSE: Version 2, June 1991](https://www.gnu.org/licenses/old-licenses/gpl-2.0.html).

## 3rd May 2001
BlueZ project [announced](https://web.archive.org/web/20010720194637/http://mhonarc.axis.se/bluetooth-dev/msg01881.html) "for Linux under GPL license".

## 9th March 2002
[Initial revision](https://github.com/bluez/bluez/tree/a48d8b4639f36e6fc2d7e87cac92e178674caaa1) git commit containing COPYRIGHT file with [GNU GENERAL PUBLIC LICENSE: Version 2, June 1991](https://github.com/bluez/bluez/blob/a48d8b4639f36e6fc2d7e87cac92e178674caaa1/COPYING#L25-L26). Importantly the version doesn't include "any later version" in the version specifier, so it's GPL v2 *only*. The `COPYING.LIB` file doesn't exist at this point, so everything seems to be under one license (the correct one).

## 30th October 2005
[Big cleanup of CVS relics](https://github.com/bluez/bluez/commit/c0d524486a50e8366c12c5ebea1a4441e9db46aa) committed with license change from GPL v2 to GPL v2 or later.

I believe this is the unintended but critical change that has caused this issue, made with little understanding of the importance of the distinction between GPL v2 only vs GPL v2 or later - especially as it was made *almost 2 years prior to the [GPL v3 being released on 29th June 2007](https://www.gnu.org/licenses/quick-guide-gplv3)* and with no consultation with project contributors about a proposed license change that I could find.

## 22nd September 2020
[lib: Add SPDX License Identifier](https://github.com/bluez/bluez/commit/3c1d875bf45d436a09e5e428c213f48222d6d8da) commit perpetuates the change from 15 years prior. The correct SPDX License Identifier should likely be `GPL-2.0-only WITH Linux-syscall-note`, possibly also with `GCC-exception-2.0`.

tysonclugg, 2025-05-06:
Not stale, even though the issue dates back many many years.

Vudentz, 2025-10-01:
In case we use SDPX then there doesn't seem to have a specific entry for GPL-2.0-with-<kernel sycall exception>, but the kernel seems to be using GPL-2.0 WITH Linux-syscall-note.

Vudentz, 2025-10-02:
> I understand the desire to close this out with a "clever hack", but I really don't think that approach at a fix ([#1579](https://github.com/bluez/bluez/pull/1579)) is the best path forward. Specifically, I don't think that GPL-2.0-only can be retained when linked with GPL-2.0-or-later code, as [the GPL-3.0 has a clause that might prevent that](https://clugg.net/blog/avoiding-license-ambiguity#gplv3-and-exceptions). The result _might_ be that the GPLv3 is the effective license for the entire source tree when used within any part of the project, _regardless of what is specified in other parts of the project_.
> 
> I suggest that _if_ all GPL-2.0-or-later parts of the software can successfully be restored to GPL-2.0-only, that my concerns above might be alleviated. However, I'm guessing that it really does depend on all GPL licensed files within the project being successfully corrected - not just the narrow set of header files referenced in the PR above.
> 
> As stated previously, I am not a lawyer. This is a complex issue that is worthy of checking by relevant copyright specialists. I urge you to seek assistance if at all possible.

This is a stopgap solution until we migrate to use the kernel UAPI to expose these headers and then libbluetooth will be deprecated.

tysonclugg, 2025-10-02:
I have grave concerns regarding your stopgap solution. My understanding is that the FSF would assert that distributing programs that combine code from different versions of the GPL (2 and 3) into one larger program would either not be allowed, or would be covered entirely by the GPL v3 for the "whole combination" (_emphasis_ mine):
> **What does it mean to say a license is “compatible with the GPL?”** ([#WhatDoesCompatMean](https://www.gnu.org/licenses/gpl-faq.html.en#WhatDoesCompatMean))
> 
 >  It means that the other license and the GNU GPL are compatible; you can combine code released under the other license with code released under the GNU GPL in one larger program.
> 
>  All GNU GPL versions permit such combinations privately; they also permit distribution of such combinations _provided the combination is released under the same GNU GPL version_. The other license is compatible with the GPL if it permits this too.
> 
> GPLv3 is compatible with more licenses than GPLv2: it allows you to make combinations with code that has specific kinds of additional requirements that are not in GPLv3 itself. Section 7 has more information about this, including the list of additional requirements that are permitted.
>
> ...
> 
> **Does the GPL have different requirements for statically vs dynamically linked modules with a covered work?** ([#GPLStaticVsDynamic](https://www.gnu.org/licenses/gpl-faq.html.en#GPLStaticVsDynamic))
> 
> No. _Linking a GPL covered work statically or dynamically with other modules is making a combined work based on the GPL covered work. Thus, the terms and conditions of the GNU General Public License cover the whole combination._ See also [What legal issues come up if I use GPL-incompatible libraries with GPL software?](https://www.gnu.org/licenses/gpl-faq.html.en#GPLIncompatibleLibs)

I am firmly of the opinion that the GPLv3 (referenced in this project via the GPL-2.0-or-later SPDX license identifier) must be removed from all parts of the project.

Vudentz, 2025-10-02:
> I have grave concerns regarding your stopgap solution. My understanding is that the FSF would assert that distributing programs that combine code from different versions of the GPL (2 and 3) into [...]

Not if you are not directly linking with any of this code, if you are using D-Bus alone (which is probably recommended), then you are not linking directly with any of this code.

IPC and message passing are not typically subject to licensing themselves since there are not directly linking with bluetoothd for example, which is another reason why we want to deprecate libbluetooth.

tysonclugg, 2025-10-06:
OK, that makes sense for "applications". But that raises one very big question: What should the Linux kernel do?