bluez/bluez #991: LE Audio bluetoothd crash

kirankrishnappa-intel, 2024-10-23:
Observed bluetoothd crash  when using as pheripheral device.

Usecase:
1. Pair and connect
2. Start music playback on central
3. Pause music streaming
4. Observed crash of bluetoothd.

> bluetoothd[104990]: src/shared/att.c:can_read_data() (chan 0x5555556fdbb0) ATT PDU received: 0x52
bluetoothd[104990]: src/shared/gatt-server.c:write_cb() Write Cmd - handle: 0x0036
bluetoothd[104990]: src/shared/bap.c:ascs_ase_cp_write() Update Metadata
bluetoothd[104990]: src/shared/bap.c:ep_metadata() ep 0x55555572fb60 id 0x01 dir 0x01
bluetoothd[104990]: src/shared/bap.c:stream_metadata() stream 0x55555571b7f0
bluetoothd[104990]: src/gatt-database.c:send_notification_to_device() GATT server sending notification

bluetoothd[104990]: src/shared/att.c:can_read_data() (chan 0x5555556fdbb0) ATT PDU received: 0x52
bluetoothd[104990]: src/shared/gatt-server.c:write_cb() Write Cmd - handle: 0x0036
bluetoothd[104990]: src/shared/bap.c:ascs_ase_cp_write() Disable
bluetoothd[104990]: src/shared/bap.c:ep_disable() ep 0x55555572fb60 id 0x01 dir 0x01
bluetoothd[104990]: src/shared/bap.c:stream_disable() stream 0x55555571b7f0
bluetoothd[104990]: src/shared/bap.c:bap_ucast_set_state() stream 0x55555571b7f0 dir 0x01: streaming -> qos
bluetoothd[104990]: src/shared/bap.c:bap_stream_io_detach() stream 0x55555571b7f0
bluetoothd[104990]: src/shared/bap.c:stream_io_free() fd 23
bluetoothd[104990]: profiles/audio/bap.c:bap_state() stream 0x55555571b7f0: streaming(4) -> qos(2)
bluetoothd[104990]: profiles/audio/bap.c:setup_create_io() setup (nil) stream 0x55555571b7f0 defer true
bluetoothd[104990]: src/shared/bap.c:bt_bap_stream_io_get_qos() in (nil) out 0x55555571b820
bluetoothd[104990]: profiles/audio/bap.c:setup_listen_io() stream 0x55555571b7f0
bluetoothd[104990]: profiles/audio/transport.c:bap_state_changed() stream 0x55555571b7f0: streaming(4) -> qos(2)
bluetoothd[104990]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_C4_75_AB_17_2F_D6/fd0 State=TRANSPORT_STATE_ACTIVE Playing=0
bluetoothd[104990]: profiles/audio/transport.c:media_transport_remove_owner() Transport /org/bluez/hci0/dev_C4_75_AB_17_2F_D6/fd0 Owner :1.332
bluetoothd[104990]: profiles/audio/transport.c:media_owner_free() Owner :1.332
bluetoothd[104990]: profiles/audio/transport.c:media_transport_suspend() Transport /org/bluez/hci0/dev_C4_75_AB_17_2F_D6/fd0 Owner
bluetoothd[104990]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_C4_75_AB_17_2F_D6/fd0: TRANSPORT_STATE_ACTIVE -> TRANSPORT_STATE_IDLE
bluetoothd[104990]: src/shared/bap.c:bap_queue_req() req 0x555555732470 (op 0x05) queue 0x55555571e850
bluetoothd[104990]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[104990]: src/shared/bap.c:stream_notify_state() stream 0x55555571b7f0
bluetoothd[104990]: src/shared/bap.c:stream_notify_qos() stream 0x55555571b7f0
bluetoothd[104990]: src/gatt-database.c:send_notification_to_device() GATT server sending notification
bluetoothd[104990]: src/shared/bap.c:bap_process_queue()
bluetoothd[104990]: src/shared/bap.c:bap_send() req 0x555555732470 len 3

Program received signal SIGSEGV, Segmentation fault.
bap_send (bap=bap@entry=0x555555730c50, req=req@entry=0x555555732470) at src/shared/bap.c:1490
1490            if (!gatt_db_attribute_get_char_data(ascs->ase_cp, NULL, &handle,
(gdb)
(gdb)
(gdb)
(gdb) bt
#0  bap_send (bap=bap@entry=0x555555730c50, req=req@entry=0x555555732470) at src/shared/bap.c:1490
#1  0x000055555563ec73 in bap_process_queue (data=0x555555730c50) at src/shared/bap.c:1537
#2  0x0000555555657b41 in timeout_callback (user_data=<optimized out>) at src/shared/timeout-glib.c:25
#3  0x00007ffff7ec12a8 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff7ec0c24 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff7f156f8 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff7ec0293 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x0000555555657db9 in mainloop_run () at src/shared/mainloop-glib.c:66
#8  0x0000555555658230 in mainloop_run_with_signal (func=func@entry=0x5555555c6f20 <signal_callback>, user_data=user_data@entry=0x0) at src/shared/mainloop-notify.c:189
#9  0x000055555557ab18 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1489
(gdb) p ascs
$1 = (struct bt_ascs *) 0x0
(gdb)

kirankrishnappa-intel, 2024-10-23:
Observed over commit 29174df00

Vudentz, 2024-10-23:
@kirankrishnappa-intel please check with the above change.

kirankrishnappa-intel, 2024-10-24:
@Vudentz  - The fix seems to be working fine. Thanks.

closed 2024-10-24 by BluezTestBot