bluez/bluez #994: when multiple servers disconnect from the same client, only the first disconnect message gets sent

abqjln, 2024-10-24:
I have a client that is connected to three servers on a different machine, each on a different BT adapter running in the same program. When I close the servers, the client receives a disconnect message only from the first server. This leaves my client unaware that the 2nd and 3rd server have disconnected and so it does not try to reconnect.

However, the remaining two disconnect messages are indeed received if I either stop or restart bluetooth on the server machine with 'systemctl stop/restart bluetooth'

It seems the server side takes all three disconnect actions and forms the messages, but bluez only sends the first message? Is there an "fflush" I can call after disconnecting servers?

Vudentz, 2024-10-28:
@abqjln you will need to elaborate what are the servers doing in this case? Are you registering a GATT application, running some socket, or bluetoothctl?

abqjln, 2024-11-03:
@Vudentz Thanks. BLUF I'll close this as I worked around it and now cannot reproduce the error.

The 3 servers are in a single program using C on different Raspberry PIs (3,4) using internal and external BT-8500 adapters, and a Pi5 as central (without the BT-8500 as I last heard there were issues) under bluez 5.78.

I register GATT services for Heart Rate 0x180d, Running Speed and Cadence 0x1814 and Cycling Speed and Cadence 0x1816 using the bluez_inc [https://github.com/weliem/bluez_inc](url) library. The central, also in C, connects to all three.

Because the servers were all running on different adapters, I disconnected them all one after the other and not waiting for a cb to confirm. bluez and binc did not give any busy errors.

I seem to have worked around it by using C->binc->bluez to re-confirm connection before disconnecting, even though I know they are connected since communication is ongoing with notifies and battery level reads. Now when closing all servers, the central properly receives all 3 disconnects. But the reason is conjecture. I refactored that function and many others afterwards and can no longer reproduce the odd delayed disconnect issue. Thank you.