bluez/bluez #995: Bus 001 Device 006: ID 13d3:3602 IMC Networks Wireless_Device

sincorchetes, 2024-10-24:
Hello, I have an issue with MT7925 in nixOS with 5.78 Bluez version.

uname -a
```console
 Linux tank0 6.11.4 #1-NixOS SMP PREEMPT_DYNAMIC Thu Oct 17 13:27:02 UTC 2024 x86_64 GNU/Linux
```

lsusb 

```console
Bus 001 Device 006: ID 13d3:3602 IMC Networks Wireless_Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 [unknown]
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x13d3 IMC Networks
  idProduct          0x3602 Wireless_Device
  bcdDevice            1.00
  iManufacturer           5 MediaTek Inc.
  iProduct                6 Wireless_Device
  iSerial                 7 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00fe
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          8 Config_01
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        224 Wireless
      bFunctionSubClass       1 Radio Frequency
      bFunctionProtocol       1 Bluetooth
      iFunction               4 BT_FUNCTION
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              1 BT_ACL_If
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
[... 362 lines cut ...]
btrtl                  32768  1 btusb
btintel                69632  1 btusb
btbcm                  24576  1 btusb
btmtk                  32768  1 btusb
bluetooth            1073152  44 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
```

dmesg
```console
[    7.365501] Bluetooth: Core ver 2.22
[    7.365509] NET: Registered PF_BLUETOOTH protocol family
[    7.365510] Bluetooth: HCI device and connection manager initialized
[    7.365512] Bluetooth: HCI socket layer initialized
[    7.365514] Bluetooth: L2CAP socket layer initialized
[    7.365516] Bluetooth: SCO socket layer initialized
[    7.865402] Bluetooth: hci0: HW/SW Version: 0x00000000, Build Time: 20240816133202
[    9.195980] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.195984] Bluetooth: BNEP socket layer initialized
[   26.250299] Bluetooth: hci0: Device setup in 18109067 usecs
[   26.250303] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   26.410455] Bluetooth: MGMT ver 1.23
[   26.429836] Bluetooth: RFCOMM TTY layer initialized
[   26.429840] Bluetooth: RFCOMM socket layer initialized
[   26.429843] Bluetooth: RFCOMM ver 1.11
```

johncronan, 2024-10-24:
> systemctl status bluetooth.service
> 
> ```
> bluetooth.service - Bluetooth service
>      Loaded: loaded (/etc/systemd/system/bluetooth.service; enabled; preset: enabled)
>     Drop-In: /nix/store/5lavl80b872zxc3xv7qaiigm2q7y44cd-system-units/bluetooth.service.d
>              └─overrides.conf
>      Active: active (running) since Tue 2024-10-22 21:15:50 CEST; 3h 0min ago
>        Docs: man:bluetoothd(8)
>    Main PID: 1107 (bluetoothd)
>      Status: "Running"
>          IP: 0B in, 0B out
>          IO: 1.9M read, 0B written
>       Tasks: 1 (limit: 75921)
>      Memory: 2.9M (peak: 3.3M)
>         CPU: 27ms
>      CGroup: /system.slice/bluetooth.service
>              └─1107 /nix/store/4jk7imhlawwja56kdvqwiiznjv19zv91-bluez-5.75/libexec/bluetooth/bluetoothd -f
> ```

We need an updated log, here. This is from 5.75.

Vudentz, 2024-10-24:
It most likely the suspend bug that was fixed by:

https://github.com/bluez/bluetooth-next/commit/a2ce7481010a32013b3c332f1f5b62897e9c3009

In the meantime it may be possible to workaround it by disabling USB auto-suspend with the likes of powertop, etc.

sincorchetes, 2024-10-27:
I've upgraded to an unstable entire system and It's working perfectly :D 

```console
bluetooth.service - Bluetooth service
     Loaded: loaded (/etc/systemd/system/bluetooth.service; enabled; preset: ignored)
    Drop-In: /nix/store/z8w9qyj16q8b0h7prg6ici8v1fzgydia-system-units/bluetooth.service.d
             └─overrides.conf
     Active: active (running) since Sun 2024-10-27 13:03:28 CET; 5min ago
 Invocation: 8aa4127d684b4da68f478ec7d11c0fd2
       Docs: man:bluetoothd(8)
   Main PID: 977 (bluetoothd)
     Status: "Running"
         IP: 0B in, 0B out
         IO: 2.2M read, 40K written
      Tasks: 1 (limit: 75918)
     Memory: 3.3M (peak: 3.9M)
        CPU: 34ms
     CGroup: /system.slice/bluetooth.service
             └─977 /nix/store/hfp3xwh0b0kiw7yai4ibf67hmfi9gg94-bluez-5.78/libexec/bluetooth/bluetoothd -f /etc/bluetooth/main.conf

oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSource/faststream
oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSource/faststream_duplex
oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSink/opus_05
oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSource/opus_05
oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSink/opus_05_duplex
oct 27 13:03:44 tank0 bluetoothd[977]: Endpoint registered: sender=:1.94 path=/MediaEndpoint/A2DPSource/opus_05_duplex
oct 27 13:05:51 tank0 bluetoothd[977]: Failed to set mode: Failed (0x03)
oct 27 13:05:55 tank0 bluetoothd[977]: /org/bluez/hci0/dev_F4_4E_FC_A1_01_79/fd0: fd(31) ready
oct 27 13:06:23 tank0 bluetoothd[977]: /org/bluez/hci0/dev_F4_4E_FC_A1_01_79/sep1/fd1: fd(31) ready
```

```console
[sincorchetes@tank0:~/Documents]$ bluetoothctl info
Device F4:4E:FC:A1:01:79 (public)
	Name: DOQAUS CARE 1
	Alias: DOQAUS CARE 1
	Class: 0x00240404 (2360324)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (0000e003-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v03E0p301Bd0100
	Battery Percentage: 0x3c (60)

```

BlueZ version: 5.78
Kernel: Linux tank0 6.11.5 #1-NixOS SMP PREEMPT_DYNAMIC Tue Oct 22 13:51:37 UTC 2024 x86_64 GNU/Linux

The kernel upgrade may have solved the issue because linux-firmware does not have updates from the 20th of October in an unstable branch.