bluez/bluez #996: bluetoothctl sends STOPPING=1 notification to systemd

Vladimir-csp, 2024-10-26:
If `bluetoothctl` is invoked inside a unit that happens to have `Type=notify`, `NotifyAccess=all`, unit erroneously enters `stopping` state.
Perhaps this shared routine was not really meant to be called from a user-facing application.
[Context of discovery](https://github.com/Vladimir-csp/uwsm/issues/45).

Vudentz, 2024-10-28:
> If `bluetoothctl` is invoked inside a unit that happens to have `Type=notify`, `NotifyAccess=all`, unit erroneously enters `stopping` state. Perhaps this shared routine was not really meant to be [...]

Never heard about running bluetoothctl as systemd unit, so Im not surprised it doesn't work with some options, actually after looking at systemd manpage it looks like notify requires the process to do something:

```

Behavior of notify is similar to exec; however, it is
               expected that the service sends a "READY=1" notification
               message via [sd_notify(3)](https://www.man7.org/linux/man-pages/man3/sd_notify.3.html) or an equivalent call when it
               has finished starting up. systemd will proceed with
               starting follow-up units after this notification message
               has been sent. If this option is used, NotifyAccess= (see
               below) should be set to open access to the notification
               socket provided by systemd. If NotifyAccess= is missing
               or set to none, it will be forcibly set to main.
```

So I don't think you should be using notify, it should probably be just exec.

Vladimir-csp, 2024-10-28:
The problem is that bluetoothctl issues notification, which is unexpected, and deactivates a unit (unrelated) it should not deactivate.

Vladimir-csp, 2024-10-28:
> Never heard about running bluetoothctl as systemd unit
It is not being run as a systemd unit. Another systemd unit runs bluetoothctl as part of its operation and gets unexpectedly deactivated.

Vudentz, 2024-10-28:
> > Never heard about running bluetoothctl as systemd unit
> 
> It is not being run as a systemd unit. Another systemd unit runs bluetoothctl as part of its operation and gets unexpectedly deactivated.

Either way I dont think it should use notify as type if it would be calling other binaries like that since they may attempt to notify system of its own process existing, which might be what is happening here.

Vladimir-csp, 2024-10-28:
bluetoothctl is not a daemon, but a control utility, it has no reason to issue such notification. This is clearly a bug.

It may be used in other units which might *have* to use notifications because they are designed to do so.

Vudentz, 2024-10-28:
> bluetoothctl is not a daemon, but a control utility, it has no reason to issue such notification. This is clearly a bug.
> 
> It may be used in other units which might _have_ to use notifications because they are designed to do so.

Well same could be said to the unit invoking bluetoothctl, besides if we turn bluetoothctl into a unit then the whole thing about writing to NOTIFY_SOCKET is based on the environment variable, if that is set all our tools will notify it not only bluetoothctl:

https://github.com/bluez/bluez/blob/master/src/shared/mainloop-notify.c#L56

So Im not sure how binaries are able to distinct when they are the sole binary of a unit file, and therefore need to handle type=notify by itself, versus when they are invoked by another binary which is doing the handling, anyway perhaps the proper solution is to run bluetoothctl as a sole binary of a separate unit file which can then be run with type=exec for example.

Vladimir-csp, 2024-10-28:
> anyway perhaps the proper solution is to run bluetoothctl as a sole binary of a separate unit file which can then be run with type=exec for example.

That is roughly the workaround I suggested for the referenced bugreport.

>  besides if we turn bluetoothctl into a unit then the whole thing about writing to NOTIFY_SOCKET is based on the environment variable, if that is set all our tools will notify it not only bluetoothctl

What is the use case for bluetoothctl or other non-daemon tools to send notifications? They just do a request then exit.

Vudentz, 2024-10-28:
> That is roughly the workaround I suggested for the referenced bugreport.
>
>
> What is the use case for bluetoothctl or other non-daemon tools to send notifications? They just do a request then exit.

bluetoothctl can be run in interactive mode and for example register services as a server (accessible over D-Bus), in that case it should be keep running.

Vladimir-csp, 2024-10-29:
Would you consider limiting notifications to interactive mode or putting them behind a cli argument?

Vudentz, 2024-10-29:
> Would you consider limiting notifications to interactive mode or putting them behind a cli argument?

Yes, perhaps we will have to do that so interactive mode still handles notify option but non-interactive doesn't.

ArtemSmaznov, 2025-05-01:
I'm actively avoiding using bluetoothctl but I think this is still an issue

Vudentz, 2025-05-02:
@ArtemSmaznov please check with the changes above

ArtemSmaznov, 2025-05-02:
@Vudentz thank you for your work. I would love to test this out, but unfortunately I don't know to build this.

I am running Arch Linux x86_64
I've checked out the PR and tried running make in the root of the repo:
```
bluez on  HEAD (571b643)
➜ make
make: *** No targets specified and no makefile found.  Stop.

bluez on  HEAD (571b643)
➜ makepkg -si
==> ERROR: PKGBUILD does not exist.
```
I've tried running `./bootstrap` just in case even if I'm not on Debian and it looks like it did something, but I don't really see the binary anywhere in the repo.
```
configure.ac:21: installing './compile'
configure.ac:36: installing './config.guess'
configure.ac:36: installing './config.sub'
configure.ac:5: installing './install-sh'
configure.ac:5: installing './missing'
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
```
If you can explain to me how to build it, I'm happy to test this out. Already replicated the issue 3 times before trying to build so I have something to compare with.

Vladimir-csp, 2025-05-05:
I've tested #1235 against bluez 5.82-1 in Debian sid. Running `bluetoothctl devices` in a terminal running directly inside `wayland-wm@hyprland.desktop.service` no longer triggers stopping state.

closed 2025-05-05 by BluezTestBot